thingsboard/thingsboard-gateway #636: Thingsboard gateway startup delay

andrkass, 2021-11-19:
I have configured thingsboard gateway with opcua connector listening to around 100 opcua endpoint ids. Even though in connector.log the messages appear that receiving a new data notification and its sending to thingsboard, the startup delay until the first messages appear at thingsboard is like 5...10 minutes
 
""2021-11-19 11:42:58" - DEBUG - [opcua_connector.py] - opcua_connector - 523 - Python: New data change event on node ns=6;s=S71500ET200MP-Station_1.BH-CPU.wd_Temp.Z3-4.act_temp, with val: 1022 and data DataChangeNotification(<opcua.common.subscription.SubscriptionItemData object at 0x7fee2058c940>, MonitoredItemNotification(ClientHandle:{self.ClientHandle}, Value:{self.Value}))"
""2021-11-19 11:42:58" - DEBUG - [opcua_connector.py] - opcua_connector - 529 - [SUBSCRIPTION] Data to ThingsBoard: {'deviceName': 'Temperature Z3-4', 'deviceType': 'temperature sensor', 'attributes': [], 'telemetry': [{'act_temp': '1022'}]}"

I verified this by sending the MQTT message to Mosquito broker instead and having a listener that prints out MQTT messages. There is significant delay during startup.

Which settings to use to have thingsboard gateway immediately publish messages to backend and not wait 5..10 minutes?

tb_gateway.yaml looks like

connectors:
- configuration: opcua.json
  name: OPC-UA
  type: opcua
storage:
  type: memory
  read_records_count: 10
  max_records_count: 100

opcua.json looks like

{
  "server": {
    "name": "OPC-UA  S7 Server",
    "url": "opc.tcp://192.168.0.25:55105",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 10000,
    "security": "None",
    "disableSubscriptions": false,
    "showMap": false,
    "subCheckPeriodInMillis": 100,
...

version 2.5.5.2 on Ubuntu as daemon.

Another question: I don't want to scan for subscriptions every 100ms, i wanna have true publish subscribe so that new data is pushed. How that can be configured?

Finally, how can I get the OPC-UA servers timestamp (I guess now it stamps the time when it receives the telemetry at the gateway)?

samson0v, 2021-11-23:
Hi @andrkass, please, try to use `"disableSubscriptions": true` in the OPC-UA connector config file and let us know if it helps.
There is no ability to use timestamps from the OPC-UA nodes.

Thanks for your interest in ThingsBoard IoT Gateway.

samson0v, 2021-11-29:
Hi @andrkass, did it help, can we close the issue?

andrkass, 2021-11-29:
Hi,
the initial startup latency of the first messages arriving at MQTT broker did not change. I am also confused a bit about 
max_records_count
does that mean that gateway will not send data to thingsboard unless this amount of messages arrive from connectors? I can put 1 but it is still not sending.

Maybe a newer version would solve it? But I had problems with OPC-UA connector in the newer versions that's why I switched back.

zbeacon, 2021-11-30:
Hi @andrkass ,

Could you provide your logs files for investigation purposes?
Also, this error, can relates to RemoteLogging feature (There was an issue) - you can try to disable it.
In order to get information what the OPC-UA connector does - you can set the value of "showMap" parameter to true and logging level in logs.conf file from INFO to DEBUG . After the gateway restart you will be able to see the search steps.
The delay on the begging of the running OPC-UA connector probably caused by recursive search over all nodes on the server on starting.
Please try to use suggestions and let us know about the results.

andrkass, 2021-11-30:
ok i did some tests. Just subscribing to one OPCUA node results in low startup latency of first message to thingsboard. Subscribing to 100 nodes results in first message arriving to thingsboard after 10 minutes. The reason seems to be that even if the first subscription is arriving at the gateway, it is only send to thingsboard once all subscriptions are processed. Is there a setting that once a subscription is processed, it is immediately sent to thingsboard even if not all subscriptions are processed?

samson0v, 2022-01-20:
Hi @andrkass, there is a fix of the OPC-UA speed (the first start will have the same speed but the next other will have x6 more speed). So you can update your Gateway via the master branch and retest it.

andrkass, 2022-01-20:
Thanks,
so i did install it on fresh machine according to https://thingsboard.io/docs/iot-gateway/install/source-installation/
I used my opcua.json configuration enabling subscriptions and I get following error (did not get that in my old 2.5.x version which works nicely

"2022-01-20 17:22:53" - |ERROR| - [_base.py] - _base - _invoke_callbacks - 330 - exception calling callback for <Future at 0x7f84330edf40 state=finished returned Buffer>"
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/home/andrkass/.local/lib/python3.8/site-packages/opcua/client/ua_client.py", line 493, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "/home/andrkass/.local/lib/python3.8/site-packages/opcua/client/ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "/home/andrkass/.local/lib/python3.8/site-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadTooManyPublishRequests: "The server has reached the maximum number of queued publish requests."(BadTooManyPublishRequests)

arunkumar790, 2022-01-29:
I am using version 2.8 and getting the same error. Any help would be great. Thanks

samson0v, 2022-02-03:
Hi @arunkumar790, @andrkass, please try to use the version from the master branch and let us know about the results.

arunkumar790, 2022-02-13:
@samson0v, Thanks for your response. I got the above error while using remote OPCUA server, i tried with local OPCUA server, it worked. i could fetch data

samson0v, 2022-02-14:
@andrkass can we close the issue?

andrkass, 2022-02-14:
sure

Kind regards, Andreas Kassler
====================================
Prof. Dr. Andreas J. Kassler
Computer Science Department
Head of Research Center for Digital Adaptive Manufacturing for Industry 4.0
Universitetsgatan 2
SE-65188 Karlstad (Sweden)
Tel:   ++46 (0)54 700-2168
Fax:   ++46 (0)54 700-1828
eMail: ***@***.***
Skype: andreaskassler
Web:   www.kau.se/forskare/andreas-kassler
Zoom: https://kau-se.zoom.us/my/andrkass

Reply to: thingsboard/thingsboard-gateway ***@***.***>
Date: Monday, 14 February 2022 at 09:20
To: thingsboard/thingsboard-gateway ***@***.***>
Cc: Andreas Kassler ***@***.***>, Mention ***@***.***>

När du skickar e-post till Karlstads universitet behandlar vi dina personuppgifter<https://www.kau.se/gdpr>.
When you send an e-mail to Karlstad University, we will process your personal data<https://www.kau.se/en/gdpr>.

closed 2022-05-20 by imbeacon