thingsboard/thingsboard-gateway #637: [HELP] No Data from OPC-UA connector with TB-gateway 2.9

hangusto, 2021-11-24:
Hello everyone,

I tried to connect the tb-gateway to Thingsboard and OPC-UA server (Kepware) by opcua connector.

[image]
[image]

### But it cannot get the data from OPC-UA server to Thingsboard.

I followed the the configuration setting and it was work on **tb-gateway version 2.6**

Here is my _opcua.json_ which is the same on **tb-gateway version 2.9**:
[image]
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://host.docker.internal:49320",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic256Sha256",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "${Root\\.Objects\\.Simulator\\.Device1}",
        "deviceNamePattern": "G-sensor",
        "attributes": [
        ],
        "timeseries": [
          {
            "key": "G-sensor-x",
            "path": "${G-sensor-x}"
          },
          {
            "key": "G-sensor-y",
            "path": "${G-sensor-y}"
          },
	  {
	    "key": "G-sensor-z",
	    "path": "${G-sensor-z}"
	  }
        ]
      }
    ]
  }
}
```
Does anyone know what are the differences from the versions of gateway?
I want to figure out it. 
Thx.

**Versions:**
 - OS: Windows docker 
 - Thingsboard IoT Gateway version: 2.9
 - Python version: 3.7

samson0v, 2021-11-26:
Hi @hangusto, alternative you can use the next:
```
{
     "key": "G-sensor-x",
     "path": "${ns=2;i=5}"
},
```
I think you can find ns and i arguments in KEPServerEX 6 node info panel.

Thanks for your interest in ThingsBoard IoT Gateway.

samson0v, 2021-12-28:
Hi @hangusto, did it help, can we close the issue?

closed 2022-01-10 by zbeacon

hangusto, 2022-02-07:
@samson0v Thx, it is work.