thingsboard/thingsboard-gateway #639: [BUG]

jayneev, 2021-11-28:
**Describe the bug**
 ‘BrokenPipeError: [Errno 32] Broken pipe’
Steps:
1.	3 BLE end devices CMx4, CMx7, CMx8 connected to TB gateway.
2.	notification data received for 3 min intervals.
3.	devices remains disconnected for 2 min.
4.	Steps #2, #3 repeated again.
After 2-3 above iterations one of the devices CMx4 error’d out during BLE notifications data read with ‘broken pipe’ error, just after stdin.flush

**Connector name (If bug in the some connector):**
BLE Connector

**Error traceback (If available):**
====================================================================================
""2021-11-24 12:20:26" - |DEBUG| - [ble_connector.py] - ble_connector - handleNotification - 337 - Notification received from device {'device_config': {'name': 'CMx4', 'MACAddress': 'E2:E1:C2:57:D7:09',"
""2021-11-24 12:20:27" - |ERROR| - [ble_connector.py] - ble_connector - __get_services_and_chars - 281 - [Errno 32] Broken pipe"
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/thingsboard_gateway/connectors/ble/ble_connector.py", line 268, in __get_services_and_chars
    data = self.__service_processing(device, characteristics_configs_for_processing_by_methods[method])
  File "/usr/lib/python3.8/site-packages/thingsboard_gateway/connectors/ble/ble_connector.py", line 359, in __service_processing
    data = characteristic.read()
  File "/usr/lib/python3.8/site-packages/bluepy/btle.py", line 197, in read
    return self.peripheral.readCharacteristic(self.valHandle)
  File "/usr/lib/python3.8/site-packages/bluepy/btle.py", line 531, in readCharacteristic
    resp = self._getResp('rd')
  File "/usr/lib/python3.8/site-packages/bluepy/btle.py", line 408, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/lib/python3.8/site-packages/bluepy/btle.py", line 362, in _waitResp
    self._stopHelper()
  File "/usr/lib/python3.8/site-packages/bluepy/btle.py", line 293, in _stopHelper
    self._helper.stdin.flush()
BrokenPipeError: [Errno 32] Broken pipe
==================================================================================

**Versions (please complete the following information):**
 - OS: Yocto Linux - version  4.9
 - Thingsboard IoT Gateway version  2.9 (latest from master branch)
 - Python - version 3.8
[Broken-Pipe-Error-debug-logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/7613983/Broken-Pipe-Error-debug-logs.txt)

samson0v, 2021-12-09:
Hi @jayneev, we are remaking the BLE connector now. We will let you know when it is done.

Thanks for your interest in ThignsBoard IoT Gateway!

ndrs-cwt, 2021-12-09:
@samson0v I have this Broken pipe issue too and after this exception happens, gateway can't reconnect to device that has this exception anymore. The only way to connect to that device again is restart gateway process.

Do you have any recommended workaround for this issue while waiting for the new BLE connector?

samson0v, 2021-12-09:
@ndrs-cwt, we fixed the problem, you can update your Gateway from the master branch and try to test it. But we are afraid that it does not help because of a very big waiting time for notification.

ndrs-cwt, 2021-12-13:
@samson0v, Thank you for the new BLE connector, but it doesn't fix my Broken pipe problem but maybe it works for OP case.

I just found some workaround for my problem by record the name of the Broken pipe device in some list
```
except BrokenPipeError:
    log.exception('Broken Pipe. Device %s, wait for remake', device)
    self.__remake_list.append(device)
    continue
```

and delete all of its Peripheral relate objects and let's connector create all related objects again
```
if time.time() - self.__previous_read_time >= self.__check_interval_seconds:
    self.__get_services_and_chars()
    if len(self.__remake_list) > 0:
        for device in self.__remake_list:
            log.warning("Clear %s entry from __devices_around", device)
            del self.__devices_around[device]["peripheral"]
            del self.__devices_around[device]["services"]
            self.__devices_around[device]["is_new_device"] = True # Trig the connector to remake all object
        self.__remake_list.clear()
    self.__previous_read_time = time.time()
```
this is from my [custom BLE connector](https://github.com/ndrs-cwt/thingsboard-gateway/blob/25f48f99d626cfb5febfa90c5d7a73aa30ca002c/thingsboard_gateway/extensions/parking/parking_lock_connector.py) that almost exact copy from TB's BLE connector but remove notify support and always disconnect from device. This band-aid just makes my gw can reconnect to the Broken pipe device again but have to wait for next read interval to recreate device object and start reading data again.

samson0v, 2021-12-17:
Hi @ndrs-cwt, we did a remake of the BLE Connector yesterday, so you can install the new version of Gateway from the master branch and test the connector.
Let us know if the new version of BLE Connector fixes your problem.

jayneev, 2021-12-18:
Thank you @samson0v and @zbeacon for sharing the ble connector remake, We are trying to find right way to get 'data notification updates from BLE peripheral devices, with latest checked out code. Below are few observations and queries , if provided feedback, will definitely help us to get notify data on gateway.

OBSERVATION: Below parameters in ble.json now removed from latest one. Are they replaced with new parameter ?
Old parameters:
======================
"rescanIntervalSeconds": 18000,
"checkIntervalSeconds": 45,
"scanTimeSeconds": 35,
=======================
New parameters: (Is this suffice above requirements?)
=======================
"scanner": {
"timeout": 10000,
"deviceName": "Device name" <==== Is this 'Device name' needs to be replaced with gateway name at TB side?
},
"devices": [
{
......
"pollPeriod": 500000, <==== per devices
.....
}
=======================

QUERY: Is the newly added above json parameter tag 'deviceName' value is the gateway name specified on Thingsboard?

QUERY: Once device connected to gateway, will they get added in TB dashboard as in previous versions?

QUERY: As in QUERY#3 above, are the below JSON parameters still serve purpose of identifying the device connected?
"attributes": [
{
"key": "name",
"method": "read",
"characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
"byteFrom": 0,
"byteTo": -1
}
],
"attributeUpdates": [
{
"attributeOnThingsBoard": "sharedName",
"characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB"
}
],

QUERY: Does server side RPC if not used, should the below feel be empty?
"serverSideRpc": [
{
"methodRPC": "rpcMethod1",
"withResponse": true,
"characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
"methodProcessing": "read"
},
{
"methodRPC": "rpcMethod2",
"withResponse": true,
"characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
"methodProcessing": "write"
},
{
"methodRPC": "rpcMethod3",
"withResponse": true,
"methodProcessing": "scan"
}
]

QUERY: Does current configuration supports ACTIVE scan mode?
If yes, then does setting "passiveScanMode": false, will suffice this requirement?

I appreciate you all for your valuable contribution and help.

Warm Regards,
@jayneev

samson0v, 2021-12-28:
Was discussed and a new issue was opened.

closed 2021-12-28 by samson0v