thingsboard/thingsboard-gateway #641: Bacnet Connector to consistently read Bacnet properties [HELP]

AhmadMagdi, 2021-12-01:
**Describe the issue**
I have The Thingsboard IoT Gateway up on a virtual machine, and I'm trying to get the BACnet connector to work. I have, on the same local network, another virtual machine running a BACnet device using [BACnet Stack](https://github.com/bacnet-stack/bacnet-stack), which can be used to simulate a Bacnet device.
I defined a device in the Bacnet connector configuration file, and gave it a few properties to read which are sent to an Mqtt Broker by the gateway. The problem is, it seems not to be consistent in reading those properties; most of the time it can read the first one and send it to the broker, sometimes it sends the first two, and only a few times does it send the three of them. 

`{"Device1 Abanoub": {"temperature": 98}}`
`{"Device1 Abanoub": {"temperature": 98, "t2": 0.0}}`
`{"Device1 Abanoub": {"temperature": 98, "t2": 0.0, "t3": 98}}`

Also, how do I control when the gateway scans for devices, or do I have to restart it each time?

**Configuration** (Attach your configuration file)

### tb_gateway.yaml:

[tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/7430445/tb_gateway.yaml.txt)

```
thingsboard:
  host: 192.168.0.133
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: EXAMPLE_ACCESS_TOKEN
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
connectors:
  -
    name: BACnet Connector
    type: bacnet
    configuration: bacnet.json

```

### bacnet.json:

[bacnet.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/7630036/bacnet.json.txt)

```
{
  "general": {
    "objectName": "TB_gateway",
    "address": "0.0.0.0:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.0.133:47808",
      "pollPeriod": 10000,
      "attributes": [
        {
          "key": "temperature",
          "type": "string",
          "objectId": "analogOutput:1",
          "propertyId": "units"
        },
        {
          "key": "t2",
          "type": "string",
          "objectId": "analogOutput:0",
          "propertyId": "presentValue"
        },
        {
          "key": "t3",
          "type": "string",
          "objectId": "analogOutput:1",
          "propertyId": "units"
        }
      ],
      "timeseries": [
        {
          "key": "state",
          "type": "bool",
          "objectId": "binaryValue:1",
          "propertyId": "presentValue"
        }
      ]
    }
  ]
}

```

**Connector name (If you need help with some connector/converter):**
BACnet Connector

**Error traceback (If it was raised):**
Those errors sometimes appear in the logs, but I'm not sure if or how they're related to the issue.
```
""2021-11-30 20:01:41" - |ERROR| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - __init__ - 40 - [Errno 98] Address already in use"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 38, in __init__
    super().__init__(self.__device, self.__config["general"]["address"])
  File "/var/lib/thingsboard_gateway/.local/lib/python3.6/site-packages/bacpypes/app.py", line 535, in __init__
    self.mux = UDPMultiplexer(self.localAddress)
  File "/var/lib/thingsboard_gateway/.local/lib/python3.6/site-packages/bacpypes/bvllservice.py", line 96, in __init__
    self.directPort = UDPDirector(self.addrTuple)
  File "/var/lib/thingsboard_gateway/.local/lib/python3.6/site-packages/bacpypes/udp.py", line 155, in __init__
    self.bind(address)
  File "/usr/lib/python3.6/asyncore.py", line 329, in bind
    return self.socket.bind(addr)
OSError: [Errno 98] Address already in use
```

```
""2021-11-30 20:10:50" - |ERROR| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - __general_cb - 152 - During processing callback, exception has been raised:"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 148, in __general_cb
    callback_params["callback"](iocb, callback_params)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/bacnet/bacnet_connector.py", line 222, in __bacnet_device_mapping_response_cb
    converter = callback_params["config"]["uplink_converter"]
KeyError: 'uplink_converter'
```

Here are some of the logs, where the errors appear, for context

[sampled_logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/7630059/sampled_logs.txt)

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04.3 LTS
 - Thingsboard IoT Gateway version: 2.8
 - Python version: 3.6.9

PaulHHolmes, 2022-06-01:
Hi,

It looks like you have a repeated attibute address:-

  "attributes": [
        {
          "key": "temperature",
          "type": "string",
          **"objectId": "analogOutput:1",**
          "propertyId": "units"
        },
        {
          "key": "t2",
          "type": "string",
          "objectId": "analogOutput:0",
          "propertyId": "presentValue"
        },
        {
          "key": "t3",
          "type": "string",
          **"objectId": "analogOutput:1",**
          "propertyId": "units"
        }
      ],

Try changing the third one to something different?

samson0v, 2022-08-02:
Hi @AhmadMagdi, please, update your Gateway to the newest version (3.1) via the master branch or other convenient for you methods and let us know if the bug is gone.

Thanks for your interest in ThingsBoard IoT Gateway!

closed 2022-08-16 by imbeacon