thingsboard/thingsboard-gateway #642: [BUG] Delete a device from thingsboard

mircopz, 2021-12-01:
**Describe the bug**
My problem is when I delete a device from thingsboard (28630045-00-21-000-000000 in the attached traceback), but it is still present in connected_devices.json list (as it should be). 
Then, when my device sends another publish with telemetry data, I see from logs that tb-gateway correctly receives the message, but it is not forwarded to thingsboard.
First question: it is because the device no longer exists on thingsboard? why device is not recreated when receiving the new publish?

Then I tried using the disconnect request to inform tb-gateway that device is disconnected from broker, but I saw the attached exception, and tb-gateway stops working! It no longer receives anything.
Second question: as before, it is because the device no longer exists on thingsboard? how can I automatically recover from an exception like this? 

Third question: how can i behave to inform tb-gateway that a device has been deleted from thingsboard?

Thank you

**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
```
""2021-12-01 10:14:06" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_message - 486 - Disconnecting device 28630045-00-21-000-000000 of type default"
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.6.1-py3.7.egg/paho/mqtt/client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.6.1-py3.7.egg/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.6.1-py3.7.egg/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.6.1-py3.7.egg/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.6.1-py3.7.egg/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.6.1-py3.7.egg/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.6.1-py3.7.egg/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.7/site-packages/paho_mqtt-1.6.1-py3.7.egg/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 487, in _on_message
    self.__gateway.del_device(found_device_name)
  File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 742, in del_device
    del self.__saved_devices[device_name]
KeyError: '28630045-00-21-000-000000'

```

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version 2.9
 - Python version 3.7

zbeacon, 2021-12-01:
Hi @mircopz ,

Please try to use the version form the master branch, it will process correctly saving persistent devices, but unfortunately, there is no way to notify the gateway about the device removal on ThingsBoard at the moment.

mircopz, 2021-12-01:
Thank you @zbeacon 

Now I can't build from sources, I'm using the current docker image, so I'll wait a new release. Do you know if it will be released shortly?

mircopz, 2022-08-03:
> Hi @mircopz ,
> 
> Please try to use the version form the master branch, it will process correctly saving persistent devices, but unfortunately, there is no way to notify the gateway about the device removal on ThingsBoard at the moment.

Hi @imbeacon , now I'm using the version from master branch, and it correctly handles persistent devices list. 
In this way, when Thingsboard sends a gateway_device_deleted RPC to the gateway, the device is removed from persistent devices file and from local __connected_devices array.
The device is still present in local __saved_devices array, so if I recreate a device with the same name, the first time it sends a message to Thingsboard the persistent devices file is not updated since that name is already present in __saved_devices

```python
 def add_device(self, device_name, content, device_type=None):
     if device_name not in self.__saved_devices:
        device_type = device_type if device_type is not None else 'default'
        self.__connected_devices[device_name] = {**content, "device_type": device_type}
        self.__saved_devices[device_name] = {**content, "device_type": device_type}
        self.__save_persistent_devices()
        self.tb_client.client.gw_connect_device(device_name, device_type)
```

Can this lead to any problems? I think the correct way is to remove the deleted device from __save_devices array too.
Is the following code better or it is not necessary?

```python
def __process_deleted_gateway_devices(self, deleted_device_name: str):
    log.info("Received deleted gateway device notification: %s", deleted_device_name)
    if deleted_device_name in list(self.__renamed_devices.values()):
        first_device_name = TBUtility.get_dict_key_by_value(self.__renamed_devices, deleted_device_name)
        del self.__renamed_devices[first_device_name]
        deleted_device_name = first_device_name
        log.debug("Current renamed_devices dict: %s", self.__renamed_devices)
    if deleted_device_name in self.__connected_devices:
        del self.__connected_devices[deleted_device_name]
        log.debug("Device %s - was removed from __connected_devices", deleted_device_name)
    if deleted_device_name in self.__saved_devices:
        del self.__saved_devices[deleted_device_name]
        log.debug("Device %s - was removed from __saved_devices", deleted_device_name)
    self.__save_persistent_devices()
    self.__load_persistent_devices()
```

Please let me know if you think this can be fixed in the master branch, or if I should make a pull request.

samson0v, 2022-08-03:
Hi @mircopz, make a pull request and I will review it.

closed 2022-08-05 by samson0v