thingsboard/thingsboard-gateway #643: [HELP] thingboard_gateway is continuously subscribe with a new id

gohm-tech, 2021-12-02:
**Describe the issue**
Hi,
When the thingsboard_gateway service was running, We execute the mqtt_firmware_client.py. After that we are getting this log. So we are wondering this status will create any problem for us?
[image]

**Server Environment**
own setup
docker deployment
OS: Linux(Ubuntu 20.04)

**Client Environment**
OS: Linux(Ubuntu)
Browser: Chrome, Firefox
Version Ubuntu Version 20.04

**Connector name**
MQTT Connector

**Configuration**
[mqtt_firmware_client.txt](https://github.com/thingsboard/thingsboard-gateway/files/7641374/mqtt_firmware_client.txt)

**Versions:**
OS: Raspbian GNU/Linux 10 (buster)
Thingsboard IoT Gateway version [e.g. 2.7]
Python version 3.7.3

Best Regards,
**Gohm Team**

zbeacon, 2021-12-02:
Hi @gohm-tech ,

Thank you for you interest in ThingsBoard IoT gateway.
This situation can appears when you used the same credentials for the both clients, ThingsBoard doesn't allow connection for two or more clients with the same credentials.
You can change the configuration of your ThingsBoard instance (parameter ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE)
Or use another device credentials for firmware client.

gohm-tech, 2021-12-03:
Hi @zbeacon,
Thank you for your quick reply, After your suggestion, we found this question about this subject
[https://github.com/thingsboard/thingsboard/issues/1210](url)
We indicate the location of the file to help other users.
`thingsboard/application/src/main/resources`

Best Regards,
**Gohm Team**

closed 2021-12-03 by gohm-tech