thingsboard/thingsboard-gateway #644: Gateway gets into endless loop trying to send telemetry data to ThingsBoard server

olafpeters1962, 2021-12-03:
I have an issue with ThingsBoard gateway version 2.9

I send telemetry data via the mqtt connector with the ${serialNumber} (i.e. the device name) set to the device name of the gateway itself:

mosquitto_pub -d -t "/sensor/data" -m '{"serialNumber": "my-test-gw", "sensorType": "EMS-edge-gw", "sensorModel": "Dell laptop", "temp":22.0, "hum":44.0}'

The telemetry data is sent only once to the ThingsBoard server, then the gateway software is getting into an endless loop trying to resend the telemetry data.
Log data of this situation is shown below, it looks like the mqtt connection to the ThingsBoard server gets disconnected all the time (the lines with "OPE - get_event_pack - self.__event_pack =" are debug output added by me to investigate the problem):

021-12-02 22:06:25" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 149 - connection SUCCESS"
""2021-12-02 22:06:26" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes - successfully completed."
""2021-12-02 22:06:26" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes/response - successfully completed."
""2021-12-02 22:06:26" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/rpc - successfully completed."
""2021-12-02 22:06:26" - |INFO| - [memory_event_storage.py] - memory_event_storage - get_event_pack - 50 - OPE - get_event_pack - self.__event_pack = ['{"deviceName": "ope-test-gw", "deviceType": "EMS-edge-gw", "attributes": [{"model": "Dell laptop"}, {"Dell laptop": "on"}], "telemetry": {"ts": 1638479183481, "values": {"temperature": "22.0", "humidity": "44.0", "combine": "44.0:22.0"}}}']"
""2021-12-02 22:06:26" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 128 - Disconnected client: <paho.mqtt.client.Client object at 0x7fab639ee640>, user data: None, result code: 1"
""2021-12-02 22:06:27" - |DEBUG| - [tb_client.py] - tb_client - _on_connect - 82 - TB client <paho.mqtt.client.Client object at 0x7fab639ee640> connected to ThingsBoard"
""2021-12-02 22:06:27" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 149 - connection SUCCESS"
""2021-12-02 22:06:27" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes - successfully completed."
""2021-12-02 22:06:27" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes/response - successfully completed."
""2021-12-02 22:06:27" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/rpc - successfully completed."
""2021-12-02 22:06:27" - |INFO| - [memory_event_storage.py] - memory_event_storage - get_event_pack - 50 - OPE - get_event_pack - self.__event_pack = ['{"deviceName": "ope-test-gw", "deviceType": "EMS-edge-gw", "attributes": [{"model": "Dell laptop"}, {"Dell laptop": "on"}], "telemetry": {"ts": 1638479183481, "values": {"temperature": "22.0", "humidity": "44.0", "combine": "44.0:22.0"}}}']"
""2021-12-02 22:06:27" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - subscribe_to_attribute - 318 - Subscribed to * with id 4"
""2021-12-02 22:06:27" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 179 - Subscribed to *|* with id 4 for device *"
""2021-12-02 22:06:27" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 128 - Disconnected client: <paho.mqtt.client.Client object at 0x7fab639ee640>, user data: None, result code: 1"
""2021-12-02 22:06:27" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __read_data_from_storage - 514 - OPE - NO success - 1: self.tb_client.is_connected() = False"
""2021-12-02 22:06:28" - |DEBUG| - [tb_client.py] - tb_client - _on_connect - 82 - TB client <paho.mqtt.client.Client object at 0x7fab639ee640> connected to ThingsBoard"
""2021-12-02 22:06:28" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 149 - connection SUCCESS"
""2021-12-02 22:06:28" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes - successfully completed."
""2021-12-02 22:06:28" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes/response - successfully completed."
""2021-12-02 22:06:28" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/rpc - successfully completed."
""2021-12-02 22:06:28" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - subscribe_to_attribute - 318 - Subscribed to * with id 5"
""2021-12-02 22:06:28" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 179 - Subscribed to *|* with id 5 for device *"
""2021-12-02 22:06:28" - |INFO| - [memory_event_storage.py] - memory_event_storage - get_event_pack - 50 - OPE - get_event_pack - self.__event_pack = ['{"deviceName": "ope-test-gw", "deviceType": "EMS-edge-gw", "attributes": [{"model": "Dell laptop"}, {"Dell laptop": "on"}], "telemetry": {"ts": 1638479183481, "values": {"temperature": "22.0", "humidity": "44.0", "combine": "44.0:22.0"}}}']"
""2021-12-02 22:06:28" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 128 - Disconnected client: <paho.mqtt.client.Client object at 0x7fab639ee640>, user data: None, result code: 1"
^C""2021-12-02 22:06:29" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __stop_gateway - 243 - Stopping..."
""2021-12-02 22:06:29" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 320 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7fab631c1730>, None, 0)"
""2021-12-02 22:06:29" - |INFO| - [mqtt_connector.py] - mqtt_connector - close - 217 - MQTT Broker Connector has been stopped."
""2021-12-02 22:06:29" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __close_connectors - 236 - Connector MQTT Broker Connector closed connection."
""2021-12-02 22:06:29" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __stop_gateway - 246 - The gateway has been stopped."
""2021-12-02 22:06:29" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_unsubscribe - 187 - Unsubscribed from *|*, subscription id '*'"
""2021-12-02 22:06:29" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 182 - <paho.mqtt.client.Client object at 0x7fab639ee640>"
""2021-12-02 22:06:29" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 183 - Disconnecting from ThingsBoard"

When I set "serialNumber" (device name) to "currentThingsBoardGateway", everythings works fine:

mosquitto_pub -d -t "/sensor/data" -m '{"serialNumber": "currentThingsBoardGateway", "sensorType": "EMS-edge-gw", "sensorModel": "Dell laptop", "temp":23.0, "hum":45.0}"

I couldn't find any documentation about this feature/behaviour. Gateway version 2.8 works fine: the data is transmitted and the gateway doesn't get into an endless loop situation.

Version info:
 - OS: Ubuntu 21.10
 - Thingsboard IoT Gateway version 2.9 (latest github version)
 - Python version 2.7.18

samson0v, 2021-12-03:
Hi @olafpeters1962, can you, pls, send your configuration file?

Thanks for your interest in ThingsBoard IoT Gateway!

olafpeters1962, 2021-12-05:
Hi @samson0v, here are my tb_config.yaml and mqtt.json config files.
[config-files.zip](https://github.com/thingsboard/thingsboard-gateway/files/7655517/config-files.zip)

samson0v, 2022-01-20:
Hi @olafpeters1962, it is not recommended to set deviceName as a Gateway name because of behaviour as above.

closed 2022-05-20 by imbeacon