thingsboard/thingsboard-gateway #649: [BUG] Use rest connector error

klonhj2015, 2021-12-10:
**Describe the bug**
I use the master version to build a version. And I use the rest connector, but when I run the gateway, it encounted an error.

**Connector name (If bug in the some connector):**
[ REST Connector]

**Error traceback (If available):**
Traceback (most recent call last):
  File "/Users/hongquanzhu/Documents/Github/thingsboard-gateway/dev-tb-gw/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/rest/rest_connector.py", line 150, in run
    self.__run_server()
  File "/Users/hongquanzhu/Documents/Github/thingsboard-gateway/dev-tb-gw/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/rest/rest_connector.py", line 144, in __run_server
    ssl_context=ssl_context)
  File "/Users/hongquanzhu/Documents/Github/thingsboard-gateway/dev-tb-gw/lib/python3.7/site-packages/aiohttp/web.py", line 514, in run_app
    loop.run_until_complete(main_task)
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
    return future.result()
  File "/Users/hongquanzhu/Documents/Github/thingsboard-gateway/dev-tb-gw/lib/python3.7/site-packages/aiohttp/web.py", line 321, in _run_app
    await runner.setup()
  File "/Users/hongquanzhu/Documents/Github/thingsboard-gateway/dev-tb-gw/lib/python3.7/site-packages/aiohttp/web_runner.py", line 279, in setup
    self._server = await self._make_server()
  File "/Users/hongquanzhu/Documents/Github/thingsboard-gateway/dev-tb-gw/lib/python3.7/site-packages/aiohttp/web_runner.py", line 373, in _make_server
    self._app._set_loop(loop)
  File "/Users/hongquanzhu/Documents/Github/thingsboard-gateway/dev-tb-gw/lib/python3.7/site-packages/aiohttp/web_app.py", line 224, in _set_loop
    "web.Application instance initialized with different loop"
RuntimeError: web.Application instance initialized with different loop

**Versions (please complete the following information):**
 - OS: [macos]
 - Thingsboard IoT Gateway version [master]
 - Python version[3.7]

klonhj2015, 2021-12-10:
If I checkout version 2.7, the rest server is running with flask, that seems ok. The version of master seems to use aiohttp, error.

samson0v, 2021-12-10:
Hi @klonhj2015, send, please, your REST config file.

Thanks for your interest in ThingsBoard IoT Gateway.

klonhj2015, 2021-12-10:
This is my config file.
{
  "host": "0.0.0.0",
  "port": "35000",
  "SSL": false,
  "mapping":[
    {
      "endpoint": "/aiAlarm",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "${name}",
        "deviceTypeExpression": "default",
        "timeseries": [
           {
             "type": "double",
             "key": "combcsp0",
             "value": "${values.combcsp0}"
           },
          {
             "type": "double",
             "key": "pactcsp0",
             "value": "${values.pactcsp0}"
           }
        ]
      }
    }
  ]
}

samson0v, 2021-12-10:
@klonhj2015 are you running Gateway under Docker container?

klonhj2015, 2021-12-10:
> @klonhj2015 are you running Gateway under Docker container?

No, I run gateway  in a python virtual env

samson0v, 2021-12-10:
@klonhj2015 Send, pls, your tb_gateway.yaml config file

klonhj2015, 2021-12-10:
thingsboard:
  host: 127.0.0.1
  port: 11883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: test
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
   - name: HTTP RestAPI Connector
     type: rest
     configuration: rest.json
  # - name: MQTT Broker Connector
  #   type: mqtt
  #   configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector

klonhj2015, 2021-12-10:
`thingsboard:
  host: 127.0.0.1
  port: 11883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: test
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
   - name: HTTP RestAPI Connector
     type: rest
     configuration: rest.json
  # - name: MQTT Broker Connector
  #   type: mqtt
  #   configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
`

klonhj2015, 2021-12-10:
thingsboard:
  host: 127.0.0.1
  port: 11883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: test
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000

connectors:
   - name: HTTP RestAPI Connector
     type: rest
     configuration: rest.json

samson0v, 2021-12-10:
@klonhj2015 can you run the Gateway on python 3.9 and let us know if it helps?

klonhj2015, 2021-12-11:
It works! I change the python version to 3.0. It seems all done. Thanks.

klonhj2015, 2021-12-11:
Sorry, the first time starting seems ok, but after i restart the gw, the same error.

Congregator, 2021-12-11:
Experiencing the same issue with the REST connector on both native and docker installations of the current release version 2.9. Using Python 3.9. I think the issue might be a change in an external dependency (aiohttp) as new installations of older gateway versions (2.8 and 2.7) show the same issue.

Congregator, 2021-12-11:
I finally figured out the issue - it appears it is related to external dependencies.
The latest pip and pip3 repos have the packages aiohttp and async_timeout at different versions than what the Gateway needs to operate.

At the latest versions, the following packages are installed:
aiohttp - version 3.8.1
async_timeout - version 4.0.1

However, the gateway requires the following versions to operate:
aiohttp - version 3.7.4
async_timeout - version 3.0.1

I believe the latest version of the Gateway REST connector does not specifically request the versions as above and just requests the latest, for which the code is not built to accommodate the changes in the packages.

Doing a standard "pip3 freeze" does not show the packages aiohttp or async_timeout installed under python, so I believe the Gateway is using some virtual environment.
Thus, as I workaround, I downloaded the above packages using pip3 and symbolic linked them to the virtual environment folder from the native collection. i.e:

Standard pip install installs packages under "/usr/local/lib/<python_version>/dist-packages"
Thingsboard stores their package under "/var/lib/thingsboard_gateway/.local/lib/<python_version/site-packages"

So symbolic link "/usr/local/lib/<python_version>/dist-packages/aiohttp" to "/var/lib/thingsboard_gateway/.local/lib/<python_version/site-packages/aiohttp"

Do above symbolic linking for folders:
- aiohttp
- aiohttp-3.7.4.dist-info
- async_timeout
- async_timeout-3.0.1.dist-info

Thereafter restart Thingsboard Gateway. 
This worked for me on Ubuntu 18.04 (as a workaround). I assume similar issues on other platforms.

samson0v, 2021-12-16:
Hi @Congregator, thanks for your reply, we are working on fixing this bug.

closed 2021-12-28 by zbeacon