thingsboard/thingsboard-gateway #651: [HELP] K8s gateway with 2 replicas

ardilom-entel, 2021-12-13:
Hello,
I created an image of thingsboard-gateway (based on v2.6) and it works fine, but now I tried to run it with a replica and it keeps failing and restarting and I don't understand why. What am I doing wrong?

**Configuration**
```
thingsboard:
  host: $THINGSBOARD_HOST
  port: $THINGSBOARD_PORT
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: $THINGSBOARD_GATEWAY_TOKEN
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
connectors:
  - name: REST Connector
    type: rest
    configuration: rest.json
```

**Error traceback (If it was raised):**
[image]

**Versions (please complete the following information):**
 - OS: Debian 10
 - Thingsboard IoT Gateway version 2.6
 - Python version 3.9

zbeacon, 2021-12-14:
Hi @ardilom2,

This behavior related to device authorization on ThingsBoard. By default only one device can be connected with one accessToken. If ThingsBoard detects more than one device for accessToken it dissconnects old session. There are 2 possible solutions for this behavior:
1. Use different accessTokens for gateways.
2. Increase the value for env variable ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE on your ThingsBoard and restart it.
Thank you for your interest in ThingsBoard IoT gateway.

ardilom2, 2021-12-15:
> Hi @ardilom2,
>
> This behavior related to device authorization on ThingsBoard. By default only one device can be connected with one accessToken. If ThingsBoard detects more than one device for [...]

thx! setting this env variable ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE=2 worked perfectly!

closed 2021-12-15 by ardilom2