thingsboard/thingsboard-gateway #652: [HELP]

jitendraspatil, 2021-12-16:
16 06:47:06" - |DEBUG| - [tb_client.py] - tb_client - _on_connect - 82 - TB client <paho.mqtt.client.Client object at 0x7fb1e8bd4350> connected to ThingsBoard"

""2021-12-16 06:47:06" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 138 - connection SUCCESS"

""2021-12-16 06:47:06" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes - successfully completed."

""2021-12-16 06:47:06" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes/response - successfully completed."

""2021-12-16 06:47:06" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/rpc - successfully completed."

""2021-12-16 06:47:06" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fb1e8b94b50>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"

""2021-12-16 06:47:06" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BACnetConnector from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet."

""2021-12-16 06:47:06" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 334 - Error on loading connector:"

""2021-12-16 06:47:06" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 335 - Expecting value: line 47 column 7 (char 1259)"

Traceback (most recent call last):

  File "/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 324, in _load_connectors

    connector_conf = load(conf_file)

  File "/usr/local/lib/python3.7/site-packages/simplejson-3.17.5-py3.7-linux-x86_64.egg/simplejson/__init__.py", line 471, in load

    use_decimal=use_decimal, **kw)

  File "/usr/local/lib/python3.7/site-packages/simplejson-3.17.5-py3.7-linux-x86_64.egg/simplejson/__init__.py", line 525, in loads

    return _default_decoder.decode(s)

  File "/usr/local/lib/python3.7/site-packages/simplejson-3.17.5-py3.7-linux-x86_64.egg/simplejson/decoder.py", line 370, in decode

    obj, end = self.raw_decode(s)

  File "/usr/local/lib/python3.7/site-packages/simplejson-3.17.5-py3.7-linux-x86_64.egg/simplejson/decoder.py", line 400, in raw_decode

    return self.scan_once(s, idx=_w(s, idx).end())

simplejson.errors.JSONDecodeError: Expecting value: line 47 column 7 (char 1259)

""2021-12-16 06:47:06" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_devices - 763 - Loaded devices:

 {'Temp Sensor': 'Modbus Default zeugy'}"

""2021-12-16 06:47:06" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __read_data_from_storage - 431 - Send data Thread has been started successfully."

""2021-12-16 06:47:06" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 152 - Gateway started."

""2021-12-16 06:47:07" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attributes_parse - 263 - Received data: {}"

""2021-12-16 06:47:15" - |WARNING| - [tb_updater.py] - tb_updater - check_for_new_version - 85 - Cannot connect to the update service. PLease check your internet connection."

**Configuration** (Attach your configuration file)
thingsboard:
  host: localhost
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken:***********************
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
connectors:
# - name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json

# -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
  -
    name: BACnet Connector
    type: bacnet
    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector

**Connector name (If you need help with some connector/converter):**
          BACnet Connector
{
  "general": {
    "objectName": "TB-gateway",
    "address": "0.0.0.0:4581",
    "objectIdentifier": 4581,
    "maxApduLengthAccepted": 1024,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 37
  },
  "devices": [
    {
      "deviceName": "BACNet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.1.24:4581",
      "pollPeriod": 10,
      "attributes": [
        {
          "key": "Ctrl_1_MRI_OUT_WTR_TEMP",
          "type": "string",
          "objectId": "analogValue:11",
          "propertyId": "presentValue"
        },
		{
          "key": "Ctrl_1_MRI_RTRN_AIR_RH_1",
          "type": "string",
          "objectId": "analogValue:12",
          "propertyId": "presentValue"
        },
		{
          "key": "Ctrl_1_MRI_RTRN_AIR_RH_2",
          "type": "string",
          "objectId": "analogValue:13",
          "propertyId": "presentValue"
        },
		{
          "key": "Ctrl_1_MRI_HMD__A_M_STS",
          "type": "string",
          "objectId": "analogValue:14",
          "propertyId": "presentValue"
        },
		{
          "key": "Ctrl_1_MRI_HEATER_ON_STS_1",
          "type": "string",
          "objectId": "analogValue:15",
          "propertyId": "presentValue"
        },
      ]     
    }
  ]
}

 Versions (21H1)
 - OS: windows 10 Pro
 - Thingsboard IoT Gateway version   latest
 - Python version 3.7

zbeacon, 2021-12-16:
Hi @jitendraspatil ,

It looks like you have an extra comma symbol after the last configuration object in section attributes. Please remove it and try to restart the gateway.
Thank you for your interest in ThingsBoard IoT Gateway.

jitendraspatil, 2021-12-16:
Hi @zbeacon 
i am not able to push the data in Device

jitendraspatil, 2021-12-16:
""2021-12-16 08:52:07" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 76 - Gateway starting..."

""2021-12-16 08:52:07" - |DEBUG| - [tb_updater.py] - tb_updater - check_for_new_version - 69 - Checking for new version"

""2021-12-16 08:52:07" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 81 - ThingsBoard IoT gateway version: 2.9"

""2021-12-16 08:52:07" - |DEBUG| - [tb_client.py] - tb_client - run - 124 - connecting to ThingsBoard"

""2021-12-16 08:52:07" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 171 - <paho.mqtt.client.Client object at 0x7f1ed68f0ad0>"

""2021-12-16 08:52:07" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 172 - Disconnecting from ThingsBoard"

""2021-12-16 08:52:07" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - subscribe_to_attribute - 301 - Subscribed to * with id 1"

""2021-12-16 08:52:07" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 178 - Subscribed to *|* with id 1 for device *"

""2021-12-16 08:52:07" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log service"

""2021-12-16 08:52:07" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log extension"

""2021-12-16 08:52:07" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log converter"

""2021-12-16 08:52:07" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log connector"

""2021-12-16 08:52:07" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log tb_connection"

""2021-12-16 08:52:07" - |DEBUG| - [memory_event_storage.py] - memory_event_storage - __init__ - 28 - Memory storage created with following configuration: 

Max size: 100000

 Read records per time: 100"

""2021-12-16 08:52:07" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_devices - 763 - Loaded devices:

 {'Temp Sensor': 'Modbus Default zeugy'}"

""2021-12-16 08:52:07" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 35 - Root path is: /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway"

""2021-12-16 08:52:07" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f1ed68f5c50>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-16 08:52:08" - |DEBUG| - [tb_client.py] - tb_client - _on_connect - 82 - TB client <paho.mqtt.client.Client object at 0x7f1ed68f0ad0> connected to ThingsBoard"

""2021-12-16 08:52:08" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 138 - connection SUCCESS"

""2021-12-16 08:52:08" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes - successfully completed."

""2021-12-16 08:52:08" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes/response - successfully completed."

""2021-12-16 08:52:08" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/rpc - successfully completed."

""2021-12-16 08:52:08" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f1ed68b6a10>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"

""2021-12-16 08:52:08" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BACnetConnector from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet."

""2021-12-16 08:52:08" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f1ebfcf1690>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-16 08:52:08" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BACnetUplinkConverter from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet."

""2021-12-16 08:52:08" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f1ebfcf18d0>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-16 08:52:08" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f1ebfcf1b90>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"

""2021-12-16 08:52:08" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f1ebfcf1850>, origin='/usr/local/lib/python3.7/site-packages/thing [... cut ...]
""2021-12-16 08:52:09" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __read_data_from_storage - 431 - Send data Thread has been started successfully."

""2021-12-16 08:52:09" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 152 - Gateway started."

""2021-12-16 08:52:09" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attributes_parse - 263 - Received data: {}"

""2021-12-16 08:52:17" - |WARNING| - [tb_updater.py] - tb_updater - check_for_new_version - 85 - Cannot connect to the update service. PLease check your internet connection."

jitendraspatil, 2021-12-16:
""2021-12-16 08:52:09" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attributes_parse - 263 - Received data: {}"

jitendraspatil, 2021-12-16:
[image]
[image]

jitendraspatil, 2021-12-16:
{
  "general": {
    "objectName": "TB-gateway",
    "address": "127.0.0.1:4581",
    "objectIdentifier": 4581,
    "maxApduLengthAccepted": 1024,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 37
  },
  "devices": [
    {
      "deviceName": "BACNet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.1.24:4581",
      "pollPeriod": 10,
      "attributes": [
        {
          "key": "Ctrl_1_MRI_RUN_STS",
          "type": "string",
          "objectId": "analogValue:1",
          "propertyId": "presentValue"
        },
		{
          "key": "Ctrl_1_MRI_FLTR_DRP_EU_4",
          "type": "string",
          "objectId": "analogValue:2",
          "propertyId": "presentValue"
        },
		{
          "key": "Ctrl_1_MRI_FLTR_DRP_EU_8",
          "type": "string",
          "objectId": "analogValue:3",
          "propertyId": "presentValue"
        }
      ]     
    }
  ]
}

zbeacon, 2021-12-16:
You can try to use 0.0.0.0:4581 for address in general section.

jitendraspatil, 2021-12-16:
{
  "general": {
    "objectName": "TB-gateway",
    "address": "0.0.0.0:4581",
    "objectIdentifier": 4581,
    "maxApduLengthAccepted": 1024,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 37
  },
  "devices": [
    {
      "deviceName": "BACNet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.1.24:4581",
      "pollPeriod": 10,
      "attributes": [
        {
          "key": "Ctrl_1_MRI_RUN_STS",
          "type": "string",
          "objectId": "analogValue:1",
          "propertyId": "presentValue"
        },
		{
          "key": "Ctrl_1_MRI_FLTR_DRP_EU_4",
          "type": "string",
          "objectId": "analogValue:2",
          "propertyId": "presentValue"
        },
		{
          "key": "Ctrl_1_MRI_FLTR_DRP_EU_8",
          "type": "string",
          "objectId": "analogValue:3",
          "propertyId": "presentValue"
        }
      ]     
    }
  ]
}

jitendraspatil, 2021-12-16:
""2021-12-16 09:18:14" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 76 - Gateway starting..."

""2021-12-16 09:18:14" - |DEBUG| - [tb_updater.py] - tb_updater - check_for_new_version - 69 - Checking for new version"

""2021-12-16 09:18:14" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 81 - ThingsBoard IoT gateway version: 2.9"

""2021-12-16 09:18:14" - |DEBUG| - [tb_client.py] - tb_client - run - 124 - connecting to ThingsBoard"

""2021-12-16 09:18:14" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 171 - <paho.mqtt.client.Client object at 0x7f2d0e620c90>"

""2021-12-16 09:18:14" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 172 - Disconnecting from ThingsBoard"

""2021-12-16 09:18:14" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - subscribe_to_attribute - 301 - Subscribed to * with id 1"

""2021-12-16 09:18:14" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 178 - Subscribed to *|* with id 1 for device *"

""2021-12-16 09:18:14" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log service"

""2021-12-16 09:18:14" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log extension"

""2021-12-16 09:18:14" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log converter"

""2021-12-16 09:18:14" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log connector"

""2021-12-16 09:18:14" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log tb_connection"

""2021-12-16 09:18:14" - |DEBUG| - [memory_event_storage.py] - memory_event_storage - __init__ - 28 - Memory storage created with following configuration: 

Max size: 100000

 Read records per time: 100"

""2021-12-16 09:18:14" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_devices - 763 - Loaded devices:

 {'Temp Sensor': 'Modbus Default zeugy'}"

""2021-12-16 09:18:14" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 35 - Root path is: /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway"

""2021-12-16 09:18:14" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f2d0e6169d0>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-16 09:18:14" - |DEBUG| - [tb_client.py] - tb_client - _on_connect - 82 - TB client <paho.mqtt.client.Client object at 0x7f2d0e620c90> connected to ThingsBoard"

""2021-12-16 09:18:14" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 138 - connection SUCCESS"

""2021-12-16 09:18:14" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes - successfully completed."

""2021-12-16 09:18:14" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes/response - successfully completed."

""2021-12-16 09:18:14" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/rpc - successfully completed."

""2021-12-16 09:18:15" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f2d0e5e7250>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"

""2021-12-16 09:18:15" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BACnetConnector from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet."

""2021-12-16 09:18:15" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f2d0c229650>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-16 09:18:15" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BACnetUplinkConverter from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet."

""2021-12-16 09:18:15" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f2d0c229890>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-16 09:18:15" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f2d0c229b50>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"

""2021-12-16 09:18:15" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f2d0c229f10>, origin='/usr/local/lib/python3.7/site-packages/thing [... cut ...]
""2021-12-16 09:18:16" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_devices - 763 - Loaded devices:

 {'Temp Sensor': 'Modbus Default zeugy'}"

""2021-12-16 09:18:16" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __read_data_from_storage - 431 - Send data Thread has been started successfully."

""2021-12-16 09:18:16" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 152 - Gateway started."

""2021-12-16 09:18:16" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attributes_parse - 263 - Received data: {}"

jitendraspatil, 2021-12-16:
No Data Received

jitendraspatil, 2021-12-16:
[image]

zbeacon, 2021-12-16:
It looks like there is no response to WhoIs request to device.

jitendraspatil, 2021-12-17:
""2021-12-17 09:16:51" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 76 - Gateway starting..."

""2021-12-17 09:16:51" - |DEBUG| - [tb_updater.py] - tb_updater - check_for_new_version - 69 - Checking for new version"

""2021-12-17 09:16:51" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 81 - ThingsBoard IoT gateway version: 2.9"

""2021-12-17 09:16:51" - |DEBUG| - [tb_client.py] - tb_client - run - 124 - connecting to ThingsBoard"

""2021-12-17 09:16:51" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 171 - <paho.mqtt.client.Client object at 0x7f2795501d10>"

""2021-12-17 09:16:51" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 172 - Disconnecting from ThingsBoard"

""2021-12-17 09:16:51" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - subscribe_to_attribute - 301 - Subscribed to * with id 1"

""2021-12-17 09:16:51" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 178 - Subscribed to *|* with id 1 for device *"

""2021-12-17 09:16:51" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log service"

""2021-12-17 09:16:51" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log extension"

""2021-12-17 09:16:51" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log converter"

""2021-12-17 09:16:51" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log connector"

""2021-12-17 09:16:51" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log tb_connection"

""2021-12-17 09:16:51" - |DEBUG| - [memory_event_storage.py] - memory_event_storage - __init__ - 28 - Memory storage created with following configuration: 

Max size: 100000

 Read records per time: 100"

""2021-12-17 09:16:51" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_devices - 763 - Loaded devices:

 {'Temp Sensor': 'Modbus Default zeugy'}"

""2021-12-17 09:16:51" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 35 - Root path is: /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway"

""2021-12-17 09:16:51" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f27954f7f10>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-17 09:16:51" - |DEBUG| - [tb_client.py] - tb_client - _on_connect - 82 - TB client <paho.mqtt.client.Client object at 0x7f2795501d10> connected to ThingsBoard"

""2021-12-17 09:16:51" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 138 - connection SUCCESS"

""2021-12-17 09:16:52" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes - successfully completed."

""2021-12-17 09:16:52" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/attributes/response - successfully completed."

""2021-12-17 09:16:52" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 67 - Service subscription to topic v1/gateway/rpc - successfully completed."

""2021-12-17 09:16:52" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f27954c6950>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"

""2021-12-17 09:16:52" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BACnetConnector from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet."

""2021-12-17 09:16:52" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f27941046d0>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-17 09:16:52" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BACnetUplinkConverter from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet."

""2021-12-17 09:16:52" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f27941048d0>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"

""2021-12-17 09:16:52" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f2794104b90>, origin='/usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"

""2021-12-17 09:16:52" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='BACnetDownlinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f2794104950>, origin='/usr/local/lib/python3.7/site-packages/thing [... cut ...]
""2021-12-17 09:18:43" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log extension"
""2021-12-17 09:18:43" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log connector"
""2021-12-17 09:18:45" - |DEBUG| - [bacnet_connector.py] - bacnet_connector - scan_network - 195 - WhoIsRequest has been
 sent."
""2021-12-17 09:18:45" - |DEBUG| - [bacnet_connector.py] - bacnet_connector - run - 74 - WhoIsRequest has been sent."
""2021-12-17 09:18:45" - |DEBUG| - [bacnet_connector.py] - bacnet_connector - scan_network - 195 - WhoIsRequest has been
 sent."
#

jitendraspatil, 2021-12-17:
[image]

jitendraspatil, 2021-12-17:
data not Resived

jitendraspatil, 2021-12-17:
BACnet.json

{
  "general": {
    "objectName": "MRI_Area_DDC",
    "address": "0.0.0.0:4581",
    "objectIdentifier": 4581,
    "maxApduLengthAccepted": 1458,
    "segmentationSupported": "None:3",
    "vendorIdentifier": 37
  },
  "devices": [
    {
      "deviceName": "BACNet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.1.24:4581",
      "pollPeriod": 10000,
      "timeseries": [
        {
          "key": "Ctrl_1_MRI_RTRN_TEMP_1",
          "objectId": "analogValue:4",
          "propertyId": "presentValue"
        }
      ]     
    }
  ]
}

jitendraspatil, 2021-12-17:
help me plzzz ?

zbeacon, 2021-12-17:
Please collect packages from the gateway to the device using Wireshark or tcpdump and send the dump to us, or check - is there IAm response from device present?

samson0v, 2022-02-18:
Hi @jitendraspatil, is this bug still actual for you? Can we close the issue?

closed 2022-03-02 by jitendraspatil