thingsboard/thingsboard-gateway #657: [BUG] Gateway configuration documentation does not match reality

nznobody, 2021-12-21:
Just an FYI that your documentation does not match reality for the tb-gateway configuration. Fix would be great since I just wasted two hours on wondering why it was not working :(

Your main configuration documentation: https://thingsboard.io/docs/iot-gateway/configuration/ specifies that the folder is `/etc/thingsboard-gateway/config`. However in other places and in the container it is `/thingsboard_gateway/config`. Same goes for other folders. See Docker environment variables: `configs=/thingsboard_gateway/config`

E.g. https://github.com/thingsboard/thingsboard-gateway/blob/master/docker/Dockerfile#L19

samson0v, 2021-12-21:
Hi @nznobody, thanks for your interest in ThingsBoard IoT Gateway! We rewrote and updated our documentation and the last thing that must be done is to publish a new release version of Gateway.
You can see the working version of the documentation in this [PR](https://github.com/thingsboard/thingsboard.github.io/pull/767)  and offer some ideas and remarks.
Thanks!

nznobody, 2021-12-22:
Looks alright, just needs to be fixed up in the docker files (still points to old paths)

samson0v, 2021-12-22:
Thanks for your remarks, we will update the documentation.

samson0v, 2021-12-22:
@nznobody Thanks for your remarks, we will update the documentation.

JosDeL24, 2022-03-10:
To reiterate on this topic, would an admin of the Thingsboard documentation warn developers to not use Gateway as a name for their Gateway. @samson0v touched this topic on another thread and this solved errors with serial modbus and rpc calls

closed 2022-05-20 by imbeacon