thingsboard/thingsboard-gateway #658: [BUG] After a while, the packet We send from the mqtt connector does not reach the thingsboard server.

gohm-tech, 2021-12-21:
The thingsboard-gateway service stops sending data to the server after running for a certain period of time. When I restart the thingsboard-gateway service, it starts sending again.

That is the server code. The part marked in yellow is our gateway's acces token. We could not see any logs containing the acces token of our gateway when it did not send the data to the server. After restarting the thingsboard-gateway service, these logs started to appear.
[image]

That is our code log.
[image]

That is connector log.
[image]

That is the MQTT broker log we connect and subscribe.
[image]

**Server Environment**
own setup
docker deployment
OS: Linux(Ubuntu 20.04)

**Client Environment**
OS: Linux(Ubuntu)
Browser: Chrome, Firefox
Version Ubuntu Version 20.04

**Connector name**
MQTT Connector

**Configuration**
[mqtt.txt](https://github.com/thingsboard/thingsboard-gateway/files/7752196/mqtt.txt)
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/7752203/tb_gateway.txt)

**Versions:**
OS: Raspbian GNU/Linux 10 (buster)
Thingsboard IoT Gateway version [e.g. 2.7]
Python version 3.7.3

Best Regards,
**Gohm Team**

samson0v, 2021-12-21:
Hi @gohm-tech, please, set DEBUG level in the `config/logs.conf` file and provide more info.
Thanks for your interest in ThingsBoard IoT Gateway!

gohm-tech, 2021-12-21:
Hi @samson0v,
We already set DEBUG level the all of the log levels. We did not see any error in the log files. Everythings look good.Best Regards,
**Gohm Team**

samson0v, 2021-12-21:
@gohm-tech, after about how long do Gateway stop sending data?

gohm-tech, 2021-12-22:
Hi @samson0v,
We test it again. There are the logs.
[service.log](https://github.com/thingsboard/thingsboard-gateway/files/7762201/service.log)
[connector.log](https://github.com/thingsboard/thingsboard-gateway/files/7762202/connector.log)
[thingsboard_server.log](https://github.com/thingsboard/thingsboard-gateway/files/7762207/thingsboard_server.log)
[image]
[image]
[image]

**Note:** The server clock is 3 hours behind.

gohm-tech, 2021-12-23:
Hi @samson0v, 

Do you think it can be related to our thingsboard-gateway version?

Regards,
**Gohm Team**

gohm-tech, 2022-01-05:
Hi @samson0v,

We created two new gateways. One of them has the 2.7 version and the other has the latest version. (The reason we use version 2.7 is high CPU usage in the latest version). We have created a service that will send data over an MQTT connector per second. We think that this way we can understand whether the error is caused by the version.

Have you encountered such an error before?
Do you have any ideas as to why this error might occur?

Regards,
**Gohm Team**

samson0v, 2022-01-05:
Hi @gohm-tech, we released a new version of Gateway 3.0 (include a High CPU usage fix), so I think it will be better if you **completely** delete all versions of Gateways you have, install the new (3.0) one and send the logs from it.
Thanks for your interest in ThingsBoard IoT Gateway!

gohm-tech, 2022-01-06:
Hi @samson0v,

Thanks for your interest. Thanks for this information too. After the testing we realized something. thingsboard-gateway and this code are working together. We have created a service for working continuously to mqtt_client_firmware.py code. This code wants to subscribe same topics. So we were getting this error #643. After getting this error we change the ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE parameter. After doing this we thought the error was resolved. But in this test environment, we disabled the service of mqq_client_firmware.py code. After doing this, the error did not occur. In short, the error occurs as a result of multiple MQTT subscribe.

Regards,
**Gohm Team**

closed 2022-05-20 by imbeacon