thingsboard/thingsboard-gateway #659: [BUG][HELP] ThingsBoard Gateway cannot store/update JSON values

janvl1234, 2021-12-21:
The ThingsBoard gateway cannot store or update JSON values.

Seems to effect all connectors, I want to use it for the REST-Connector. 

In the database there is a field json_v. The JSON data can be created and updated via the GUI/Thingsboard.
There seems to be no possibility to update the JSON field data via de ThignsBoard gateway. 

Can you add the type json like there is a string, double, boolean? 
Or if it excists can you explain me how to use it properly. 

The rest,json connector looks like this: 
...
        {
            "type": "string",
            "key": "My new string",
            "value": "${string}"
        },
        {
            "type": "json",
            "key": "My new JSON",
            "value": "${json}"
        }
...

samson0v, 2021-12-21:
Hi @janvl1234, thanks for your interest in ThingsBoard IoT Gateway, we will discuss your offer and let you know in the coming days.

zzpandaman, 2023-04-03:
> Hi @janvl1234, thanks for your interest in ThingsBoard IoT Gateway, we will discuss your offer and let you know in the coming days.

I found built-in transport protocol implementation——HTTP Device API has the feature of  JSON value support. And it's really useful for simplifying configuration. But when i come to REST-Connector, this featrue is missing.

imbeacon, 2024-06-20:
Fixed on ThingsBoard.

closed 2024-06-20 by imbeacon