thingsboard/thingsboard-gateway #660: [BUG] BLE data converter not working with latest BLE connector remake from master branch

jayneev, 2021-12-22:
**Describe the bug**

**Connector name (If bug in the some connector):**
BLE Connector

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: Linux 4.9
 - Thingsboard IoT Gateway version 2.9 (latest master branch)
 - Python version - 3.8

samson0v, 2021-12-22:
Hi @jayneev, we will investigate the problem with data converting.
Thanks for your interest in ThingsBoard IoT Gateway!

jayneev, 2021-12-30:
Hi @samson0v, we are ready for testing the BLE connector for this fixes at our end. Is there any opportunity to test development fixes on this issue?
Thanking in advance.
Warm Regards,
@jayneev

samson0v, 2021-12-30:
@jayneev, we have investigated the problem with converting data, so for now I am working on fixing it. I will let you know when it is done.

closed 2021-12-30 by zbeacon