thingsboard/thingsboard-gateway #661: [HELP] How can we download the sensor firmware file into the gateway?

gohm-tech, 2021-12-23:
Hi,
[image]
We have a gateway and sensors. Sensors connected the sensor. We want to update sensor firmware via Narrow Band (RF). But first we need to download the sensor's firmware file into the gateway. There is a relation between sensor and gateway. When We changed the sensor firmware version using Thingsboard Dashboard.
[image]
mqtt_firmware_client.py code is already working in our gateway. The gateway's acces_token is given as an argument. When we changed the sensor firmware, we just see attribute_update_request log on service.log.

How can we download the sensor firmware file into the gateway?

**Client Environment**
OS: Linux(Ubuntu)
Browser: Chrome, Firefox
Version Ubuntu Version 20.04

**Connector name**
MQTT Connector

**Configuration**
[mqtt.txt](https://github.com/thingsboard/thingsboard-gateway/files/7769043/mqtt.txt)

**Versions:**
OS: Raspbian GNU/Linux 10 (buster)
Thingsboard IoT Gateway version [e.g. 2.7]
Python version 3.7.3

Best Regards,
**Gohm Team**

samson0v, 2021-12-24:
Hi @gohm-tech, you have to use [ThingsBoard Python SDK](https://github.com/thingsboard/thingsboard-python-client-sdk) and run it on your sensors for downloading firmware.
Thanks for your interest in ThingsBoard IoT Gateway.

gohm-tech, 2022-01-03:
Hi @samson0v,

We tried to implement this code (tb_device_mqtt.py and tb_gateway_mqtt.py) into thingsboard-gateway. But we got an error.
When will you push the new version of thingsboard-gateway which includes this option?

Best Regards,
**Gohm Team**

samson0v, 2022-01-04:
Hi @gohm-tech, as far as I know, in the near future we won't implement this in Gateway because we have this functionality in [Python SDK](https://github.com/thingsboard/thingsboard-python-client-sdk) and [Arduino SDK](https://github.com/thingsboard/thingsboard-arduino-sdk).

closed 2022-01-05 by gohm-tech