thingsboard/thingsboard-gateway #663: AttributeError: 'str' object has no attribute 'get' [ThingsBoard IoT gateway version: 2.9]

santobedi, 2021-12-27:
**Describe the issue**
The error arises after installing ThingsBoard IoT gateway version: 2.9 and updating the config files. I didn't have this error with earlier version of the gateway.   

**Connector name (If you need help with some connector/converter):**
Modbus and MQTT connectors

**Error traceback (If it was raised):**
```
iots@iot2056:/etc/thingsboard-gateway/config$ sudo python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("./tb_gateway.yaml")'
""2021-12-27 15:18:28" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 76 - Gateway starting..."
""2021-12-27 15:18:28" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 81 - ThingsBoard IoT gateway version: 2.9"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 91, in __init__
    self.tb_client = TBClient(self.__config["thingsboard"], self._config_dir)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 36, in __init__
    self.__tls = bool(credentials.get('tls', False) or credentials.get('caCert', False))
AttributeError: 'str' object has no attribute 'get'
```

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04.3 LTS (GNU/Linux 5.4.0-1047-raspi armv7l)
 - Thingsboard IoT Gateway version 2.9
 - Python version 3.8

santobedi, 2021-12-28:
[Update]
I reinstalled the gateway from [pip](https://thingsboard.io/docs/iot-gateway/install/pip-installation/). Now, I get **AttributeError: 'ModbusTcpClient' object has no attribute '_in_waiting'**. 

Is it a BUG? 

In addition, the telemetry value of MQTT is strange! 

Following is a part of the config file (timeseries): 
```

	{
					"type": "double",
					"key": "current_phase_1",
					"value": "${current_phase_1}"
				}
```
 
The telemetry received for this config is: 

 (Last update time) 2021-12-28 09:01:41	
(Key) current_phase_1	
(Value) **${current_phase_1}**

In the earlier gateway version, I used to get a proper value.

samson0v, 2021-12-28:
Hi @santobedi, there is no new release, so you have to use the newest Gateway version via the master branch and install it from the source.

santobedi, 2022-01-03:
Hi @samson0v,
I installed the newest Gateway from [here](https://thingsboard.io/docs/iot-gateway/install/source-installation/). Now I get ,

**AttributeError: 'ModbusTcpClient' object has no attribute '_in_waiting'**

Is it a bug or did I miss something else?

samson0v, 2022-01-04:
Hi @santobedi, I think you have the same problem as [this](https://github.com/thingsboard/thingsboard-gateway/issues/666).

samson0v, 2022-03-14:
Hi @santobedi, any updates? Can we close the issue?

santobedi, 2022-03-14:
Hello @samson0v,

The issue with Modbus connector was still there. I used an alternative option in my project (used only MQTT).

samson0v, 2022-08-05:
Hi @santobedi, update, please, your Gateway to the newest version and send the logs if the problem isn't gone.

closed 2022-10-13 by imbeacon