thingsboard/thingsboard-gateway #666: [BUG] after Update from 2.7 to 2.9 mqtt parser did not workcat

hansju-11, 2021-12-29:
**Describe the bug**
I did a update from 2.7 to 2.9 via deb pack. 
Now I can still receive mqtt message but I did not get the values into thingsboard. For me it looks like that the parser is not working. Instead of the values I got "pattern string" into the value field.

**Connector name (If bug in the some connector):**
mqtt broker.

my config:

{
  "broker": {
        "name":"ttn-mqtt",
    "host":"eu1.cloud.thethings.network",
    "port":1883,
    "clientId": "TheThingNetwork_gateway_1",
    "security": {
      "type": "basic",
      "username": "dragino-......",
      "password": ....."
    }
  },
 "mapping": [
    {
      "topicFilter": "v3/dragino-dev-kit-caffee-13092021@ttn/devices/+/up",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${$.end_device_ids.device_id}",
        "deviceTypeJsonExpression": "MultiSensor",
        "timeout": 120000,
        "attributes": [],
        "timeseries": [
         {
            "type": "double",
            "key": "temperature",
            "value": "${$.uplink_message.decoded_payload.temperature_1}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${$.uplink_message.decoded_payload.relative_humidity_2}"
          },
                  {
            "type": "number",
            "key": "digital_out",
            "value": "${$.uplink_message.decoded_payload.digital_out_3}"
          },
                  {
            "type": "number",
            "key": "digital_in",
            "value": "${$.uplink_message.decoded_payload.digital_in_4}"
          }

        ]
      }
    }
  ]
}

the payload I received via an mqtt browser:

{
  "end_device_ids": {
    "device_id": "eui-a.........",
    "application_ids": {
      "application_id": "dragino-........."
    },
    "dev_eui": "......",
    "join_eui": ".......",
    "dev_addr": ".........."
  },
  "correlation_ids": [
    "as:up:01FQW2FNNQ8D42NAQEZAAN0SK3",
    "gs:conn:01FQCAK5ZHHF7MG7C4Z5R2VC67",
    "gs:up:host:01FQCAK60AMAHATD489KDRA2Y3",
    "gs:uplink:01FQW2FNEMYMZAGSH4VNGGKC26",
    "ns:uplink:01FQW2FNENK88S9HAX0EFYWH9Z",
    "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01FQW2FNENEPAPTCK9D6XX34YY",
    "rpc:/ttn.lorawan.v3.NsAs/HandleUplink:01FQW2FNNPH08J86ZJY6W7ZTYH"
  ],
  "received_at": "2021-12-26T19:11:40.217180464Z",
  "uplink_message": {
    "session_key_id": ".............",
    "f_port": 1,
    "f_cnt": 42620,
    "frm_payload": "AWcAAAJokgMBAAQAAA==",
    "decoded_payload": {
      "digital_in_4": 0,
      "digital_out_3": 0,
      "relative_humidity_2": 73,
      "temperature_1": 0
    },
    "rx_metadata": [
      {
        "gateway_ids": {
          "gateway_id": "hs-el-leer",
          "eui": "A840411F7E38FFFF"
        },
        "time": "1980-01-06T00:00:19.728Z",
        "timestamp": 444857843,
        "rssi": -96,
        "channel_rssi": -96,
        "snr": 9.8,
        "uplink_token": "..........=",
        "channel_index": 3
      },
      {
        "gateway_ids": {
          "gateway_id": "eui-40d63cfffe1f8525",
          "eui": "40D63CFFFE1F8525"
        },
        "timestamp": 377034931,
        "rssi": -47,
        "channel_rssi": -47,
        "snr": 9,
        "location": {
          "latitude": 53.23535,
          "longitude": 7.46611,
          "altitude": 12,
          "source": "SOURCE_REGISTRY"
        },
        "uplink_token": "........",
        "channel_index": 3
      },
      {
        "gateway_ids": {
          "gateway_id": "agrnf007",
          "eui": "7076FF0056070CCB"
        },
        "time": "2021-12-26T19:11:39.955Z",
        "timestamp": 2754590787,
        "rssi": -86,
        "channel_rssi": -86,
        "snr": 10.8,
        "location": {
          "latitude": 53.236845308013045,
          "longitude": 7.467157956489246,
          "altitude": 50,
          "source": "SOURCE_REGISTRY"
        },
        "uplink_token": "C........="
      },
      {
        "gateway_ids": {
          "gateway_id": "leer-bremerstr9-indor",
          "eui": "58A0CBFFFE8022C9"
        },
        "time": "2021-12-26T19:11:39.938013076Z",
        "timestamp": 412456963,
        "rssi": -112,
        "channel_rssi": -112,
        "snr": -5.5,
        "location": {
          "latitude": 53.23408857416256,
          "longitude": 7.466425001621247,
          "source": "SOURCE_REGISTRY"
        },
        "uplink_token": ".........."
      }
    ],
    "settings": {
      "data_rate": {
        "lora": {
          "bandwidth": 125000,
          "spreading_factor": 7
        }
      },
      "coding_rate": "4/5",
      "frequency": "867100000",
      "timestamp": 444857843,
      "time": "1980-01-06T00:00:19.728Z"
    },
    "received_at": "2021-12-26T19:11:39.989269118Z",
    "consumed_airtime": "0.061696s",
    "network_ids": {
      "net_id": "000013",
      "tenant_id": "ttn",
      "cluster_id": "ttn-eu1"
    }
  }
}

** Debug log File **

""2021-12-26 12:09:43" - DEBUG - [mqtt_connector.py] - mqtt_connector - 352 - Successfully converted message from topic v3/dragino-..............1@ttn/devices/eui-aa1................../up"
""2021-12-26 12:10:50" - DEBUG - [mqtt_connector.py] - mqtt_connector - 718 - {'deviceName': '${end_device_ids.device_id}', 'deviceType': 'MultiSensor', 'attributes': [], 'telemetry': [{'temperature': '${uplink_message.decoded_payloa    d.temperature_1}'}, {'humidity': '${uplink_message.decoded_payload.relative_humidity_2}'}, {'digital_out': '${uplink_message.decoded_payload.digital_out_3}'}, {'digital_in': '${uplink_message.decoded_payload.digital_in_4}'}]}"
""2021-12-26 12:10:50" - DEBUG - [mqtt_connector.py] - mqtt_connector - 352 - Successfully converted message from topic v3/dragino-..............1@ttn/devices/eui-aa1................../up"
""2021-12-26 12:11:57" - DEBUG - [mqtt_connector.py] - mqtt_connector - 718 - {'deviceName': '${end_device_ids.device_id}', 'deviceType': 'MultiSensor', 'attributes': [], 'telemetry': [{'temperature': '${uplink_message.decoded_payloa    d.temperature_1}'}, {'humidity': '${uplink_message.decoded_payload.relative_humidity_2}'}, {'digital_out': '${uplink_message.decoded_payload.digital_out_3}'}, {'digital_in': '${uplink_message.decoded_payload.digital_in_4}'}]}"
""2021-12-26 12:11:57" - DEBUG - [mqtt_connector.py] - mqtt_connector - 352 - Successfully converted message from topic v3/dragino-..............1@ttn/devices/eui-aa1................../up"
""2021-12-26 12:13:03" - DEBUG - [mqtt_connector.py] - mqtt_connector - 718 - {'deviceName': '${end_device_ids.device_id}', 'deviceType': 'MultiSensor', 'attributes': [], 'telemetry': [{'temperature': '${uplink_message.decoded_payloa    d.temperature_1}'}, {'humidity': '${uplink_message.decoded_payload.relative_humidity_2}'}, {'digital_out': '${uplink_message.decoded_payload.digital_out_3}'}, {'digital_in': '${uplink_message.decoded_payload.digital_in_4}'}]}"
""2021-12-26 12:13:03" - DEBUG - [mqtt_connector.py] - mqtt_connector - 352 - Successfully converted message from topic v3/dragino-..............1@ttn/devices/eui-aa1................../up"
""2021-12-26 12:14:10" - DEBUG - [mqtt_connector.py] - mqtt_connector - 718 - {'deviceName': '${end_device_ids.device_id}', 'deviceType': 'MultiSensor', 'attributes': [], 'telemetry': [{'temperature': '${uplink_message.decoded_payloa    d.temperature_1}'}, {'humidity': '${uplink_message.decoded_payload.relative_humidity_2}'}, {'digital_out': '${uplink_message.decoded_payload.digital_out_3}'}, {'digital_in': '${uplink_message.decoded_payload.digital_in_4}'}]}"
""2021-12-26 12:14:10" - DEBUG - [mqtt_connector.py] - mqtt_connector - 352 - Successfully converted message from topic v3/dragino-..............1@ttn/devices/eui-aa1................../up"
""2021-12-26 12:15:17" - DEBUG - [mqtt_connector.py] - mqtt_connector - 718 - {'deviceName': '${end_device_ids.device_id}', 'deviceType': 'MultiSensor', 'attributes': [], 'telemetry': [{'temperature': '${uplink_message.decoded_payloa    d.temperature_1}'},

**Versions (please complete the following information):**
 - OS: [ Ubuntu 20.04]
 - Thingsboard IoT Gateway version [2.9]
 - Python version
        [Python 3.8.10 (default, Nov 26 2021, 20:14:08)   [GCC 9.3.0] on linux]]

samson0v, 2021-12-29:
HI @hansju-11, the bus was fixed, please, update your Gateway to the newest version via the master branch.
Thanks for your interest in ThingsBoard IoT Gateway.

hansju-11, 2021-12-29:
Hello samson0v,

I did an update https://thingsboard.io/docs/iot-gateway/install/source-installation/ 

but still I have the same problem.

samson0v, 2021-12-29:
@hansju-11, pleas, try to use `${uplink_message.decoded_payload.temperature_1}` insted of `${$.uplink_message.decoded_payload.temperature_1}`

hansju-11, 2021-12-29:
sorry but this is not working.

sysadmin@thingsboard:~/thingsboard-gateway$ sudo python3 ./thingsboard_gateway/tb_gateway.py
""2021-12-29 12:13:11" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 76 - Gateway starting..."
""2021-12-29 12:13:11" - |DEBUG| - [tb_updater.py] - tb_updater - check_for_new_version - 69 - Checking for new version"
""2021-12-29 12:13:11" - |DEBUG| - [connectionpool.py] - connectionpool - _new_conn - 957 - Starting new HTTPS connection (1): updates.thingsboard.io:443"
""2021-12-29 12:13:11" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 81 - ThingsBoard IoT gateway version: 2.9"
""2021-12-29 12:13:11" - |DEBUG| - [tb_client.py] - tb_client - run - 124 - connecting to ThingsBoard"
""2021-12-29 12:13:11" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 171 - <paho.mqtt.client.Client object at 0x7f0a135852e0>"
""2021-12-29 12:13:11" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - disconnect - 172 - Disconnecting from ThingsBoard"
""2021-12-29 12:13:11" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - subscribe_to_attribute - 301 - Subscribed to * with id 1"
""2021-12-29 12:13:11" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 178 - Subscribed to *|* with id 1 for device *"
""2021-12-29 12:13:11" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log service"
""2021-12-29 12:13:11" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log extension"
""2021-12-29 12:13:11" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log converter"
""2021-12-29 12:13:11" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log connector"
""2021-12-29 12:13:11" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 38 - Added remote handler to log tb_connection"
""2021-12-29 12:13:11" - |INFO| - [event_storage_reader.py] - event_storage_reader - read_state_file - 148 - FileStorage_reader -- Initializing from state file: [./data/data_1640779602739.txt:24]"
""2021-12-29 12:13:11" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_devices - 763 - Loaded devices:
 {'${$.end_device_ids.device_id}': 'MQTT Broker Dragion Dev Kit Connector'}"
""2021-12-29 12:13:11" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 35 - Root path is: /usr/local/lib/python3.8/dist-packages/thingsboard_gateway"
""2021-12-29 12:13:11" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 37 - Debian installation extensions folder exists."
""2021-12-29 12:13:11" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f0a13537850>, origin='/var/lib/thingsboard_gateway/extensions/mqtt/custom_mqtt_uplink_converter.py')"
""2021-12-29 12:13:11" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f0a135561f0>, origin='/usr/local/lib/python3.8/dist-packages/thingsboard_gateway/extensions/mqtt/custom_mqtt_uplink_converter.py')"
""2021-12-29 12:13:11" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f0a13556310>, origin='/usr/local/lib/python3.8/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_uplink_converter.py')"
""2021-12-29 12:13:11" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 57 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f0a13556790>, origin='/usr/local/lib/python3.8/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py')"
""2021-12-29 12:13:11" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import MqttConnector from /usr/local/lib/python3.8/dist-packages/thingsboard_gateway/connectors/mqtt."
""2021-12-29 12:13:11" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 154 - Mandatory key 'topicFilter' found in mapping handler: {"topicFilter": "v3/dragino-dev-kit-caffee-13092021@ttn/devices/+/up", "converter": {"type": "json", "deviceNameJsonExpression": "${end_device_ids.device_id}", "deviceTypeJsonExpression": "MultiSensor", "timeout": 120000, "attributes": [], "timeseries": [{"type": "double", "key": "temperature", "value": "${uplink_message.decoded_payload.temperature_1}"}, {"type": "double", "key": "humidity", "value": "${uplink_message.decoded_payload.relative_humidity_2}"}, {"type": "number", "key": "digital_out", "value": "${uplink_message.decoded_payload.digital_out_3}"}, {"type": "number", "key": "digital_in", "value": "${uplink_message.decoded_payload.digital_in_4}"}]}}"
""2021-12-29 12:13:11" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 154 - Mandatory key 'converter' found in mapping handler: {"topicFilter": "v3/dragino-dev-kit-caffee-13092021@ttn/devices/+/up", "converter": {"type": "json", "deviceNameJsonExpression": "${end_device_ids.device_id}", "deviceTypeJsonExpression": "MultiSensor", "timeout": 120000, "attributes": [], "timeseries": [{"type": "double", "key": "temperature", "value": "${uplink_message.decoded_payload.temperature_1}"}, {"type": "double", "key": "humidity", "value": "${uplink_message.decoded_payload.relative_humidity_2}"}, {"type": "number", "key": "digital_out", "value": "${uplink_message.decoded_pay [... cut ...] _2}'}, {'digital_out': '${uplink_message.decoded_payload.digital_out_3}'}, {'digital_in': '${uplink_message.decoded_payload.digital_in_4}'}]}"
""2021-12-29 12:13:37" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _save_converted_msg - 352 - Successfully converted message from topic v3/dragino-dev-kit-caffee-13092021@ttn/devices/eui-aa11bb22cc33dd44/up"
""2021-12-29 12:13:37" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __save_persistent_devices - 786 - Saved connected devices."

samson0v, 2021-12-29:
@hansju-11 send, please, the message that you are sending to the broker topic

samson0v, 2021-12-29:
@hansju-11 your device has to send the next message to the broker topic 
`mosquitto_pub -h 127.0.0.1 -p 1883 -t /sensor/data -m '{"serialNumber": "asd", "uplink_message": {"decoded_payload": {"temperature_1": 12, "relative_humidity_2": 123}}}'`

Result:
[image]

So, make sure that your message structure is equal to that you want Gateway convert.

hansju-11, 2021-12-29:
This is the message I got from ttn :

{
  "end_device_ids": {
    "device_id": "eui-aa11b*********44",
    "application_ids": {
      "application_id": "drag***************************"
    },
    "dev_eui": "aa11b*********44",
    "join_eui": "70B3*********579",
    "dev_addr": "260****D"
  },
  "correlation_ids": [
    "as:up:01FR33H8NJ931ZMZSCS8ZTG0R1",
    "gs:conn:01FR0P22R01AJ3Y2RP40YB4NCW",
    "gs:up:host:01FR0P22RBH51MVNFVMRBYPYWD",
    "gs:uplink:01FR33H8EYKCVMJ0Y234C47F09",
    "ns:uplink:01FR33H8EZ048X008XMA501F6C",
    "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01FR33H8EZZZ159QEAYTT9F8VV",
    "rpc:/ttn.lorawan.v3.NsAs/HandleUplink:01FR33H8NH3JJXXK7S73R6063V"
  ],
  "received_at": "2021-12-29T12:44:42.034971083Z",
  "uplink_message": {
    "session_key_id": "AX1O6JLKm/9G5O9d4yJroA==",
    "f_port": 1,
    "f_cnt": 46162,
    "frm_payload": "AWcAZAJonAMBAAQAAA==",
    "decoded_payload": {
      "digital_in_4": 0,
      "digital_out_3": 0,
      "relative_humidity_2": 78,
      "temperature_1": 10
    },
    "rx_metadata": [
      {
        "gateway_ids": {
          "gateway_id": "eui-40d63cfffe1f8525",
          "eui": "40D63CFFFE1F8525"
        },
        "timestamp": 135382939,
        "rssi": -47,
        "channel_rssi": -47,
        "snr": 9.8,
        "location": {
          "latitude": 53.23535,
          "longitude": 7.46611,
          "altitude": 12,
          "source": "SOURCE_REGISTRY"
        },
        "uplink_token": "CiIKIAoUZXVpLTQwZDYzY2ZmZmUxZjg1MjUSCEDWPP/+H4UlEJuPx0AaDAi5sLGOBhCpgPSHAyD46s6r+MoS",
        "channel_index": 4
      },
      {
        "gateway_ids": {
          "gateway_id": "hs-el-leer",
          "eui": "A840411F7E38FFFF"
        },
        "time": "1980-01-06T00:00:20.833Z",
        "timestamp": 203155707,
        "rssi": -98,
        "channel_rssi": -98,
        "snr": 9.5,
        "uplink_token": "ChgKFgoKaHMtZWwtbGVlchIIqEBBH344//8Q+9HvYBoMCLmwsY4GENOv2IMDIPiomOj0760BKgwIlPvUlgEQwKSajQM=",
        "channel_index": 4
      },
      {
        "gateway_ids": {
          "gateway_id": "agrnf007",
          "eui": "7076FF0056070CCB"
        },
        "time": "2021-12-29T12:44:41.782Z",
        "timestamp": 2513168643,
        "rssi": -90,
        "channel_rssi": -90,
        "snr": 10.8,
        "location": {
          "latitude": 53.236845308013045,
          "longitude": 7.467157956489246,
          "altitude": 50,
          "source": "SOURCE_REGISTRY"
        },
        "uplink_token": "ChYKFAoIYWdybmYwMDcSCHB2/wBWBwzLEIPSr64JGgwIubCxjgYQp6GokwMguOeYpJKzrgEqDAi5sLGOBhCAv/H0Ag==",
        "channel_index": 1
      },
      {
        "gateway_ids": {
          "gateway_id": "leer-bremerstr9-indor",
          "eui": "58A0CBFFFE8022C9"
        },
        "time": "2021-12-29T12:44:41.757863998Z",
        "timestamp": 1766443907,
        "rssi": -106,
        "channel_rssi": -106,
        "snr": 2.75,
        "location": {
          "latitude": 53.23408857416256,
          "longitude": 7.466425001621247,
          "source": "SOURCE_REGISTRY"
        },
        "uplink_token": "CiMKIQoVbGVlci1icmVtZXJzdHI5LWluZG9yEghYoMv//oAiyRCDl6fKBhoMCLmwsY4GEPb+y5UDILjvi8K0ngcqDAi5sLGOBhC+rLDpAg=="
      }
    ],
    "settings": {
      "data_rate": {
        "lora": {
          "bandwidth": 125000,
          "spreading_factor": 7
        }
      },
      "coding_rate": "4/5",
      "frequency": "867300000",
      "timestamp": 135382939
    },
    "received_at": "2021-12-29T12:44:41.823575071Z",
    "consumed_airtime": "0.061696s",
    "network_ids": {
      "net_id": "000013",
      "tenant_id": "ttn",
      "cluster_id": "ttn-eu1"
    }
  }
}

hansju-11, 2021-12-29:
my topic is 
v3/drag******************@ttn/devices/eui-aa1****************/up

hansju-11, 2021-12-29:
do you know a way to debug the incoming message in raw (the mqtt client will get from ttn). For now I use a external mqtt browser with the same credential to see the messages from ttn?

samson0v, 2021-12-29:
@hansju-11 I use your config file to test the Gateway
```json
{
      "topicFilter": "v3/dragino-dev-kit-caffee-13092021@ttn/devices/+/up",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${$.end_device_ids.device_id}",
        "deviceTypeJsonExpression": "default",
        "timeout": 60000,
        "attributes": [
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${$.uplink_message.decoded_payload.temperature_1}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${$.uplink_message.decoded_payload.relative_humidity_2}"
          },
          {
            "type": "number",
            "key": "digital_out",
            "value": "${$.uplink_message.decoded_payload.digital_out_3}"
          },
          {
            "type": "number",
            "key": "digital_in",
            "value": "${$.uplink_message.decoded_payload.digital_in_4}"
          }
        ]
      }
    }
```
also I use your message and I didn't see any problem with converting data (the last screenshot)
[image]

[image]

samson0v, 2021-12-29:
@hansju-11 it seems that you have a problem with your device or you use the old version of Gateway

hansju-11, 2021-12-29:
ok, I will uninstall the gateway and install it again, to see. maybe old code at some hidden places.
I will keep you updated. 
Thanks for your help

hansju-11, 2021-12-29:
IT IS WORKING !!!

but I had to reinstall the whole python, not only the thingsboard gateway. 
I dont know, but I'm happy now.  

Thx Vitalii for being with me ....

closed 2021-12-29 by hansju-11