thingsboard/thingsboard-gateway #670: [BUG] High CPU usage because of while loop in mqtt_firmware_client.py

gohm-tech, 2021-12-31:
Hi,
We are using the [mqtt_firmware_client.py](https://thingsboard.io/docs/user-guide/resources/firmware/mqtt_firmware_client.py) code for the [OTA updates](https://thingsboard.io/docs/user-guide/ota-updates/?remoteintegrationdockerinstall=mqtt). But when we used this code, we realized that our device was using a high CPU. Afterward, we investigated this and changed the code as follows.

[mqtt_firmware_client.txt](https://github.com/thingsboard/thingsboard-gateway/files/7795587/mqtt_firmware_client.txt)

[image]
[image]

**Client Environment**
OS: Linux(Ubuntu)
Browser: Chrome, Firefox
Version Ubuntu Version 20.04

**Connector name**
MQTT Connector

**Versions:**
OS: Raspbian GNU/Linux 10 (buster)
Thingsboard IoT Gateway version [e.g. 2.7]
Python version 3.7.3

Best Regards,
**Gohm Team**

samson0v, 2021-12-31:
Hi @gohm-tech, this bug doesn't relate to Gateway. Create an issue in [thingsboard-python-client-sdk
](https://github.com/thingsboard/thingsboard-python-client-sdk).

samson0v, 2021-12-31:
@gohm-tech This bug was fixed via the [commit](https://github.com/thingsboard/thingsboard-python-client-sdk/commit/f13e8d2f6e123c91238c701bccf16328e01879e2). The file from documentation will change to a new version soon.
Thanks for your interest in ThingsBoard.

closed 2021-12-31 by samson0v