thingsboard/thingsboard-gateway #672: [BUG]BLE connector remake

jayneev, 2021-12-31:
**Describe the bug**
Clear and concise description of what the bug is.

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
 #```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

jayneev, 2021-12-31:
@samson0v, it seems that the BLE converter still not working with latest fixes named "with fixes 'Fixed data converting for BLE Connector #669'". Could you please check the above logs and ble.json?

samson0v, 2021-12-31:
Hi @jayneev, it seems that your BLE device hasn't that characteristic, make sure that you are using the right one. To see all characteristics set `"showMap": true` on your device config section and use them (we were doing those things on our call).

`"valueExpression": "[:]"` - it is in what format data will be sent to the Gateway. When the data come from the device to Gateway, it is a byte array, so you must specify what specific byte to read. You can easily set [:] to read all bytes in the array if you don't know what exactly the data device send.

jayneev, 2021-12-31:
@samson0v,
Can we have discussion on Teams if you are available now?

samson0v, 2021-12-31:
@jayneev today no, but on Tuesday (04.01.2022) we can make a call.

jayneev, 2021-12-31:
@samson0v, 
1) It's standard BLE device Name characteristic UUID - "00002A00-0000-1000-8000-00805F9B34FB" which is default generic access attributes sent as default by BLE device connected. In all past releases it worked OK and returned CMx4 as 'Device Name', which  was published from Thingsboard Gateway to Thingsboard.
2) "valueExpression": "[:]" -  we set it to read all bytes in the array. Not sure what meant by "[0:2]cm [2:]A"?

samson0v, 2021-12-31:
@jayneev, use `showMap` to find out what characteristics exist and send the list here.

jayneev, 2021-12-31:
""2021-12-31 11:06:56" - |INFO| - [ble_connector.py] - ble_connector - __show_map - 65 - FOUND DEVICES"
""2021-12-31 11:06:56" - |INFO| - [ble_connector.py] - ble_connector - __show_map - 74 - E2:E1:C2:57:D7:09: CMx4"
""2021-12-31 11:06:56" - |INFO| - [ble_connector.py] - ble_connector - __show_map - 74 - 62:9F:15:CD:EC:17: 62-9F-15-CD-EC-17"
""2021-12-31 11:06:56" - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1640947770384.txt!"
""2021-12-31 11:06:56" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 154 - Gateway started."
""2021-12-31 11:06:59" - |INFO| - [device.py] - device - run_client - 164 - Connected to CMx4 device"
""2021-12-31 11:07:00" - |INFO| - [device.py] - device - __show_map - 204 - MAP FOR CMX4
| [Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
|       [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 11): Service Changed (indicate), Value: None
|               [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 13): Client Characteristic Configuration) | Value: b'\x02\x00'
| [Service] 1e821400-4a30-45b4-9e17-93f34e283081 (Handle: 14): Unknown
|       [Characteristic] 1e821401-4a30-45b4-9e17-93f34e283081 (Handle: 15): Unknown (notify), Value: None
|               [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 17): Client Characteristic Configuration) | Value: b'\x00\x00'
|       [Characteristic] 1e820101-4a30-45b4-9e17-93f34e283081 (Handle: 18): Unknown (read,write), Value: b'P\x00\x00'
|       [Characteristic] 1e820102-4a30-45b4-9e17-93f34e283081 (Handle: 20): Unknown (read,write), Value: b'\x00'
|       [Characteristic] 1e820103-4a30-45b4-9e17-93f34e283081 (Handle: 22): Unknown (read,write), Value: b'\x00'
|       [Characteristic] 1e820104-4a30-45b4-9e17-93f34e283081 (Handle: 24): Unknown (read,write), Value: b'\x01'
|       [Characteristic] 1e820105-4a30-45b4-9e17-93f34e283081 (Handle: 26): Unknown (read,write), Value: b'\x03\x00\x02\x00\x02'
|       [Characteristic] 1e820106-4a30-45b4-9e17-93f34e283081 (Handle: 28): Unknown (read,write), Value: b'\x00\x08\x05\x17\x01'"
Exception in thread CMx4:

samson0v, 2021-12-31:
@jayneev and send, pls, the config file that you are using

jayneev, 2021-12-31:
{
  "name": "BLE Connector",
  "passiveScanMode": false,
  "showMap": true,
  "devices": [
    {
      "name": "CMx4",
      "MACAddress": "E2:E1:C2:57:D7:09",
      "pollPeriod": 5000,
      "showMap": true,
      "timeout": 10000,
      "telemetry": [
        {
          "key": "Temp",
          "method": "notify",
          "characteristicUUID": "1E821401-4A30-45B4-9E17-93F34E283081",
          "valueExpression": "[0]"
        }
      ],
      "attributes": [
        {
          "key": "name",
          "method": "read",
          "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
          "valueExpression": "[0:2]cm [2:]A"                           
        },                                                             
        {                                                              
          "key": "values",                                             
          "method": "read",                                            
          "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB",
          "valueExpression": "[:]"                         
        }                                                              
      ],                                                               
      "attributeUpdates": [                                            
        {                                                              
          "attributeOnThingsBoard": "sharedName",                      
          "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB" 
        }                                                              
      ],                                                               
      "serverSideRpc": [                                               
      ]                                                                
    }                                                                  
  ]                                                                    
}

samson0v, 2021-12-31:
@jayneev I don't understand why you are using a characteristic UUID that doesn't exist in the device
[image]
The first telemetry section is OK because you use the right characteristic, the next attributes section is not OK, so Gateway raises the exception.

To fix it, easily delete all obj in the attributes section (but not the attributes section, it can be empty but not without it).

jayneev, 2021-12-31:
Deleting attribute section, will update on it soon online

jayneev, 2021-12-31:
@samson0v, below are the logs and ble configuration. Hope this helps.

<<<<<<<<<<<<<<<<<<<<<< logs START <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
thingsboard_gateway# ""2021-12-31 13:30:20" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 77 - Gateway starting..."
""2021-12-31 13:30:20" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 82 - ThingsBoard IoT gateway version: 2.9"
""2021-12-31 13:30:20" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 178 - Subscribed to *|* with id 1 for device *"
""2021-12-31 13:30:20" - |INFO| - [event_storage_reader.py] - event_storage_reader - read_state_file - 148 - FileStorage_reader -- Initializing from state file: [./data/data_1640948803974.txt:0]"
""2021-12-31 13:30:20" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 138 - connection SUCCESS"
""2021-12-31 13:30:22" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BLEConnector from /usr/lib/python3.8/site-packages/thingsboard_gateway/connectors/ble."
""2021-12-31 13:30:33" - |INFO| - [ble_connector.py] - ble_connector - __show_map - 65 - FOUND DEVICES"
""2021-12-31 13:30:33" - |INFO| - [ble_connector.py] - ble_connector - __show_map - 74 - 48:00:E5:10:A3:30: 48-00-E5-10-A3-30"
""2021-12-31 13:30:33" - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1640948803974.txt!"
""2021-12-31 13:30:33" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 154 - Gateway started."
""2021-12-31 13:30:44" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:30:55" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:31:06" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:31:17" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:31:28" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:31:39" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:31:50" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:32:01" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:32:12" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
""2021-12-31 13:32:23" - |ERROR| - [device.py] - device - connect_to_device - 155 - Device with address E2:E1:C2:57:D7:09 was not found."
<<<<<<<<<<<<<<<<<<<<<<<<<<< logs END <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

<<<<<<<<<<<<<<<<<<<<<<<<<<ble.json START <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
{
  "name": "BLE Connector",
  "passiveScanMode": false,
  "showMap": true,
  "devices": [
    {
      "name": "CMx4",
      "MACAddress": "E2:E1:C2:57:D7:09",
      "pollPeriod": 5000,
      "showMap": true,
      "timeout": 10000,
      "telemetry": [
        {
          "key": "Temp",
          "method": "notify",
          "characteristicUUID": "1E821401-4A30-45B4-9E17-93F34E283081",
          "valueExpression": "[0]"
        }
      ],
      "attributes": [                                
      ],                                                               
      "attributeUpdates": [                                            
        {                                                              
          "attributeOnThingsBoard": "sharedName",                      
          "characteristicUUID": "00002A00-0000-1000-8000-00805F9B34FB" 
        }                                                              
      ],                                                               
      "serverSideRpc": [                                               
      ]                                                                
    }                                                                  
  ]                                                                    
} 
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<ble.json END<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

samson0v, 2021-12-31:
@jayneev BLE device not found, make sure that your device is on and BLE is on too.

jayneev, 2021-12-31:
checking it.

samson0v, 2021-12-31:
@jayneev don't send logs that make no sense, I think there is no problem firstly check device connection, check logs and check if these logs relate to the problem of this issue.

jayneev, 2021-12-31:
Do you understand that this UUID devices sent which leads to it's identification as CMx4? Device was reconnected and tested. Below are the logs. Note that no CMx4 device updates on the THINGSBOARD dashboard side.

<<<<<<<<<<<<<<<<<<<<<< logs START <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
thingsboard_gateway# ""2021-12-31 13:48:28" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 77 - Gateway starting..."
""2021-12-31 13:48:28" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 82 - ThingsBoard IoT gateway version: 2.9"
""2021-12-31 13:48:28" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 178 - Subscribed to *|* with id 1 for device *"
""2021-12-31 13:48:28" - |INFO| - [event_storage_reader.py] - event_storage_reader - read_state_file - 148 - FileStorage_reader -- Initializing from state file: [./data/data_1640957420864.txt:0]"
""2021-12-31 13:48:28" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 138 - connection SUCCESS"
""2021-12-31 13:48:30" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import BLEConnector from /usr/lib/python3.8/site-packages/thingsboard_gateway/connectors/ble."
""2021-12-31 13:48:40" - |INFO| - [ble_connector.py] - ble_connector - __show_map - 65 - FOUND DEVICES"
""2021-12-31 13:48:40" - |INFO| - [ble_connector.py] - ble_connector - __show_map - 74 - E2:E1:C2:57:D7:09: CMx4"
""2021-12-31 13:48:40" - |INFO| - [ble_connector.py] - ble_connector - __show_map - 74 - 7C:90:CA:01:46:92: 7C-90-CA-01-46-92"
""2021-12-31 13:48:40" - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1640957420864.txt!"
""2021-12-31 13:48:40" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 154 - Gateway started."
""2021-12-31 13:48:43" - |INFO| - [device.py] - device - run_client - 164 - Connected to CMx4 device"
""2021-12-31 13:48:44" - |INFO| - [device.py] - device - __show_map - 204 - MAP FOR CMX4
| [Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
| [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 11): Service Changed (indicate), Value: None
| [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 13): Client Characteristic Configuration) | Value: b'\x02\x00'
| [Service] 1e821400-4a30-45b4-9e17-93f34e283081 (Handle: 14): Unknown
| [Characteristic] 1e821401-4a30-45b4-9e17-93f34e283081 (Handle: 15): Unknown (notify), Value: None
| [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 17): Client Characteristic Configuration) | Value: b'\x00\x00'
| [Characteristic] 1e820101-4a30-45b4-9e17-93f34e283081 (Handle: 18): Unknown (read,write), Value: b'P\x00\x00'
| [Characteristic] 1e820102-4a30-45b4-9e17-93f34e283081 (Handle: 20): Unknown (read,write), Value: b'\x00'
| [Characteristic] 1e820103-4a30-45b4-9e17-93f34e283081 (Handle: 22): Unknown (read,write), Value: b'\x00'
| [Characteristic] 1e820104-4a30-45b4-9e17-93f34e283081 (Handle: 24): Unknown (read,write), Value: b'\x01'
| [Characteristic] 1e820105-4a30-45b4-9e17-93f34e283081 (Handle: 26): Unknown (read,write), Value: b'\x03\x00\x02\x00\x02'
| [Characteristic] 1e820106-4a30-45b4-9e17-93f34e283081 (Handle: 28): Unknown (read,write), Value: b'\x00\x08\x05\x17\x01'"
<<<<<<<<<<<<<<<<<<<<<< logs END <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

samson0v, 2021-12-31:
@jayneev I can't see any problems, you are using notify method, so make sure that data on your device changing.

jayneev, 2021-12-31:
@samson0v, device sending updated changed data notification periodically OK, was working and tested OK with TB gateway previous version release since 2.5.2 and onwards.

jayneev, 2022-01-02:
@samson0v, device sent updated data notification tested OK with TB gateway previous release versions 2.5.2 and onwards till release v2.9 (with bluepy) . So, why with same devices sending data to TB gateway (latest version 2.9 from master branch using bleak), updated data not received?

samson0v, 2022-01-05:
@jayneev we released a new version of Gateway and updated documentation where describe all config parameters.
You are welcome https://thingsboard.io/docs/iot-gateway/config/ble/

samson0v, 2022-01-10:
Opened another issue

closed 2022-01-10 by samson0v