thingsboard/thingsboard-gateway #678: [HELP] Create .deb package failed

ndrs-cwt, 2022-01-05:
Hello all
I try to create .deb package with command ```sudo ./generate_deb_package.sh clean``` but I got this error and build process stopped
```
======================================================================
ERROR: thingsboard_gateway (unittest.loader._FailedTest)
----------------------------------------------------------------------
ImportError: Failed to import test module: thingsboard_gateway
Traceback (most recent call last):
  File "/usr/lib/python3.8/unittest/loader.py", line 470, in _find_test_path
    package = self._get_module_from_name(name)
  File "/usr/lib/python3.8/unittest/loader.py", line 377, in _get_module_from_name
    __import__(name)
  File "/home/ndrs-cwt/ndrs_wks/vcs/thingsboard-gateway/deb_dist/thingsboard-gateway-3.0/.pybuild/cpython3_3.8_thingsboard-gateway/build/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/home/ndrs-cwt/ndrs_wks/vcs/thingsboard-gateway/deb_dist/thingsboard-gateway-3.0/.pybuild/cpython3_3.8_thingsboard-gateway/build/thingsboard_gateway/gateway/tb_gateway_service.py", line 33, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
  File "/home/ndrs-cwt/ndrs_wks/vcs/thingsboard-gateway/deb_dist/thingsboard-gateway-3.0/.pybuild/cpython3_3.8_thingsboard-gateway/build/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
    from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
  File "/home/ndrs-cwt/ndrs_wks/vcs/thingsboard-gateway/deb_dist/thingsboard-gateway-3.0/.pybuild/cpython3_3.8_thingsboard-gateway/build/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 25, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_downlink_converter import GrpcDownlinkConverter
  File "/home/ndrs-cwt/ndrs_wks/vcs/thingsboard-gateway/deb_dist/thingsboard-gateway-3.0/.pybuild/cpython3_3.8_thingsboard-gateway/build/thingsboard_gateway/gateway/grpc_service/grpc_downlink_converter.py", line 21, in <module>
    from thingsboard_gateway.gateway.proto.messages_pb2 import *
  File "/home/ndrs-cwt/ndrs_wks/vcs/thingsboard-gateway/deb_dist/thingsboard-gateway-3.0/.pybuild/cpython3_3.8_thingsboard-gateway/build/thingsboard_gateway/gateway/proto/messages_pb2.py", line 22, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
```

full command log here
[conmand_log.txt](https://github.com/thingsboard/thingsboard-gateway/files/7814041/conmand_log.txt)

Code based from https://github.com/thingsboard/thingsboard-gateway/commit/246c6c9595c7d442f2399a373702053f7e83d676

```
sudo pip list | grep grpcio
grpcio                 1.43.0  
```
Ubuntu 20.04
 Python 3.8

Do I have to install a specific grpc version or maybe something in my environment is wrong?

Best regard
Chaiwat

zbeacon, 2022-01-05:
Hi @ndrs-cwt ,

Please try to install the gateway first, with `sudo pip3 setup.py install` and then run `sudo bash generate_deb_package.sh clean`.
In order to generate the deb package - you have to install the gateway, because, unfortunately we cannot use relative import in grpc files (It is a problem with generation for python).

Please try to follow to suggested steps and let us know about the results.

ndrs-cwt, 2022-01-05:
@zbeacon 
I install the gateway with ```sudo python setup.py install```
[install_tb_gw_log.txt](https://github.com/thingsboard/thingsboard-gateway/files/7815128/install_tb_gw_log.txt)

but after run ```sudo bash generate_deb_package.sh clean``` I still got the same ```AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'``` error.

[generate_deb_package.sh_clean.txt](https://github.com/thingsboard/thingsboard-gateway/files/7815214/generate_deb_package.sh_clean.txt)

Try to run thingboard-gateway after install, got the same error too.
```
ndrs-cwt@ndrscwt-ubuntu:~$ sudo thingsboard-gateway 
[sudo] password for ndrs-cwt: 
Traceback (most recent call last):
  File "/usr/local/bin/thingsboard-gateway", line 11, in <module>
    load_entry_point('thingsboard-gateway==3.0', 'console_scripts', 'thingsboard-gateway')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 33, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
    from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 25, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_downlink_converter import GrpcDownlinkConverter
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/grpc_service/grpc_downlink_converter.py", line 21, in <module>
    from thingsboard_gateway.gateway.proto.messages_pb2 import *
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/proto/messages_pb2.py", line 22, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
```
```
ndrs-cwt@ndrscwt-ubuntu:~$ sudo python
Python 3.8.10 (default, Nov 26 2021, 20:14:08) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import grpc
>>> import thingsboard_gateway
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 33, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
    from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 25, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_downlink_converter import GrpcDownlinkConverter
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/grpc_service/grpc_downlink_converter.py", line 21, in <module>
    from thingsboard_gateway.gateway.proto.messages_pb2 import *
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/gateway/proto/messages_pb2.py", line 22, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
>>> 

```

zbeacon, 2022-01-05:
Please try to run the following command and then try to generate again:
`sudo pip install --upgrade protobuf`

artofdestroy, 2022-01-06:
@zbeacon 
confirmed ```sudo pip install --upgrade protobuf``` Fixed my problem.

ndrs-cwt, 2022-01-06:
@zbeacon 
Thank you ```sudo pip install --upgrade protobuf``` Fixed my problem.

closed 2022-01-06 by ndrs-cwt