thingsboard/thingsboard-gateway #679: [HELP] Modbus

l33tkrew, 2022-01-05:
**Describe the issue**
I want to test MODBUS, BLE, CAN Connectors. However, I do not know how to send messages? I read all documentation about tb-gateway. I think testing the connectors is missing in the documentation.

For example, I use the following format to text MQTT and REST(POST). As you can see the message formats are different. For example, I could not a way to test MODBUS connector because I do not know the command/message format.

```
mosquitto_pub -h "192.168.1.60" -p "1883" -t "/sensor/data" -m '{"serialNumber": "SN-001", "sensorType": "Thermometer", "sensorModel": "T1000", "temp": 42, "hum": 58}'
curl -v -X POST -d "{\"name\":\"DEV-01\",\"sensorModel\":\"snn-01\",\"temp\":33,\"hum\":23}" http://192.168.1.X:5000/device1 -H "Content-Type:application/json"
```

**Configuration** (Attach your configuration file)
I use the config files;
https://github.com/thingsboard/thingsboard-gateway/tree/master/thingsboard_gateway/config

**Connector name (If you need help with some connector/converter):**
[e.g. MODBUS, BLE, CAN Connectors]

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [laters]
 - Python version[e.g. 3.8]

samson0v, 2022-01-05:
Hi @l33tkrew, mosquitto_pub is used only if you are using an MQTT connector, if you want to use Modbus Connector, you have to use device compatible or create slave using, for example, [pymodbus](https://pymodbus.readthedocs.io/en/latest/source/example/modbus_payload_server.html).
So, for example, if you want to use BLE Connector, there are two options:
1. Use real BLE Device
2. Use a simulator on a smartphone (BLE Peripheral Simulator), so you can use the MAC address of your smartphone and GATT objects of your simulator and voila, Gateway will read your configured data options.

l33tkrew, 2022-01-06:
So, if I want to use Modbus, should start modbus server manually. REST connector automatically creates a server and listen on 5000 port. But, It seems I should create modbus server manually with python.

Is [pymodbus](https://pymodbus.readthedocs.io/en/latest/source/example/modbus_payload_server.html) server script that should be run on tb-gateway server? Without this script, I got a connection problem to port 5021

However, When the connection is established, I got the following error(s). I haven't searched about the Modbus connector. It may be my fault.
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-2.9-py3.8.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 77, in convert
    decoded_data = self.__decode_from_registers(decoder, configuration)
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-2.9-py3.8.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 128, in __decode_from_registers
    decoded = decoder_functions[lower_type]()
  File "/root/.local/lib/python3.8/site-packages/pymodbus/payload.py", line 420, in decode_64bit_uint
    handle = self._unpack_words(fstring, handle)
  File "/root/.local/lib/python3.8/site-packages/pymodbus/payload.py", line 361, in _unpack_words
    handle = unpack(up, handle)
struct.error: unpack requires a buffer of 8 bytes
""2022-01-06 19:52:07" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 91 - 'utf-8' codec can't decode byte 0xee in position 7: unexpected end of data"
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-2.9-py3.8.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 77, in convert
    decoded_data = self.__decode_from_registers(decoder, configuration)
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-2.9-py3.8.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 151, in __decode_from_registers
    result_data = decoded.decode('UTF-8')

```

closed 2022-01-06 by l33tkrew