thingsboard/thingsboard-gateway #684: [HELP] OPC-UA Connection refused

l33tkrew, 2022-01-06:
**Describe the issue**
When I activate the OPC-UA connector, I got a connection refused problem. Also, I cannot see the 4840 port on listening ports.

I tried localhost, 0.0.0.0, 192.168.1.115 IP address. But, none of them worked.

Is it a problem to use both REST and OPC-UA connectors at the same time?

**Configuration** (Attach your configuration file)
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "0.0.0.0:4840/server/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "temperature °C",
            "path": "${ns=2;i=5}"
          }
        ],
        "timeseries": [
          {
            "key": "humidity",
            "path": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Humidity}"
          },
          {
            "key": "batteryLevel",
            "path": "${Battery\\.batteryLevel}"
          }
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [
              2,
              4
            ]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}
```

**Connector name (If you need help with some connector/converter):**
[e.g. OPC-UA Connector]

**Error traceback (If it was raised):**
```
""2022-01-06 21:57:18" - |ERROR| - [opcua_connector.py] - opcua_connector - run - 114 - Connection refused on connection to OPC-UA server with url 0.0.0.0:4840/server/"
""2022-01-06 21:57:28" - |ERROR| - [opcua_connector.py] - opcua_connector - run - 114 - Connection refused on connection to OPC-UA server with url 0.0.0.0:4840/server/"
""2022-01-06 21:57:38" - |ERROR| - [opcua_connector.py] - opcua_connector - run - 114 - Connection refused on connection to OPC-UA server with url 0.0.0.0:4840/server/"
""2022-01-06 21:57:48" - |ERROR| - [opcua_connector.py] - opcua_connector - run - 114 - Connection refused on connection to OPC-UA server with url 0.0.0.0:4840/server/"
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0]
 - Python version[e.g. 3.8]

ysimonx, 2022-01-07:
Hello

This connector is not a OPCUA Server
it is a OPCUA Client

You need a OPCUA Server (for instance a nodered server)

l33tkrew, 2022-01-08:
When I run this server from another virtual machine in my local network. The gateway can connect to it but I cannot get the data to be sent to tb through mqtt protocol.

Can you help me how can i send the correct data to opc protocol?

```
from opcua import Server
from random import randint
import datetime
import time

server = Server()

url = "opc.tcp://0.0.0.0:4840/server/"
server.set_endpoint(url)

name = "OPC_SIMULATION_SERVER"
addspace = server.register_namespace(name)

node = server.get_objects_node()

Param = node.add_object(addspace, "Parameters")

Temp = Param.add_variable(addspace, "Temperature", 0)
Press = Param.add_variable(addspace, "Pressure", 0)
Time = Param.add_variable(addspace, "Time", 0)

Temp.set_writable()
Press.set_writable()
Time.set_writable()

server.start()
print("Server started at {}".format(url))

while True:
    #Temperature = randint(10,50)
    Temperature = 11
    Pressure = randint(200, 999)
    TIME = datetime.datetime.now()

    print(Temperature, Pressure, TIME)

    Temp.set_value(Temperature)
    Press.set_value(Pressure)

    Time.set_value(TIME)

    time.sleep(0.1)
```

OPC confs
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "192.168.1.55:4840/server/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "temperature °C",
            "path": "${ns=2;i=5}"
          }
        ],
        "timeseries": [
          {
            "key": "humidity",
            "path": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Humidity}"
          },
          {
            "key": "batteryLevel",
            "path": "${Battery\\.batteryLevel}"
          }
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [
              2,
              4
            ]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}
```

ysimonx, 2022-01-08:
can you provide tb_gateway.yaml please ?

ysimonx, 2022-01-08:
can you provide your tb_connection.log too ?

l33tkrew, 2022-01-11:
```
thingsboard:
  host: MY_IP
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: MY_TOKEN
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
connectors:
  - name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#
  -
    name: OPC-UA Connector
    type: opcua
    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
  -
    name: REST Connector
    type: rest
    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
```

I'll send the log file tomorrow. But it does not include any error.

samson0v, 2022-02-03:
Hi @l33tkrew, please try to use the version from the master branch and let us know about the results.

closed 2022-03-03 by l33tkrew