thingsboard/thingsboard-gateway #685: [BUG] Latest docker image fails on start (arm64)

greg-mcnamara, 2022-01-07:
**Describe the bug**
I'm trying to run the latest (3.0) docker image for thingsboard-gateway and it fails to start up with the error message below in the logs. If I downgrade to the 2.9 version it starts up ok.

tb-gateway_1  |   File "//./thingsboard_gateway/tb_gateway.py", line 17, in <module>
tb-gateway_1  |     from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
tb-gateway_1  | ModuleNotFoundError: No module named 'thingsboard_gateway'

**Connector name (If bug in the some connector):**

**Error traceback (If available):**

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04 (arm64 on rpi4)
 - Thingsboard IoT Gateway version 3.0
 - Python version 3.8.10

basubhaskar, 2022-01-08:
This is correct. I have added 'RUN python /setup.py install' in Dockerfile and did build it locally. This resolved the issue. But I am facing a new issue. Its not connecting to opcua endpoint with the error

File "/usr/local/lib/python3.8/site-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 509, in OpcUaConnector
tb-gateway    |     def get_node_path(self, node: Node):
tb-gateway    | NameError: name 'Node' is not defined

The same works without Docker 

Import OpcUaConnector from /usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0-py3.8.egg/thingsboard_gateway/connectors/opcua."
""2022-01-08 10:11:36" - |INFO| - [opcua_connector.py] - opcua_connector - open - 100 - Starting OPC-UA Connector"
""2022-01-08 10:11:36" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - Gateway started."
""2022-01-08 10:11:36" - |INFO| - [opcua_connector.py] - opcua_connector - run - 125 - OPC-UA connector OPC-UA Connector connected to server <OPCUA-server-IP>:<OPCUA-Server-Port>/OPCUA/SimulationServer"

greg-mcnamara, 2022-01-08:
Thanks @basubhaskar I noticed /setup.py had been removed from Dockerfile after 2.9 so I also tried adding it back in but the build failed (were you building on ARM or AMD?). I ended up checking out the 2.9 release code, updating the code specifically for the aiohttp fix in 3.0 (the main reason I wanted 3.0) and rebuilding. So I have a working custom docker image but would like to see a working 3.0 release for arm64.

basubhaskar, 2022-01-09:
Its amd64 for me. I finally fixed this issue by adding following four installables in setup.py       
        'pytz',
        'python-dateutil',
        'lxml',
        'opcua'
and building the docker with setup.py install included as I mentioned in last post.

zbeacon, 2022-01-10:
Hi @basubhaskar, @greg-mcnamara thanks for your participating. Could you try to use the latest image now?

greg-mcnamara, 2022-01-10:
Hi @basubhaskar, the latest image works ok for me now with no modifications. Thanks!

greg-mcnamara, 2022-01-11:
Closing this issue. Resolved by updated docker image for v3.0.

closed 2022-01-11 by greg-mcnamara