thingsboard/thingsboard-gateway #687: [BUG] OPCUA Connector very long MQTT connection to Thingsboard

ysimonx, 2022-01-08:
**Describe the bug**
Connection to MQTT Thingsboard is very long with TB Gateway
i have lots of "connecting to ThingsBoard" before any connection
However, when i try to telnet thingsboard.cloud on port 1883, the connection is made very quickly

**OPC-UA Connector:**

**Error traceback :**
```
''2022-01-08 13:54:00' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:01' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:02' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:03' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:04' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:05' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:06' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:07' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:09' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:10' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
''2022-01-08 13:54:11' - DEBUG - tb_client - 124 - connecting to ThingsBoard'
```

**Versions :**
 - OS: Mac OS
 - Thingsboard IoT Gateway version 3.0]
 - Python version 3.9

ysimonx, 2022-01-08:
on
thingsboard_gateway/tb_client/tb_device_mqtt.py

if i add a "sleep" , the connection is made without waiting

```
def connect(self, callback=None, min_reconnect_delay=1, timeout=120, tls=False, ca_certs=None, cert_file=None, key_file=None, keepalive=120):
        if tls:
            try:
                self._client.tls_set(ca_certs=ca_certs,
                                     certfile=cert_file,
                                     keyfile=key_file,
                                     cert_reqs=ssl.CERT_REQUIRED,
                                     tls_version=ssl.PROTOCOL_TLSv1_2,
                                     ciphers=None)
                self._client.tls_insecure_set(False)
            except ValueError:
                pass
        self._client.connect(self.__host, self.__port, keepalive=keepalive)
        **sleep(5)**
        self.reconnect_delay_set(min_reconnect_delay, timeout)
        self._client.loop_start()
        self.__connect_callback = callback
```
        
with less than 3 seconds, the connection is not available before a long time

I have a very good connection to Internet from the TB Gateway

ysimonx, 2022-01-08:
FYI : I have the impression that as long as the connector founds the OPCUA attributes nodes, telemetry nodes , the MQTT connection is difficult to establish

samson0v, 2022-01-10:
Hi @ysimonx, we investigating the problem. I will let you know about the results.

ysimonx, 2022-01-14:
DDOS on thingsboard.cloud,
[image]

closed 2022-01-14 by ysimonx