thingsboard/thingsboard-gateway #688: [BUG] BLE connector not connecting with multiple BLE devices on startup with latest Bleak Client introduced- TB gateway release ver3.0

jayneev, 2022-01-10:
**Describe the bug**
BLE Connector clients - TB Gateway release ver 3.0,  not able to connect to multiple devices on startup. Bleak multiple clients fail to connect concurrently to all devices. Only max 1-2 devices connected at some instances and that too not sending telemetry data to Thingsboard Application.
Query:
Is this Bleak client able to connect to more multiple BLE devices (e.g 3 devices ) in parallel and data communications OK? Was this tested and verified before > 1 BLE device connections?

**Connector name (If bug in the some connector):**
BLE Connector

**Error traceback (If available):**
```ble.json''''
{
  "name": "BLE Connector",
  "passiveScanMode": false,
  "showMap": false,
  "scanner": {
    "timeout": 10000,
    "deviceName": "Device name"
  },
  "devices": [
    {
      "name": "CMx10",
      "MACAddress": "F7:A2:AD:F4:4E:D9",
      "pollPeriod": 5000,
      "showMap": true,
      "addrType": "public",
      "timeout": 10000,
      "telemetry": [
        {
          "key": "Temp",
          "method": "notify",
          "characteristicUUID": "1e821401-4a30-45b4-9e17-93f34e283081",
          "valueExpression": "[0:3]"
        }
       ],
      "attributes": [                                                  
        {                                                              
          "key": "name",                                               
          "method": "read",                                            
          "characteristicUUID": "00002a00-0000-1000-8000-00805f9b34fb",
          "valueExpression": "[:]"                                     
        }                                                              
      ],                                                               
      "attributeUpdates": [                                            
        {                                                              
          "attributeOnThingsBoard": "sharedName",                      
          "characteristicUUID": "00002a00-0000-1000-8000-00805f9b34fb" 
        }                                                              
      ],                                                               
      "serverSideRpc": [                                               
      ]                                                                
    },
    {                                                                  
      "name": "CMx32",                                                 
      "MACAddress": "F9:C2:A9:B4:12:D9",                               
      "pollPeriod": 5000,                                              
      "showMap": true,                                                 
      "addrType": "public",                                            
      "timeout": 10000,                                                
      "telemetry": [                                                   
        {                                                              
          "key": "Temp",                                               
          "method": "notify",                                          
          "characteristicUUID": "1e821401-4a30-45b4-9e17-93f34e283081",
          "valueExpression": "[0:3]"                                   
        }
      ],                                                               
      "attributes": [                                                  
        {                                                              
          "key": "name",                                               
          "method": "read",                                            
          "characteristicUUID": "00002a00-0000-1000-8000-00805f9b34fb",
          "valueExpression": "[:]"                                     
        }                                                              
      ],                                                               
      "attributeUpdates": [                                            
        {                                                              
          "attributeOnThingsBoard": "sharedName",                      
          "characteristicUUID": "00002a00-0000-1000-8000-00805f9b34fb" 
        }                                                              
      ],                                                               
      "serverSideRpc": [                                               
      ]                                                                
    },                                                                 
    {                                                                  
      "name": "CMx99",                                                 
      "MACAddress": "17:C2:BF:74:5D:5E",                               
      "pollPeriod": 5000,                                              
      "showMap": true,                                                 
      "addrType": "public",                                            
      "timeout": 10000,                                                
      "telemetry": [                                                   
        {                                                              
          "key": "Temp",                                               
          "method": "notify",                                          
          "characteristicUUID": "1e821401-4a30-45b4-9e17-93f34e283081",
          "valueExpression": "[0:3]"                                   
        }
       ],
      "attributes": [                                                  
        {                                                              
          "key": "name",                                               
          "method": "read",                                            
          "characteristicUUID": "00002a00-0000-1000-8000-00805f9b34fb",
          "valueExpression": "[:]"                                     
        }                                                              
      ],                                                               
      "attributeUpdates": [                                            
        {                                                              
          "attributeOnThingsBoard": "sharedName",                      
          "characteristicUUID": "00002a00-0000-1000-8000-00805f9b34fb" 
        }                                                              
      ],                                                               
      "serverSideRpc": [                                               
      ]                                                                
    }                                                                  
  ]                                                                    
} 

**Versions (please complete the following information):**
 - OS:  Linux 4.9
 - Thingsboard IoT Gateway version 3.0 latest master branch
 - Python version 3.8

samson0v, 2022-01-10:
Hi @jayneev, please, send the logs of the Gateway.

jayneev, 2022-01-10:
@samson0v, mailed you the logs to your e-mail. Please check.

samson0v, 2022-01-14:
Hi @jayneev, we found the bug and will fix it. I will let you know.

samson0v, 2022-01-17:
Hi @jayneev, let's make a call. I am available from now till 17:00 Ukrainian time via Microsoft teams. Or if you can't today, we can make a call tomorrow.
Make sure you have installed the latest version of Gateway.

jayneev, 2022-01-17:
Hi @samson0v, saw your message very late, let's connect tommorrow 02:00 PM Ukrainian time (5:00 PM IST). I will send you Microsoft teams requests separately.