thingsboard/thingsboard-gateway #690: [HELP] Unable to start TB Gateway 3.0.1 & Question about Remote Configurator

jeffreylauck, 2022-01-11:
**Describe the issue**
I have done a clean installation of the TB Gateway on my Pi Zero today, however I am unable to start the gateway. Is it something to do with new v3.0.1 of the Gateway?

Also, I am reinstalling the TB Gateway in the hope of making the "Remote Configurator" to work on the gateway. I was trying to use the Remote Configurator widget on the server to configure the gateway connectors but had no luck yet. May I know if it is a completed feature, and if any documentation is available?

Thanks!

**Configuration** (Attach your configuration file)
I am using the config from the doc
```
thingsboard:
  host: 192.168.0.24
  port: 1883
  security:
    accessToken: ***REMOVED***
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:

  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

```

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
Illegal instruction
```

**Versions (please complete the following information):**
 - OS: Raspbian 11
 - Thingsboard IoT Gateway version: 3.0.1
 - Python version: 3.9

jeffreylauck, 2022-01-11:
I have also tried to install the latest v3.0.1 Gateway onto the same Raspberry Pi Zero (OS: Raspbian 10), no luck.
But when I then install it on a Raspberry Pi 4 (OS: Raspbian 11), it is able to run.
Does it mean the latest TB Gateway cannot be run on Pi Zero?

Hkrie, 2022-01-14:
@jeffreylauck did you find a solution to your problem? I think my problem is quite similar.

samson0v, 2022-01-20:
Hi @jeffreylauck, can you send the complete log of Gateway starting?
Thanks for your interest in ThingsBoard IoT Gateway!

jeffreylauck, 2022-01-21:
@samson0v Hi, thanks for following up.

I have found a solution, it seems that the wheel for grpcio published on PyWheel doesn't support armv6 (see [https://github.com/grpc/grpc/issues/26474](https://github.com/grpc/grpc/issues/26474))

As a workaround, I tried installing grpcio from the deb packages first, as mentioned in the above issue post
```
sudo apt-get install python3-grpcio
```

And then following the installation guide on thingsboard-gateway website, 
```
git clone https://github.com/thingsboard/thingsboard-gateway.git
cd thingsboard-gateway
python3 setup.py install
mkdir logs
python3 ./thingsboard_gateway/tb_gateway.py
```

Now the gateway is able to run. It seems that the problem is due to incompatibility of grpcio from pywheel with armv6 architecture.

Thanks and I will close the issue.

closed 2022-01-21 by jeffreylauck

jeffreylauck, 2022-01-21:
Hi Harald,

I have found a solution, it seems that the wheel for grpcio published on
PyWheel doesn't support armv6 (see grpc/grpc#26474
<https://github.com/grpc/grpc/issues/26474>)

As a workaround, I tried installing grpcio from the deb packages first, as
mentioned in the above issue post

sudo apt-get install python3-grpcio

And then following the installation guide on thingsboard-gateway website,

git clone https://github.com/thingsboard/thingsboard-gateway.git
cd thingsboard-gateway
python3 setup.py install
mkdir logs
python3 ./thingsboard_gateway/tb_gateway.py

Now the gateway is able to run.
[image: image]
<https://user-images.githubusercontent.com/70197122/150465606-892e9ee5-e26d-46c3-95eb-4cc2c55346f6.png>

It seems that the problem is due to incompatibility of grpcio from pywheel
with armv6 architecture.

Cheers,

Jeffrey