thingsboard/thingsboard-gateway #691: [HELP] OPC-UA connection fails with BadSessionNotActivated error

greg-mcnamara, 2022-01-11:
**Describe the issue**
I'm testing the OPC-UA connector for the first time with KepServerEx v6 OPC server. I get the following error in the logs when it tries to connect to the server. I'm really not sure if the configuration is correct at all, I'm trying to adapt from the documentation examples.

```
""2022-01-11 04:13:42" - |ERROR| - [opcua_connector.py] - opcua_connector - run - 121 - "The session cannot be used because ActivateSession has not been called."(BadSessionNotActivated)"
```

Screenshots from Kepware configuration below:

[image]

[image]

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://192.168.50.210:49320",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic256Sha256",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "${Root\\.Objects\\.Channel1\\.Device1}",
        "deviceNamePattern": "opcdevice-001",
        "attributes": [
          {
            "key": "DeviceRunning",
            "path": "${ns=2;i=5}"
          }
        ],
        "timeseries": [
          {
            "key": "DeviceSpeed",
            "path": "${ns=2;i=5}"
          }
	]
      }
    ]
  }
}
```

**Connector name (If you need help with some connector/converter):**
OPC-UA

**Error traceback (If it was raised):**

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.0.1
 - Python version 3.8.10

samson0v, 2022-01-20:
Hi @greg-mcnamara, try to set `"url": "192.168.50.210:49320"`.
Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2022-01-20:
Hi @greg-mcnamara, please, send the full log of the Gateway.
Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2022-02-03:
Hi @greg-mcnamara, please try to use the version from the master branch and let us know about the results.

samson0v, 2022-02-18:
Hi @greg-mcnamara, can we close the issue?

closed 2022-02-18 by greg-mcnamara