thingsboard/thingsboard-gateway #692: [BUG] After sending a remote configuration to gateway from Dashboard, the gateway crash upon next check_connector_configuration_updates

jeffreylauck, 2022-01-11:
**Describe the bug**
Here is the steps I took when encountered the bug
1. I added a "Gateway configuration (Single device)" widget to the Dashboard on TB Cloud.
[image]
2. And then I added a new connector to the config 
[image]
3. After I save the configuration, the Gateway changed to connect to the new connectors, with `Error Traceback No. 1`. 
4. Despite the gateway seemed to have applied the new connectors and started uploading telemetry to ThingsBoard, the telemetry are not saved on ThingsBoard.
4. However, after 60 seconds, when `check_connector_configuration_updates` is called, the Gateway crashed with the below error output `Error Traceback No. 2`.
5. If I restart the gateway, it will run normally for 60 seconds and then crash again upon `check_connector_configuration_updates` .

**Connector name (If bug in the some connector):**
I used a Request connector in the testing, and below is the connector json (Just as a reference, as the connector was working fine).
```
{
  "host": "http://192.168.1.220:1882",
  "SSLVerify": false,
  "security": {
    "type": "basic",
    "username": "",
    "password": ""
  },
  "mapping": [
    {
      "url": "test-endpoint",
      "httpMethod": "GET",
      "httpHeaders": {
        "ACCEPT": "application/json"
      },
      "allowRedirects": true,
      "timeout": 0.5,
      "scanPeriod": 5,
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "Test Request Device",
        "deviceTypeJsonExpression": "${deviceType}",
        "attributes": [
          {
            "key": "generatedNumber",
            "type": "string",
            "value": "${value}"
          }
        ]
      }
    }
  ],
  "#attributeUpdates": [
      {
        "httpMethod": "POST",
        "httpHeaders": {
          "CONTENT-TYPE": "application/json"
        },
        "timeout": 0.5,
        "tries": 3,
        "allowRedirects": true,
        "deviceNameFilter": "SD.*",
        "attributeFilter": "send_data",
        "requestUrlExpression": "sensor/${deviceName}/${attributeKey}",
        "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
      }
  ],
  "#serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTimeout": 1,
      "httpMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 0.5,
[... 127 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 243, in __init__
    self.check_connector_configuration_updates()
  File "/usr/local/lib/python3.9/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 487, in check_connector_configuration_updates
    if stat(connector_config["config_file_path"]) != connector_config["config_updated"]:
KeyError: 'config_file_path'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/thingsboard-gateway", line 8, in <module>
    sys.exit(daemon())
  File "/usr/local/lib/python3.9/dist-packages/thingsboard_gateway/tb_gateway.py", line 27, in daemon
    TBGatewayService("/etc/thingsboard-gateway/config/tb_gateway.yaml".replace('/', path.sep))
  File "/usr/local/lib/python3.9/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 254, in __init__
    self.__stop_gateway()
  File "/usr/local/lib/python3.9/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 271, in __stop_gateway
    self.__grpc_manager.stop()
AttributeError: 'NoneType' object has no attribute 'stop'
```

**Versions (please complete the following information):**
 - OS: Raspbian OS 11 on Raspberry Pi 4
 - Thingsboard IoT Gateway version: 3.0.1.post0
 - Python version: 3.9
 - Thingsboard version: Thingsboard v.3.3.3PAAS-RC1

zbeacon, 2022-01-20:
Hi @jeffreylauck ,

Thank you for your interest in ThingsBoard IoT gateway. We have fixed this issue in master branch, so you can try to use in.

jeffreylauck, 2022-01-20:
Hi @zbeacon ,

Thanks for fixing the issue. I have pulled the latest code from the master branch and try using the remote configurator on it. However, this time, the gateway terminated with a `Segmentation fault` error message after ThingsBoard pushes the Remote Config to the Gateway.

Below is the full log for your reference:-

```
root@pifour1:/home/pi/thingsboard-gateway# python3 ./thingsboard_gateway/tb_gateway.py
Cannot load GRPC connector!
""2022-01-20 16:14:12" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 100 - Gateway starting..."
""2022-01-20 16:14:12" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 105 - ThingsBoard IoT gateway version: 3.0.1"
""2022-01-20 16:14:12" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 1 for device *"
""2022-01-20 16:14:12" - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 154 - Remote shell is enabled. Please be carefully with this feature."
""2022-01-20 16:14:12" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 466 - Connectors - not found! Check your configuration!"
""2022-01-20 16:14:12" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 468 - Remote configuration is enabled forcibly!"
""2022-01-20 16:14:12" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 187 - Gateway started."
""2022-01-20 16:14:12" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 139 - connection SUCCESS"
""2022-01-20 16:14:12" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 2 for device *"
""2022-01-20 16:14:48" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __rpc_gateway_processing - 782 - Received RPC request to the gateway, id: 0, method: getTermInfo"
""2022-01-20 16:15:48" - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - process_configuration - 65 - Remote configuration received: 
 b'{"thingsboard":{"thingsboard":{"host":"thingsboard.cloud","remoteConfiguration":"truee","port":1883,"security":{"accessToken":"**REMOVED**"}},"connectors":[{"configuration":"request.json","name":"Request","type":"request"}],"storage":{"type":"memory","read_records_count":100,"max_records_count":100000},"logs":"W2xvZ2dlcnNdfX1rZXlzPXJvb3QsIHNlcnZpY2UsIGNvbm5lY3RvciwgY29udmVydGVyLCB0Yl9jb25uZWN0aW9uLCBzdG9yYWdlLCBleHRlbnNpb259fVtoYW5kbGVyc119fWtleXM9Y29uc29sZUhhbmRsZXIsIHNlcnZpY2VIYW5kbGVyLCBjb25uZWN0b3JIYW5kbGVyLCBjb252ZXJ0ZXJIYW5kbGVyLCB0Yl9jb25uZWN0aW9uSGFuZGxlciwgc3RvcmFnZUhhbmRsZXIsIGV4dGVuc2lvbkhhbmRsZXJ9fVtmb3JtYXR0ZXJzXX19a2V5cz1Mb2dGb3JtYXR0ZXJ9fVtsb2dnZXJfcm9vdF19fWxldmVsPUVSUk9SfX1oYW5kbGVycz1jb25zb2xlSGFuZGxlcn19W2xvZ2dlcl9jb25uZWN0b3JdfX1sZXZlbD1ERUJVR319aGFuZGxlcnM9Y29ubmVjdG9ySGFuZGxlcn19Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19cXVhbG5hbWU9Y29ubmVjdG9yfX1bbG9nZ2VyX3N0b3JhZ2VdfX1sZXZlbD1ERUJVR319aGFuZGxlcnM9c3RvcmFnZUhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fXF1YWxuYW1lPXN0b3JhZ2V9fVtsb2dnZXJfdGJfY29ubmVjdGlvbl19fWxldmVsPURFQlVHfX1oYW5kbGVycz10Yl9jb25uZWN0aW9uSGFuZGxlcn19Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19cXVhbG5hbWU9dGJfY29ubmVjdGlvbn19W2xvZ2dlcl9zZXJ2aWNlXX19bGV2ZWw9REVCVUd9fWhhbmRsZXJzPXNlcnZpY2VIYW5kbGVyfX1mb3JtYXR0ZXI9TG9nRm9ybWF0dGVyfX1xdWFsbmFtZT1zZXJ2aWNlfX1bbG9nZ2VyX2NvbnZlcnRlcl19fWxldmVsPURFQlVHfX1oYW5kbGVycz1jb252ZXJ0ZXJIYW5kbGVyfX1mb3JtYXR0ZXI9TG9nRm9ybWF0dGVyfX1xdWFsbmFtZT1jb252ZXJ0ZXJ9fVtsb2dnZXJfZXh0ZW5zaW9uXX19bGV2ZWw9REVCVUd9fWhhbmRsZXJzPWNvbm5lY3RvckhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fXF1YWxuYW1lPWV4dGVuc2lvbn19W2hhbmRsZXJfY29uc29sZUhhbmRsZXJdfX1jbGFzcz1TdHJlYW1IYW5kbGVyfX1sZXZlbD1ERUJVR319Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19YXJncz0oc3lzLnN0ZG91dCwpfX1baGFuZGxlcl9jb25uZWN0b3JIYW5kbGVyXX19bGV2ZWw9REVCVUd9fWNsYXNzPWxvZ2dpbmcuaGFuZGxlcnMuVGltZWRSb3RhdGluZ0ZpbGVIYW5kbGVyfX1mb3JtYXR0ZXI9TG9nRm9ybWF0dGVyfX1hcmdzPSgiLi9sb2dzL2Nvbm5lY3Rvci5sb2ciLCAiZCIsIDEsIDcsKX19W2hhbmRsZXJfc3RvcmFnZUhhbmRsZXJdfX1sZXZlbD1ERUJVR319Y2xhc3M9bG9nZ2luZy5oYW5kbGVycy5UaW1lZFJvdGF0aW5nRmlsZUhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fWFyZ3M9KCIuL2xvZ3Mvc3RvcmFnZS5sb2ciLCAiZCIsIDEsIDcsKX19W2hhbmRsZXJfc2VydmljZUhhbmRsZXJdfX1sZXZlbD1ERUJVR319Y2xhc3M9bG9nZ2luZy5oYW5kbGVycy5UaW1lZFJvdGF0aW5nRmlsZUhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fWFyZ3M9KCIuL2xvZ3Mvc2VydmljZS5sb2ciLCAiZCIsIDEsIDcsKX19W2hhbmRsZXJfY29udmVydGVySGFuZGxlcl19fWxldmVsPURFQlVHfX1jbGFzcz1sb2dnaW5nLmhhbmRsZXJzLlRpbWVkUm90YXRpbmdGaWxlSGFuZGxlcn19Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19YXJncz0oIi4vbG9ncy9jb252ZXJ0ZXIubG9nIiwgImQiLCAxLCAzLCl9fVtoYW5kbGVyX2V4dGVuc2lvbkhhbmRsZXJdfX1sZXZlbD1ERUJVR319Y2xhc3M9bG9nZ2luZy5oYW5kbGVycy5UaW1lZFJvdGF0aW5nRmlsZUhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fWFyZ3M9KCIuL2xvZ3MvZXh0ZW5zaW9uLmxvZyIsICJkIiwgMSwgMywpfX1baGFuZGxlcl90Yl9jb25uZWN0aW9uSGFuZGxlcl19fWxldmVsPURFQlVHfX1jbGFzcz1sb2dnaW5nLmhhbmRsZXJzLlRpbWVkUm90YXRpbmdGaWxlSGFuZGxlcn19Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19YXJncz0oIi4vbG9ncy90Yl9jb25uZWN0aW9uLmxvZyIsICJkIiwgMSwgMywpfX1bZm9ybWF0dGVyX0xvZ0Zvcm1hdHRlcl19fWZvcm1hdD0iJShhc2N0aW1lKXMgLSAlKGxldmVsbmFtZSlzIC0gWyUoZmlsZW5hbWUpc10gLSAlKG1vZHVsZSlzIC0gJShsaW5lbm8pZCAtICUobWVzc2FnZSlzIiB9fWRhdGVmbXQ9IiVZLSVtLSVkICVIOiVNOiVTIg=="},"request":[{"name":"Request","config":{"host":"http://192.168.1.220:1882","SSLVerify":false,"security":{"type":"basic","username":"","password":""},"mapping":[{"url":"test-endpoint","httpMethod":"GET","httpHeaders":{"ACCEPT":"application/json"},"allowRedirects":true,"timeout":0 [... cut ...]
""2022-01-20 16:15:48" - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - __process_connectors_configuration - 103 - Processing remote connectors configuration..."
""2022-01-20 16:15:48" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import RequestConnector from /usr/local/lib/python3.7/dist-packages/thingsboard_gateway-3.0.1-py3.7.egg/thingsboard_gateway/connectors/request."
""2022-01-20 16:15:48" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_unsubscribe - 183 - Unsubscribed from *|*, subscription id '*'"
Segmentation fault
```

closed 2022-01-26 by zbeacon