thingsboard/thingsboard-gateway #695: grpcio not supported on OpenWrt

serifis84, 2022-01-13:
After pip3 upgrading thingsboard-gateway i realised grpcio is needed, but it is not supported yet by openwrt. I cannot compile it from source. I am using tb-gateway on Teltonika and Mikrotik (with openwrt firmware) routers.

zbeacon, 2022-01-17:
Hi @serifis84 ,

You can try to remove grpcio dependency form setup.py file and try to compile again. 
We have updated the master branch and if the gateway cannot load grpc in the code - it will be ignored, but you won't be able to use this connector.
In order to build the gateway image - you can pull the latest master branch version, remove the following dependencies from setup.py: 'grpcio', 'protobuf'. Then you are able to build the image as you did it before.

serifis84, 2022-01-17:
@zbeacon Oh! thank you very much sir! I will try it. The issue is now fixed by installing it with: 
pip3 install thingsboard-gateway==2.9

closed 2022-01-17 by serifis84