thingsboard/thingsboard-gateway #696: [HELP] Connecting to Thingsboard and using rest connector

Hkrie, 2022-01-14:
**Describe the issue**
I try to connect to thingsboard and use the rest connector without success.
I have a Thingsboard instance running on:
http://thingsboard.demo.local/
and a rest-api-endpoint:
http://localhost:8052/mmsp/status/systemStatus/get
==> These work correctly in browser.

I followed the installation guide:
https://thingsboard.io/docs/iot-gateway/install/source-installation/ 
and configured the tb_gateway.yml using the tb-gateway-configurator:
https://thingsboard.io/docs/iot-gateway/guides/how-to-configure-gateway-using-configurator/
After that I adjusted the rest.json.

To see that I can actually connect to the thingsboard and it's not a problem with my thingsboard instance I ran a simple node script:
```
const mqtt = require('mqtt')

const client = mqtt.connect('mqtt://thingsboard.demo.local/', {
    username: MY_ACCESS_TOKEN
})

client.on('connect', function () {
    console.log('connected')
})
```

**Configuration** (Attach your configuration file)
tb_gateway.yaml
```
thingsboard:
  host: thingsboard.demo.local
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 30
  security:
    accessToken: MY_ACCESS_TOKEN
  qos: 1
storage:
  type: memory
  read_records_count: 10
  max_records_count: 1000
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
    - 
    type: rest
    name: Rest Connector
    configuration: rest.json
```
rest.json
```
{
  "host": "127.0.0.1",
  "port": "8052",
  "SSL": false,
  "mapping":[
    {
      "endpoint": "/mmsp/status/systemStatus/get",
      "HTTPMethods": [
[... 77 lines cut ...]
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 162 - connectRequests handler has all mandatory keys => accepted: {"topicFilter": "sensor/connect", "deviceNameJsonExpression": "${SerialNumber}"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 154 - Mandatory key 'topicFilter' found in connectRequests handler: {"topicFilter": "sensor/+/connect", "deviceNameTopicExpression": "(?<=sensor/)(.*?)(?=/connect)"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 162 - connectRequests handler has all mandatory keys => accepted: {"topicFilter": "sensor/+/connect", "deviceNameTopicExpression": "(?<=sensor/)(.*?)(?=/connect)"}'
''2022-01-14 11:34:19' - INFO - mqtt_connector - 165 - Number of accepted connectRequests handlers: 2'
''2022-01-14 11:34:19' - INFO - mqtt_connector - 169 - Number of rejected connectRequests handlers: 0'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 154 - Mandatory key 'topicFilter' found in disconnectRequests handler: {"topicFilter": "sensor/disconnect", "deviceNameJsonExpression": "${SerialNumber}"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 162 - disconnectRequests handler has all mandatory keys => accepted: {"topicFilter": "sensor/disconnect", "deviceNameJsonExpression": "${SerialNumber}"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 154 - Mandatory key 'topicFilter' found in disconnectRequests handler: {"topicFilter": "sensor/+/disconnect", "deviceNameTopicExpression": "(?<=sensor/)(.*?)(?=/disconnect)"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 162 - disconnectRequests handler has all mandatory keys => accepted: {"topicFilter": "sensor/+/disconnect", "deviceNameTopicExpression": "(?<=sensor/)(.*?)(?=/disconnect)"}'
''2022-01-14 11:34:19' - INFO - mqtt_connector - 165 - Number of accepted disconnectRequests handlers: 2'
''2022-01-14 11:34:19' - INFO - mqtt_connector - 169 - Number of rejected disconnectRequests handlers: 0'
''2022-01-14 11:34:19' - ERROR - mqtt_connector - 142 - 'attributeRequests' section missing from configuration'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 154 - Mandatory key 'deviceNameFilter' found in attributeUpdates handler: {"deviceNameFilter": "SmartMeter.*", "attributeFilter": "uploadFrequency", "topicExpression": "sensor/${deviceName}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 154 - Mandatory key 'attributeFilter' found in attributeUpdates handler: {"deviceNameFilter": "SmartMeter.*", "attributeFilter": "uploadFrequency", "topicExpression": "sensor/${deviceName}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 154 - Mandatory key 'topicExpression' found in attributeUpdates handler: {"deviceNameFilter": "SmartMeter.*", "attributeFilter": "uploadFrequency", "topicExpression": "sensor/${deviceName}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 154 - Mandatory key 'valueExpression' found in attributeUpdates handler: {"deviceNameFilter": "SmartMeter.*", "attributeFilter": "uploadFrequency", "topicExpression": "sensor/${deviceName}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}'
''2022-01-14 11:34:19' - DEBUG - mqtt_connector - 162 - attributeUpdates handler has all mandatory keys => accepted: {"deviceNameFilter": "SmartMeter.*", "attributeFilter": "uploadFrequency", "topicExpression": "sensor/${deviceName}/${attributeKey}", "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"}'
''2022-01-14 11:34:19' - INFO - mqtt_connector - 165 - Number of accepted attributeUpdates handlers: 1'
''2022-01-14 11:34:19' - INFO - mqtt_connector - 169 - Number of rejected attributeUpdates handlers: 0'
```

**Versions (please complete the following information):**
 - OS: Windows 10 Subsystem: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.0.1
 - Python version 3.8.10

samson0v, 2022-01-14:
Hi @Hkrie, I can't see from logs that you are using REST Connector, I see only MQTT Connector and it seems that you are using the wrong ThingsBoard Gateway token.

Hkrie, 2022-01-14:
@samson0v well i did configure the rest connector though... I edited the rest.json that was by the updated tb_gateway.yaml.
(The other config files where created automatically when installing thingsboard-gateway since they are not used in the tb_gateway.yaml i will delete them later on)
[image]

The token should be correct since i can connect with it using nodejs.
[image]

Hkrie, 2022-01-14:
@samson0v Do you have a working example on github that i could see for reference? Maybe the folder structure that i got with the source installation (https://thingsboard.io/docs/iot-gateway/install/source-installation/) is malformed somehow...

zbeacon, 2022-01-14:
Hi @Hkrie,

Could you try to use IP address instead of thingsboard.demo.local, it may be the issue with dns resolving. Because we can see that the gateway receives a response with message ` connection FAIL with error 3 server unavailable`. 
Do you can connect to ThingsBoard using any MQTT client?

Hkrie, 2022-01-14:
@zbeacon i am using a vpn to access the thingsboard instance over vpn. I just used the ip address like you advised but the output seems to be the same. Did I maybe edit the wrong files?

I am starting the thingsboard gateway with the command:
thingsboard-gateway

zbeacon, 2022-01-14:
Do you have files in /etc/thingsboard-gateway/config ?

Hkrie, 2022-01-14:
no. should i create those files (like descriped here: https://thingsboard.io/docs/iot-gateway/configuration/) next to the thingsboard-gateway folder that i currently have?

zbeacon, 2022-01-14:
Yes, please move your config folder to /etc/thingsboard-gateway/
And then run thingsboard-gateway

Hkrie, 2022-01-14:
should the etc folder be here:
parent-folder\thingsboard_gateway\etc
or here:
parent-folder\thingsboard_gateway\thingsboard_gateway\etc
or here:
parent-folder\etc

Hkrie, 2022-01-14:
I get the following output:
```
root@LT000985:~/thingsboard/example-gateway# thingsboard-gateway
[STREAM ONLY] 2022-01-14 13:53:35,348 - ERROR - [mqtt_connector.py] - mqtt_connector - 142 - 'attributeRequests' section missing from configuration
[STREAM ONLY] 2022-01-14 13:53:35,349 - ERROR - [mqtt_connector.py] - mqtt_connector - 207 - [Errno 111] Connection refused
[STREAM ONLY] 2022-01-14 13:53:35,731 - ERROR - [tb_device_mqtt.py] - tb_device_mqtt - 146 - connection FAIL with error 3 server unavailable
[STREAM ONLY] 2022-01-14 13:53:35,732 - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 123 - Disconnected client: <paho.mqtt.client.Client object at 0x7fc5fabbe5e0>, user data: None, result code: 0
[STREAM ONLY] 2022-01-14 13:53:45,360 - ERROR - [mqtt_connector.py] - mqtt_connector - 207 - [Errno 111] Connection refused
[STREAM ONLY] 2022-01-14 13:53:55,371 - ERROR - [mqtt_connector.py] - mqtt_connector - 207 - [Errno 111] Connection refused
[STREAM ONLY] 2022-01-14 13:54:05,381 - ERROR - [mqtt_connector.py] - mqtt_connector - 207 - [Errno 111] Connection refused
```

[image]

Hkrie, 2022-01-14:
If i put the etc folder where the config folder was before:
[image]

and run the command:
/example-gateway/thingsboard-gateway# thingsboard-gateway

i get pretty much the same error like before:
[complete-command-line-output.txt](https://github.com/thingsboard/thingsboard-gateway/files/7870580/complete-command-line-output.txt)

Hkrie, 2022-01-14:
@zbeacon Does Thingsboard provide a barebone example-demo project where I could see the complete folder structure for a gateway?

zbeacon, 2022-01-20:
@Hkrie ,

It looks like you have a daemon installation before, so please check folder /etc/thingsboard-gateway/config in your system, not in parent folder.

samson0v, 2022-02-18:
Hi @Hkrie, can we close the issue?