thingsboard/thingsboard-gateway #698: [BUG]

ysimonx, 2022-01-14:
**Scanning OPCUA path is very long**
Each time I start thingsboard-gateway, the process scans for UA variables specified in a opcua.json file with a path 
example
`{ 
            "key": "deltap100",
            "path": "${User Registers\\.cPURE\\.Fan\\.DeltaP_100}"
          }`
          
The process is very long before the gateway is ready
It takes at least 3 minutes for 15 variables (7 telemetry, 7 attributes, 1 shared attribute)
 before being able to process MQTT uplink and downlink

** OPC-UA Connector:**

**Versions :**
 - OS: Mac OS
 - Thingsboard IoT Gateway version 3.0
 - Python version 3.9

ysimonx, 2022-01-14:
reverse-engineering : 

`
def get_node_path(self, node: Node):
        return '\\.'.join(node.get_browse_name().Name for node in node.get_path(200000))
`

builds a fullpath for each node and is too long

if I add a "LRU Cache" for this function
`@lru_cache(maxsize=None)
    def get_node_path(self, node: Node):
        return '\\.'.join(node.get_browse_name().Name for node in node.get_path(200000))
`

Instead of 3 minutes, I have to wait only 30 seconds
6 times faster

ysimonx, 2022-01-14:
Same result with
```

@cachetools.func.ttl_cache(maxsize=1000, ttl=10 * 60)
    def get_node_path(self, node: Node):
```

samson0v, 2022-01-18:
Hi @ysimonx, is it ok to close this issue?

ysimonx, 2022-01-18:
yes @samson0v :)
thank you

closed 2022-01-18 by ysimonx