thingsboard/thingsboard-gateway #700: [BUG] 3.0.1 gateway docker not starting?

rduivenvoorde, 2022-01-17:
**Describe the bug**

Updating th official gateway docker from 2.8 to 3.0.1, there is no connection to ThingsBoard...

Rolling back to 2.8 all is ok again.

In service logs I see:

**Error traceback (If available):**
```
 ""2022-01-17 12:12:38" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 98 - ThingsBoard IoT gateway version: 3.0.1"
 27 ""2022-01-17 12:12:38" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 112 - cannot join thread before it is started"
 28 Traceback (most recent call last):
 29   File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 110, in __init__
 30     self.tb_client.disconnect()
 31   File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/gateway/tb_client.py", line 106, in disconnect
 32     self.client.disconnect()
 33   File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/tb_client/tb_device_mqtt.py", line 175, in disconnect
 34     self._client.loop_stop()
 35   File "/usr/local/lib/python3.9/site-packages/paho_mqtt-1.6.1-py3.9.egg/paho/mqtt/client.py", line 1813, in loop_stop
 36     self._thread.join()
 37   File "/usr/local/lib/python3.9/threading.py", line 1048, in join
 38     raise RuntimeError("cannot join thread before it is started")
 39 RuntimeError: cannot join thread before it is started
 40 ""2022-01-17 12:12:38" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import MqttConnector from /usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/mqtt."
```
Could the above be related to the issue of gateway not connecting to the Thingsboard server?

samson0v, 2022-01-20:
Hi @rduivenvoorde, try to completely delete the Gateway container and reinstall it.
Thanks for your interest in ThingsBoard IoT Gateway!

rduivenvoorde, 2022-01-20:
@samson0v thank, but this it is a fresh install/deployment... I think?
We deploy containers via ansible, so I upload the version there, and a new image is uploaded and then a fresh container is started (with the mounts to the config etc)?
DO you have an indication what could be the issue then? Because it it (what I think) a fresh image/container, there can be no Pythonpath issues or so?

samson0v, 2022-01-20:
@rduivenvoorde I see exception as in you for the first time, so will investigate)

rduivenvoorde, 2022-01-21:
@samson0v thanks! But, seeing the exception, do you then also confirm that there is no info flowing between gateway and TB?
Or do you only see the exception, but you still have a working gateway (which in my case is **not** working)?

rduivenvoorde, 2022-01-31:
Some more trials:

Creating an Image from current master does not fix this issue yet (I was hoping it was fixed by one of the recent commits :-)) 

Using current master, in tb_connection.log I see only : 
```""2022-01-31 08:23:53" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 1 for device *"```
and no 'connection SUCCESS' message...

While going back to 2.8:
```
""2022-01-31 11:14:38" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 178 - Subscribed to *|* with id 1 for device *"
""2022-01-31 11:14:38" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 138 - connection SUCCESS"
```

Is this reproducable for others?

samson0v, 2022-02-07:
Hi @rduivenvoorde, any progress? because we can't reproduce this issue. I have another Docker file written by me, you can try to use it. You have to create Dockerfile in the project root folder (thingsboard-gateway/).
```
FROM python:3.9-slim-buster
ENV PYTHONUNBUFFERED=1
WORKDIR /thingsboard-gateway
COPY thingsboard_gateway thingsboard_gateway
COPY setup.py setup.py
COPY README.md README.md
RUN apt-get update && apt-get install -y gcc python3-dev build-essential
RUN pip install --upgrade pip
RUN python setup.py install
RUN apt-get autoremove -y gcc
ENV configs /thingsboard_gateway/config
ENV extensions /thingsboard_gateway/extensions
ENV logs /thingsboard_gateway/logs
VOLUME ["${configs}", "${extensions}", "${logs}"]
CMD ["python3", "thingsboard_gateway/tb_gateway.py"]
```

rduivenvoorde, 2022-02-07:
@samson0v thanks for this. Not sure what 'fixed' it, but taking the latest image 3.0.1 now just works...

Sorry for the noise...

Will close this one.

closed 2022-02-07 by rduivenvoorde