thingsboard/thingsboard-gateway #706: [BUG] Modbus converter not reading the full register when decoding to binary format (or bits)

VeryBitter, 2022-01-22:
Hi everyone

There is a bug in the file bytes_modbus_uplink_converter.py. When decoding into bits the register obtained with function code 3 or 4, the decoded result is limited to bit 0 to 7 (one byte), while the full register is 16 bit long (two bytes as usual with modbus). 

When I add a device to be read from the modbus server to the modbus.json file, like this one
        {
            "tag": "test",
            "address": 2,
            "type": "bits",
            "bit":9, # which larger than a byte>7 
            "functionCode": 3,
            "objectsCount": 1
        }

I get an error, which is too bad because I cannot read the complete set of the binary (boolean) states of my device.

Here is an easy (dirty?) fix from the pymodbus GitHub issues [https://github.com/riptideio/pymodbus/issues/634#issuecomment-845807380](https://github.com/riptideio/pymodbus/issues/634#issuecomment-845807380)

line 127 in  bytes_modbus_uplink_converter.py
elif decoder_functions.get(lower_type) is not None:
            decoded = decoder_functions[lower_type]()
#add this line
            decoded += decoder_functions[lower_type]()

It works for me, hope this will not affect other conversion and that will help. Be careful that the wordoerder and the bytorder values have no effect on the bit order.
So you have to figure it out  (I am not sure of it so I let you find out).

Thingsboard is great! thanks for making it open-source!

zbeacon, 2022-01-24:
Hi @VeryBitter ,

Thank you for your interest and participating in the project life. Could you make a PR with your fix to the repository?

samson0v, 2022-01-28:
Hi @VeryBitter, resolve your problem via PR #716

VeryBitter, 2022-01-28:
Guys,
this is my first pull request ever, I may have not done things the right way, I am very sorry.
I pulled a request from my fork since I found out that the bit order was not right with my first solution.
This one is better
```
        elif lower_type == "bits":
            decoded_lastbyte= decoder_functions[type_]()
            decoded= decoder_functions[type_]()
            decoded+=decoded_lastbyte
```
I am sure my pull request was effective.

VeryBitter, 2022-02-06:
Hi I think I succeeded in creating my first pull request see #728. I changed the order of the two bytes, to get the correct 16 bit order, I think. Works with my modbus devices. Could be different on some other applications. We would need then to get byte_order to work during bit decoding. If anyone knows...

closed 2022-02-07 by zbeacon