thingsboard/thingsboard-gateway #707: [HELP] Attaching Serial Device to Docker on Windows 10

cowherdboy, 2022-01-23:
I would like to know how to attach RS485 serial device to Docker on Windows 10. The device shows as COM13 on Windows but I do not see any changes in /dev directory on Docker before and after attaching the device. I am trying to use the modbus_serial connector and need to know what to put for 'port' parameter. I am using Thingsboard IoT Gateway version 3.0.1.

zbeacon, 2022-01-24:
Hi @cowherdboy ,

Please try to add --device parameter to docker run command, it may help to expose the port to the container - official article from Microsoft is [here](https://docs.microsoft.com/en-us/virtualization/windowscontainers/deploy-containers/hardware-devices-in-containers).
Then please check /dev folder in container and device port should be there.

cowherdboy, 2022-01-24:
Hi @zbeacon 

I tried the instructions in the page but no luck. The page does say it is only for Windows container.

cowherdboy, 2022-01-26:
Hi @zbeacon 

There seems to be a way to do this for linux containers in Windows:
https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/
https://docs.microsoft.com/en-us/windows/wsl/connect-usb

I tried and it works for Ubuntu-20.04 WSL instance but not for thingsboard-gateway docker. Even for Ubuntu-20.04 the drivers for my USB-RS485 dongle were not present as per:
https://github.com/microsoft/WSL/issues/7649

I had to recompile the kernel to get it to work:
https://askubuntu.com/questions/1373910/ch340-serial-device-doesnt-appear-in-dev-wsl/

Perhaps something to consider in future releases of thingsboard gateway. I have now installed thingboard gateway on raspberry pi 2 but I am having a different issue.

closed 2022-01-26 by cowherdboy