thingsboard/thingsboard-gateway #709: [BUG]Gateway bug

redasdasd, 2022-01-24:
I found a Gateway bug. When I upgrade Gateway on an old raspberry pi. Upgrading from version 2.6 to the latest version, I get this error.
I also got the same error installing Gateway on a new raspberry pi.
### ### ### ### ### ### ### ### ### ### ### ### 
pi@raspberrypi:~ $ systemctl status thingsboard-gateway

● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2022-01-24 11:25:53 CET; 290ms ago
 Main PID: 1191 (python3)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/thingsboard-gateway.service
           └─1191 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Jan 24 11:25:53 raspberrypi python3[1191]:     from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
Jan 24 11:25:53 raspberrypi python3[1191]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
Jan 24 11:25:53 raspberrypi python3[1191]:     from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
Jan 24 11:25:53 raspberrypi python3[1191]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 21, in <module>
Jan 24 11:25:53 raspberrypi python3[1191]:     import grpc
Jan 24 11:25:53 raspberrypi python3[1191]:   File "/usr/local/lib/python3.7/dist-packages/grpc/__init__.py", line 22, in <module>
Jan 24 11:25:53 raspberrypi python3[1191]:     from grpc import _compression
Jan 24 11:25:53 raspberrypi python3[1191]:   File "/usr/local/lib/python3.7/dist-packages/grpc/_compression.py", line 15, in <module>
Jan 24 11:25:53 raspberrypi python3[1191]:     from grpc._cython import cygrpc
Jan 24 11:25:53 raspberrypi python3[1191]: ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /usr/local/lib/python3.7/dist-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so)
### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### 
pi@raspberrypi:~ $ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2022-01-24 11:25:53 CET; 6s ago
  Process: 1191 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
 Main PID: 1191 (code=exited, status=1/FAILURE)

Jan 24 11:25:53 raspberrypi systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
Jan 24 11:25:53 raspberrypi systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
Jan 24 11:25:53 raspberrypi systemd[1]: Stopped ThingsBoard Gateway.
Jan 24 11:25:53 raspberrypi systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jan 24 11:25:53 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
Jan 24 11:25:53 raspberrypi systemd[1]: Failed to start ThingsBoard Gateway.

redasdasd, 2022-01-24:
I tried your command：

pi@raspberrypi:/tmp $ sudo pip3 install grpcio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: grpcio in /usr/local/lib/python3.7/dist-packages (1.43.0)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio) (1.12.0)

### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### 
pi@raspberrypi:/tmp $ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2022-01-24 13:16:23 CET; 47ms ago
  Process: 1173 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
 Main PID: 1173 (code=exited, status=1/FAILURE)
pi@raspberrypi:/tmp $ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2022-01-24 13:16:25 CET; 3s ago
  Process: 1177 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
 Main PID: 1177 (code=exited, status=1/FAILURE)

Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
Jan 24 13:16:25 raspberrypi systemd[1]: Stopped ThingsBoard Gateway.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
Jan 24 13:16:25 raspberrypi systemd[1]: Failed to start ThingsBoard Gateway.

redasdasd, 2022-01-24:
> 嗨@redasdasd ，
> 
> 请尝试运行以下命令：
> 
> sudo pip3 install grpcio

I tried your command：

pi@raspberrypi:/tmp $ sudo pip3 install grpcio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: grpcio in /usr/local/lib/python3.7/dist-packages (1.43.0)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio) (1.12.0)

### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### 
pi@raspberrypi:/tmp $ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2022-01-24 13:16:23 CET; 47ms ago
  Process: 1173 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
 Main PID: 1173 (code=exited, status=1/FAILURE)
pi@raspberrypi:/tmp $ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2022-01-24 13:16:25 CET; 3s ago
  Process: 1177 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
 Main PID: 1177 (code=exited, status=1/FAILURE)

Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
Jan 24 13:16:25 raspberrypi systemd[1]: Stopped ThingsBoard Gateway.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
Jan 24 13:16:25 raspberrypi systemd[1]: Failed to start ThingsBoard Gateway.

redasdasd, 2022-01-24:
> Hi @redasdasd ,
> 
> Please try to run the following command:
> 
> sudo pip3 install grpcio

I tried your command：

pi@raspberrypi:/tmp $ sudo pip3 install grpcio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: grpcio in /usr/local/lib/python3.7/dist-packages (1.43.0)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio) (1.12.0)

### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### 
pi@raspberrypi:/tmp $ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2022-01-24 13:16:23 CET; 47ms ago
  Process: 1173 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
 Main PID: 1173 (code=exited, status=1/FAILURE)
pi@raspberrypi:/tmp $ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2022-01-24 13:16:25 CET; 3s ago
  Process: 1177 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
 Main PID: 1177 (code=exited, status=1/FAILURE)

Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
Jan 24 13:16:25 raspberrypi systemd[1]: Stopped ThingsBoard Gateway.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jan 24 13:16:25 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
Jan 24 13:16:25 raspberrypi systemd[1]: Failed to start ThingsBoard Gateway.

redasdasd, 2022-01-24:
> Hi @redasdasd ,
> 
> Please try to run the following command:
> 
> sudo pip3 install grpcio
### ### ### ### ### ### ### 
this my system log
### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### 
pi@raspberrypi:/tmp $ sudo journalctl -xe
Jan 24 13:34:28 raspberrypi python3[1308]:     from grpc import _compression
Jan 24 13:34:28 raspberrypi python3[1308]:   File "/usr/local/lib/python3.7/dist-packages/grpc/_compression.py", line 15, in <module>
Jan 24 13:34:28 raspberrypi python3[1308]:     from grpc._cython import cygrpc
Jan 24 13:34:28 raspberrypi python3[1308]: ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /usr/local/lib/python3.7/dist-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so)
Jan 24 13:34:28 raspberrypi systemd[1]: thingsboard-gateway.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit thingsboard-gateway.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Jan 24 13:34:28 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit thingsboard-gateway.service has entered the 'failed' state with result 'exit-code'.
Jan 24 13:34:28 raspberrypi systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
Jan 24 13:34:28 raspberrypi systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit thingsboard-gateway.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jan 24 13:34:28 raspberrypi systemd[1]: Stopped ThingsBoard Gateway.
-- Subject: A stop job for unit thingsboard-gateway.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit thingsboard-gateway.service has finished.
--
-- The job identifier is 3320 and the job result is done.
Jan 24 13:34:28 raspberrypi systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jan 24 13:34:28 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit thingsboard-gateway.service has entered the 'failed' state with result 'exit-code'.
Jan 24 13:34:28 raspberrypi systemd[1]: Failed to start ThingsBoard Gateway.
-- Subject: A start job for unit thingsboard-gateway.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit thingsboard-gateway.service has finished with a failure.
--
-- The job identifier is 3320 and the job result is failed.
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx_port_irq: 197928 callbacks suppressed
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 13:34:32 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 13:34:35 raspberrypi sudo[1311]:       pi : TTY=pts/0 ; PWD=/tmp ; USER=root ; COMMAND=/usr/bin/journalctl -xe
Jan 24 13:34:35 raspberrypi sudo[1311]: pam_unix(sudo:session): session opened for user root by pi(uid=0)

zbeacon, 2022-01-24:
Ok, than please try to install grpc using the following command:

sudo apt install python3-grpcio

redasdasd, 2022-01-24:
> Ok, than please try to install grpc using the following command:
> 
> sudo apt install python3-grpcio
### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### 
like this:
Jan 24 14:39:22 raspberrypi systemd[1]: Started ThingsBoard Gateway.
-- Subject: A start job for unit thingsboard-gateway.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit thingsboard-gateway.service has finished successfully.
--
-- The job identifier is 2236.
Jan 24 14:39:22 raspberrypi python3[1058]: Traceback (most recent call last):
Jan 24 14:39:22 raspberrypi python3[1058]:   File "<string>", line 1, in <module>
Jan 24 14:39:22 raspberrypi python3[1058]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/__init__.py", line 15, in <module>
Jan 24 14:39:22 raspberrypi python3[1058]:     from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
Jan 24 14:39:22 raspberrypi python3[1058]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 32, in <module>
Jan 24 14:39:22 raspberrypi python3[1058]:     from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
Jan 24 14:39:22 raspberrypi python3[1058]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
Jan 24 14:39:22 raspberrypi python3[1058]:     from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
Jan 24 14:39:22 raspberrypi python3[1058]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 21, in <module>
Jan 24 14:39:22 raspberrypi python3[1058]:     import grpc
Jan 24 14:39:22 raspberrypi python3[1058]:   File "/usr/local/lib/python3.7/dist-packages/grpc/__init__.py", line 22, in <module>
Jan 24 14:39:22 raspberrypi python3[1058]:     from grpc import _compression
Jan 24 14:39:22 raspberrypi python3[1058]:   File "/usr/local/lib/python3.7/dist-packages/grpc/_compression.py", line 15, in <module>
Jan 24 14:39:22 raspberrypi python3[1058]:     from grpc._cython import cygrpc
Jan 24 14:39:22 raspberrypi python3[1058]: ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /usr/local/lib/python3.7/dist-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so)
Jan 24 14:39:22 raspberrypi systemd[1]: thingsboard-gateway.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit thingsboard-gateway.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Jan 24 14:39:22 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit thingsboard-gateway.service has entered the 'failed' state with result 'exit-code'.
Jan 24 14:39:22 raspberrypi systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
Jan 24 14:39:22 raspberrypi systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit thingsboard-gateway.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jan 24 14:39:22 raspberrypi systemd[1]: Stopped ThingsBoard Gateway.
-- Subject: A stop job for unit thingsboard-gateway.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit thingsboard-gateway.service has finished.
--
-- The job identifier is 2296 and the job result is done.
Jan 24 14:39:22 raspberrypi systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jan 24 14:39:22 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit thingsboard-gateway.service has entered the 'failed' state with result 'exit-code'.
Jan 24 14:39:22 raspberrypi systemd[1]: Failed to start ThingsBoard Gateway.
-- Subject: A start job for unit thingsboard-gateway.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
[... 6 lines cut ...]
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 14:39:24 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
[... cut ...]
Jan 24 14:39:29 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 14:39:29 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 14:39:29 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 14:39:34 raspberrypi kernel: sc16is7xx_port_irq: 196894 callbacks suppressed
Jan 24 14:39:34 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 14:39:34 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0

zbeacon, 2022-01-24:
Ok, It is strange and I have not discovered behavior like this, please try also 

`sudo pip3 install grpcio -tools -U --no-binary=grpcio --no-binary=grpcio-tools` - if it doesn't help please let us know.

redasdasd, 2022-01-24:
> Ok, It is strange and I have not discovered behavior like this, please try also
> 
> `sudo pip3 install grpcio -tools -U --no-binary=grpcio --no-binary=grpcio-tools` - if it doesn't help please let us know.
### ### ### ### ### ### ### ### ### ### ### ### ### 
same error as log
### ### ### ### ### ### ### ### ### ### ### ### ### 
Jan 24 16:49:47 raspberrypi systemd[1]: Started ThingsBoard Gateway.
-- Subject: A start job for unit thingsboard-gateway.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit thingsboard-gateway.service has finished successfully.
--
-- The job identifier is 1032.
Jan 24 16:49:47 raspberrypi python3[992]: Traceback (most recent call last):
Jan 24 16:49:47 raspberrypi python3[992]:   File "<string>", line 1, in <module>
Jan 24 16:49:47 raspberrypi python3[992]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/__init__.py", line 15, in <module>
Jan 24 16:49:47 raspberrypi python3[992]:     from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
Jan 24 16:49:47 raspberrypi python3[992]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 32, in <module>
Jan 24 16:49:47 raspberrypi python3[992]:     from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
Jan 24 16:49:47 raspberrypi python3[992]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
Jan 24 16:49:47 raspberrypi python3[992]:     from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
Jan 24 16:49:47 raspberrypi python3[992]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 21, in <module>
Jan 24 16:49:47 raspberrypi python3[992]:     import grpc
Jan 24 16:49:47 raspberrypi python3[992]:   File "/usr/local/lib/python3.7/dist-packages/grpc/__init__.py", line 22, in <module>
Jan 24 16:49:47 raspberrypi python3[992]:     from grpc import _compression
Jan 24 16:49:47 raspberrypi python3[992]:   File "/usr/local/lib/python3.7/dist-packages/grpc/_compression.py", line 15, in <module>
Jan 24 16:49:47 raspberrypi python3[992]:     from grpc._cython import cygrpc
Jan 24 16:49:47 raspberrypi python3[992]:  ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /usr/local/lib/python3.7/dist-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so)
Jan 24 16:49:47 raspberrypi systemd[1]: thingsboard-gateway.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit thingsboard-gateway.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Jan 24 16:49:47 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit thingsboard-gateway.service has entered the 'failed' state with result 'exit-code'.
Jan 24 16:49:48 raspberrypi systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
Jan 24 16:49:48 raspberrypi systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit thingsboard-gateway.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jan 24 16:49:48 raspberrypi systemd[1]: Stopped ThingsBoard Gateway.
-- Subject: A stop job for unit thingsboard-gateway.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit thingsboard-gateway.service has finished.
--
-- The job identifier is 1092 and the job result is done.
Jan 24 16:49:48 raspberrypi systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jan 24 16:49:48 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit thingsboard-gateway.service has entered the 'failed' state with result 'exit-code'.
Jan 24 16:49:48 raspberrypi systemd[1]: Failed to start ThingsBoard Gateway.
-- Subject: A start job for unit thingsboard-gateway.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
[... 8 lines cut ...]
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- Documentation: https://www.freedesktop.org/wiki/Software/systemd/multiseat
--
-- A new session with the ID 4 has been created for the user pi.
--
-- The leading process of the session is 966.
Jan 24 16:49:48 raspberrypi systemd[1]: Started Session 4 of user pi.
-- Subject: A start job for unit session-4.scope has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit session-4.scope has finished successfully.
--
-- The job identifier is 1152.
Jan 24 16:49:51 raspberrypi systemd[1]: systemd-fsckd.service: Succeeded.
-- Subject: Unit succeeded
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit systemd-fsckd.service has successfully entered the 'dead' state.
Jan 24 16:49:52 raspberrypi kernel: sc16is7xx_port_irq: 196671 callbacks suppressed
Jan 24 16:49:52 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0
Jan 24 16:49:52 raspberrypi kernel: sc16is7xx spi0.0: ttySC1: Unexpected interrupt: 0
Jan 24 16:49:52 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0

redasdasd, 2022-01-24:
> Ok, It is strange and I have not discovered behavior like this, please try also
> 
> `sudo pip3 install grpcio -tools -U --no-binary=grpcio --no-binary=grpcio-tools` - if it doesn't help please let us know.
### ### ### ### ### ### ### ### 
I found an almost identical bug. but no solution.

https://github.com/thingsboard/thingsboard-gateway/issues/366

luolitao, 2022-01-25:
pip3 install --upgrade protobuf

redasdasd, 2022-01-25:
> pip3 install --upgrade protobuf
### ### ### ### ### ### ### ### ### ### ### 
pip3 install --upgrade protobuf
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: protobuf in /usr/local/lib/python3.7/dist-packages (3.19.3)
### ### ### ### ### ### ### ### ### ### ### ### 
Jan 25 09:25:51 raspberrypi python3[1165]: Traceback (most recent call last):
Jan 25 09:25:51 raspberrypi python3[1165]:   File "<string>", line 1, in <module>
Jan 25 09:25:51 raspberrypi python3[1165]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/__init__.py", line 15, in <module>
Jan 25 09:25:51 raspberrypi python3[1165]:     from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
Jan 25 09:25:51 raspberrypi python3[1165]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 32, in <module>
Jan 25 09:25:51 raspberrypi python3[1165]:     from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
Jan 25 09:25:51 raspberrypi python3[1165]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
Jan 25 09:25:51 raspberrypi python3[1165]:     from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
Jan 25 09:25:51 raspberrypi python3[1165]:   File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 21, in <module>
Jan 25 09:25:51 raspberrypi python3[1165]:     import grpc
Jan 25 09:25:51 raspberrypi python3[1165]:   File "/usr/local/lib/python3.7/dist-packages/grpc/__init__.py", line 22, in <module>
Jan 25 09:25:51 raspberrypi python3[1165]:     from grpc import _compression
Jan 25 09:25:51 raspberrypi python3[1165]:   File "/usr/local/lib/python3.7/dist-packages/grpc/_compression.py", line 15, in <module>
Jan 25 09:25:51 raspberrypi python3[1165]:     from grpc._cython import cygrpc
Jan 25 09:25:51 raspberrypi python3[1165]: ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /usr/local/lib/python3.7/dist-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so)
Jan 25 09:25:51 raspberrypi systemd[1]: thingsboard-gateway.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit thingsboard-gateway.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Jan 25 09:25:51 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit thingsboard-gateway.service has entered the 'failed' state with result 'exit-code'.
Jan 25 09:25:51 raspberrypi systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
Jan 25 09:25:51 raspberrypi systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit thingsboard-gateway.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jan 25 09:25:51 raspberrypi systemd[1]: Stopped ThingsBoard Gateway.
-- Subject: A stop job for unit thingsboard-gateway.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit thingsboard-gateway.service has finished.
--
-- The job identifier is 2243 and the job result is done.
Jan 25 09:25:51 raspberrypi systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jan 25 09:25:51 raspberrypi systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit thingsboard-gateway.service has entered the 'failed' state with result 'exit-code'.
Jan 25 09:25:51 raspberrypi systemd[1]: Failed to start ThingsBoard Gateway.
-- Subject: A start job for unit thingsboard-gateway.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit thingsboard-gateway.service has finished with a failure.
--
-- The job identifier is 2243 and the job result is failed.
Jan 25 09:25:54 raspberrypi kernel: sc16is7xx_port_irq: 198184 callbacks suppressed
Jan 25 09:25:54 raspberrypi kernel: sc16is7xx spi0.0: ttySC0: Unexpected interrupt: 0

redasdasd, 2022-01-25:
I installed Gateway version 2.9 on a new raspberry pi. Everything works fine. The error seems to be on the latest version.

ndrs-cwt, 2022-01-27:
In my case, ```ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33'``` is shown because my libc6 on my system is too old to use the lastest grpcio. 

IIRC my workaround back then is 
-remove grpcio and protobuf from setup.py
-create new deb package
-install that deb package
-manual install grpcio by ```sudo apt install python3-grpcio``` to get the lib that compatible with libc on my system
-install/upgrade protobuf from pip

Not sure if this can help.

ferguscan, 2022-02-09:
I can confirm that the latest Docker image is broken on ARMv7.  The issue is resolved by downgrading grpcio.  I created a pull request: https://github.com/thingsboard/thingsboard-gateway/pull/738

dh-jy, 2022-05-07:
Hi  I also got the same error installing Gateway(3.0.1.post0) on a new raspberry pi.
  my install command: sudo pip3 install thingsboard-gateway
Could you give some help?
[image]
[image]

santobedi, 2022-05-13:
The `grpc` support in the latest version of TB gateway sadly gives **GLIBC** error, for example:  `ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33'` in Raspberry Pi. Following solutions work for me:

**Step 1:**

```
pip3 install cython
export GRPC_BUILD_WITH_BORING_SSL_ASM=false
export GRPC_PYTHON_BUILD_SYSTEM_OPENSSL=true
export GRPC_PYTHON_BUILD_WITH_CYTHON=true
export GRPC_PYTHON_DISABLE_LIBC_COMPATIBILITY=true
export GRPC_PYTHON_LDFLAGS="-lpthread -Wl,-wrap,memcpy -static-libgcc"
```

**Step 2:** 

```
pip3 install --upgrade grpcio-tools --force-reinstall --no-binary=grpcio-tools
pip3 install --upgrade grpcio --force-reinstall --no-binary=grpcio
```

**Step 2** took a long time and was sometimes never completed (Raspberry Pi 3). So I did next short cut solution:
Uninstall grpcio;
```
pip uninstall grpcio 
pip uninstall grpcio-status 
```

Then,

```
pip install grpcio==1.44.0 --no-binary=grpcio 
pip install grpcio-tools==1.44.0 --no-binary=grpcio-tools
```

However, it also took a long time and didn't work with Raspberry Pi 3, so I didn't install them and tried to run the TB gateway service (after uninstalling `grpcio`). That gives me `import grpc` error. Locate the files that give '`import grpc`' error and edit them by commenting `#import grpc`.  Now, run the TB gateway service. 

I'm not sure what harm uninstalling `grpcio `does, but it solves the problem with the TB gateway.

samson0v, 2022-06-30:
Hi @redasdasd, any updates? Can we close the issue?

closed 2022-08-08 by imbeacon