thingsboard/thingsboard-gateway #712: [BUG] Unable to use Modbus Connector on Raspberry Pi 2

cowherdboy, 2022-01-26:
**Describe the bug**
I wasn't able to use modbus connector on Raspberry Pi 2. I got the following warning:
|WARNING| - [tb_gateway_service.py] - tb_gateway_service - _connect_with_connectors - 473 - Connector implementation not found for Modbus Connector"

**Connector name (If bug in the some connector):**
Modbus Connector

**Error traceback (If available):**
```
pi@raspberrypi:/var/log/thingsboard-gateway $ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-01-26 10:35:32 +08; 6s ago
   Main PID: 1180 (python3)
      Tasks: 8 (limit: 1597)
        CPU: 5.347s
     CGroup: /system.slice/thingsboard-gateway.service
             └─1180 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Jan 26 10:35:32 raspberrypi systemd[1]: Started ThingsBoard Gateway.
Jan 26 10:35:37 raspberrypi python3[1180]: ""2022-01-26 10:35:37" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 93 - Gateway starting..."
Jan 26 10:35:37 raspberrypi python3[1180]: ""2022-01-26 10:35:37" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 98 - ThingsBoard IoT gateway version: 3.0.1"
Jan 26 10:35:38 raspberrypi python3[1180]: ""2022-01-26 10:35:38" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 1 for device *"
Jan 26 10:35:38 raspberrypi python3[1180]: ""2022-01-26 10:35:38" - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _connect_with_connectors - 473 - Connector implementation not found for Modbus Connector"
Jan 26 10:35:38 raspberrypi python3[1180]: ""2022-01-26 10:35:38" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - Gateway started."
Jan 26 10:35:38 raspberrypi python3[1180]: ""2022-01-26 10:35:38" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 139 - connection SUCCESS"
Jan 26 10:35:38 raspberrypi python3[1180]: ""2022-01-26 10:35:38" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 2 for device *"```

**Versions (please complete the following information):**
 - OS: Linux raspberrypi 5.10.92-v7+
 - Thingsboard IoT Gateway version 3.0.1
 - Python version: 3.9.2

zbeacon, 2022-01-26:
Hi @cowherdboy ,

It looks like something wasn't installed, required by Modbus connector, could you try to run the following command and restart the gateway:

```
sudo pip3 install pymodbus pyserial twisted
```

cowherdboy, 2022-01-26:
Hi @zbeacon,

Thank you. After installing I got the following:

Jan 26 18:20:14 raspberrypi python3[1962]: ""2022-01-26 18:20:14" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 93 - Gateway starting..."
Jan 26 18:20:14 raspberrypi python3[1962]: ""2022-01-26 18:20:14" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 98 - ThingsBoard IoT gateway version: 3.0.1"
Jan 26 18:20:14 raspberrypi python3[1962]: ""2022-01-26 18:20:14" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 175 - Subscribed to *|* with id 1 for device *"
Jan 26 18:20:14 raspberrypi python3[1962]: ""2022-01-26 18:20:14" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 139 - connection SUCCESS"
Jan 26 18:20:16 raspberrypi python3[1962]: :0: UserWarning: You do not have a working installation of the service_identity module: 'No module named 'service_identity''.  Please install it from <https://pypi.python.org/pypi/service_identit
y> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Jan 26 18:20:16 raspberrypi python3[1962]: ""2022-01-26 18:20:16" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import ModbusConnector from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus."
Jan 26 18:20:16 raspberrypi python3[1962]: ""2022-01-26 18:20:16" - |WARNING| - [backward_compability_adapter.py] - backward_compability_adapter - convert - 47 - You are using old configuration structure for Modbus connector. It will be D
EPRECATED in the future version! New config file "modbus_new.json" was generated in config/ folder. Please, use it."
Jan 26 18:20:16 raspberrypi python3[1962]: ""2022-01-26 18:20:16" - |ERROR| - [sync.py] - sync - connect - 662 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'"
Jan 26 18:20:16 raspberrypi python3[1962]: ""2022-01-26 18:20:16" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - Gateway started."
Jan 26 18:20:21 raspberrypi python3[1962]: ""2022-01-26 18:20:21" - |ERROR| - [sync.py] - sync - connect - 662 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'"

To resolve the warnings and error, I did the following:
1. service_identity warning: sudo pip3 install service_identity
2. new modbus warning:  In my case, I updated the connector name in tb_gateway.yaml to modbus_new_1.json (which was in my directory)
3. permission denied /dev/ttyUSB0 error: Need to add user thingsboard_gateway to group dialout with this command: 
sudo usermod -a -G dialout thingsboard_gateway (source: https://github.com/esp8266/source-code-examples/issues/26#issuecomment-631138497)

The modbus data is now coming to Thingsboard. Thank you so much for your help @zbeacon

closed 2022-01-26 by cowherdboy

tamerfs, 2023-01-12:
Hi! @imbeacon and @cowherdboy 

I already tried to run 'sudo pip3 install pymodbus pyserial twisted' and got this:

pi@raspberrypi:/etc/thingsboard-gateway/config $ sudo pip3 install pymodbus pyserial twisted
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pymodbus in /usr/local/lib/python3.9/dist-packages (3.0.2)
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (3.5b0)
Requirement already satisfied: twisted in /usr/local/lib/python3.9/dist-packages (22.10.0)
Requirement already satisfied: zope.interface>=4.4.2 in /usr/local/lib/python3.9/dist-packages (from twisted) (5.5.2)
Requirement already satisfied: attrs>=19.2.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.2.0)
Requirement already satisfied: hyperlink>=17.1.1 in /usr/local/lib/python3.9/dist-packages (from twisted) (21.0.0)
Requirement already satisfied: constantly>=15.1 in /usr/local/lib/python3.9/dist-packages (from twisted) (15.1.0)
Requirement already satisfied: typing-extensions>=3.6.5 in /usr/lib/python3/dist-packages (from twisted) (3.7.4.3)
Requirement already satisfied: incremental>=21.3.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.10.0)
Requirement already satisfied: Automat>=0.8.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.10.0)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from Automat>=0.8.0->twisted) (1.16.0)
Requirement already satisfied: idna>=2.5 in /usr/lib/python3/dist-packages (from hyperlink>=17.1.1->twisted) (2.10)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from zope.interface>=4.4.2->twisted) (52.0.0)

im having the same [WARNING] but it was  having this Traceback:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 43, in <module>
    from twisted.internet import reactor
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/reactor.py", line 38, in <module>
    from twisted.internet import default
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/default.py", line 55, in <module>
    install = _getInstallFunction(platform)
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/default.py", line 51, in _getInstallFunction
    from twisted.internet.selectreactor import install
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/selectreactor.py", line 18, in <module>
    from twisted.internet import posixbase
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/posixbase.py", line 16, in <module>
    from twisted.internet import error, tcp, udp
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/tcp.py", line 99, in <module>
    from twisted.internet import abstract, address, base, error, fdesc, main
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/base.py", line 34, in <module>
    from twisted.internet import abstract, defer, error, fdesc, main, threads
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/defer.py", line 42, in <module>
    from typing_extensions import Literal, ParamSpec, Protocol
ImportError: cannot import name 'ParamSpec' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)
""2023-01-12 12:08:56" - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 547 - Connector implementation not found for Modbus Connector"

[error logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/10403560/error.logs.txt)
[modbus.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/10403591/modbus.json.txt)
[tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/10403592/tb_gateway.yaml.txt)

__versions:__
Raspbian GNU Linux 11 (bullseye)
Python 3.9.2
python3-thingsboard-gateway/now 3.2-1 all [installed,local]

can you guys help me?
thank you so much!

imbeacon, 2023-01-12:
Hi @tamerfs,

Please try to upgrade libraries, using the following command: `sudo pip3 install pymodbus pyserial twisted --upgrade`
If you discover some problem, please open a new issue, because this one was closed a year ago and I noticed it just because you mentioned me.