thingsboard/thingsboard-gateway #714: [HELP] Increase sampling rate in Modbus Connector

qwerty13, 2022-01-26:
**Describe the issue**

How to increase sample rate in modbus serial (rtu) mode?
as my attampts, the max telemetry recieve rate (i count them in a Timeseries table gadget) is 5 per second (and sometimes 4 or 3 or even 1). how can i increase it? i just set polling periods less than 200 ms but rate is still 5/s (= 200ms)...

**Configuration** (Attach your configuration file)

thingsboard:
  host: 192.168.1.220
  port: 1883
  security:
    accessToken: ************************
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
connectors:

  -
    name: Modbus Connector
    type: modbus
    configuration: modbus_new_1.json

**Connector name (If you need help with some connector/converter):**

Modbus Serial Connector:

{
  "master": {
    "slaves": [
      {
        "unitId": 105,
        "deviceName": "Mahda TR10 - 3",
        "attributesPollPeriod": 50,
        "timeseriesPollPeriod": 50,
        "sendDataOnlyOnChange": false,
        "timeseries": [
          {
            "tag": "test365",
            "type": "32float",
            "functionCode": 4,
            "registerCount": 2,
            "address": 0
          },
          {
            "tag": "test366",
            "type": "32float",
            "functionCode": 4,
            "registerCount": 2,
            "address": 2
          }
        ],
        "host": null,
        "port": "/dev/ttyUSB0",
        "type": "serial",
        "method": "rtu",
        "timeout": 50,
        "byteOrder": "BIG",
        "wordOrder": null,
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "baudrate": 9600,
        "pollPeriod": 50
      }
    ]
  },
  "slave": null
}

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version: 3.0
 - Python version: 3.6

qwerty13, 2022-01-26:
There's some "sleep(.2)" (ex. in thingsboard_gateway/connectors/modbus/slave.py).
when i change it to sleep(2) sampling rate drops to 1 sample in 2 seconds but sleep(.1) don't increase speed more than sleep(.2)...

qwerty13, 2022-02-10:
Are you look at my issue?
what is the fastest sample rate we can get from a modbus rtu (at 9600) with Thingsboard Gateway? Assume that server device (slave) has no limits.

dh-jy, 2022-08-04:
> **Describe the issue**
> 
> How to increase sample rate in modbus serial (rtu) mode? as my attampts, the max telemetry recieve rate (i count them in a Timeseries table gadget) is 5 per second (and sometimes 4 or 3 or even 1). how can i increase it? i just set polling periods less than 200 ms but rate is still 5/s (= 200ms)... edit: i used modpoll and it can poll even in 10ms
> 
> **Configuration** (Attach your configuration file)
> 
> thingsboard: host: 192.168.1.220 port: 1883 security: accessToken: ************************ storage: type: memory read_records_count: 100 max_records_count: 100000 connectors:
> 
> * name: Modbus Connector
>   type: modbus
>   configuration: modbus_new_1.json
> 
> **Connector name (If you need help with some connector/converter):**
> 
> Modbus Serial Connector:
> 
> { "master": { "slaves": [ { "unitId": 105, "deviceName": "Mahda TR10 - 3", "attributesPollPeriod": 50, "timeseriesPollPeriod": 50, "sendDataOnlyOnChange": false, "timeseries": [ { "tag": "test365", "type": "32float", "functionCode": 4, "registerCount": 2, "address": 0 }, { "tag": "test366", "type": "32float", "functionCode": 4, "registerCount": 2, "address": 2 } ], "host": null, "port": "/dev/ttyUSB0", "type": "serial", "method": "rtu", "timeout": 50, "byteOrder": "BIG", "wordOrder": null, "retries": true, "retryOnEmpty": true, "retryOnInvalid": true, "baudrate": 9600, "pollPeriod": 50 } ] }, "slave": null }
> 
> **Versions (please complete the following information):**
> 
> * OS: Ubuntu 20.04
> * Thingsboard IoT Gateway version: 3.0
> * Python version: 3.6

> **Describe the issue**
> 
> How to increase sample rate in modbus serial (rtu) mode? as my attampts, the max telemetry recieve rate (i count them in a Timeseries table gadget) is 5 per second (and sometimes 4 or 3 or even 1). how can i increase it? i just set polling periods less than 200 ms but rate is still 5/s (= 200ms)... edit: i used modpoll and it can poll even in 10ms
> 
> **Configuration** (Attach your configuration file)
> 
> thingsboard: host: 192.168.1.220 port: 1883 security: accessToken: ************************ storage: type: memory read_records_count: 100 max_records_count: 100000 connectors:
> 
> * name: Modbus Connector
>   type: modbus
>   configuration: modbus_new_1.json
> 
> **Connector name (If you need help with some connector/converter):**
> 
> Modbus Serial Connector:
> 
> { "master": { "slaves": [ { "unitId": 105, "deviceName": "Mahda TR10 - 3", "attributesPollPeriod": 50, "timeseriesPollPeriod": 50, "sendDataOnlyOnChange": false, "timeseries": [ { "tag": "test365", "type": "32float", "functionCode": 4, "registerCount": 2, "address": 0 }, { "tag": "test366", "type": "32float", "functionCode": 4, "registerCount": 2, "address": 2 } ], "host": null, "port": "/dev/ttyUSB0", "type": "serial", "method": "rtu", "timeout": 50, "byteOrder": "BIG", "wordOrder": null, "retries": true, "retryOnEmpty": true, "retryOnInvalid": true, "baudrate": 9600, "pollPeriod": 50 } ] }, "slave": null }
> 
> **Versions (please complete the following information):**
> 
> * OS: Ubuntu 20.04
> * Thingsboard IoT Gateway version: 3.0
> * Python version: 3.6

dh-jy, 2022-08-04:
HI

> There's some "sleep(.2)" (ex. in thingsboard_gateway/connectors/modbus/slave.py). when i change it to sleep(2) sampling rate drops to 1 sample in 2 seconds but sleep(.1) don't increase speed more than sleep(.2)...

HI, i have the same problem。Have you solved the problem?  i am looking forward to your early replay

qwerty13, 2022-08-04:
> HI
>
>
> HI, i have the same problem。Have you solved the problem? i am looking forward to your early replay

Unfortunately no.

dh-jy, 2022-08-04:
fine.

dh-jy, 2022-08-04:
I think the gateway may not support high sampling, I use the serial port to directly access the device, can achieve high sampling。

qwerty13, 2022-08-04:
It most be a limitation in gateway's codes or pymodbus... as i remember i had written a python program used pymodbus an it have no problem to achieve fast polling...

dh-jy, 2022-08-04:
ok. 
I extremely  wonder how you solved the problem of high sampling? Can I friend you

qwerty13, 2022-08-04:
> ok. I extremely wonder how you solved the problem of high sampling? Can I friend you

1. Making a python application from scratch
2. Making a ESP-32 based modbus to Wifi Gateway

You may find a better way according to your situation.

dh-jy, 2022-08-04:
Well, thank you very much for your opinion

closed 2022-08-08 by imbeacon