thingsboard/thingsboard-gateway #719: [HELP]

avipala, 2022-02-01:
**Describe the issue**
Create description about your issue, and your actions to solve it.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

avipala, 2022-02-01:
I am using a Gateway that is sending list of devices that is scans periodically. Once it detects a new device it adds it to the list of devices. But once the device is no longer transmitting (for a period of 30 sec), I want to remove it from the device list (which will also remove it from the MAP). It doesn't work automatically, How can I do it?

zbeacon, 2022-02-01:
Hi @avipala ,

Actually, there is no functionality like you describe. But we will add it and set this issue as a feature request.

Thank you for your interest in ThingsBoard IoT gateway.

avipala, 2022-02-01:
This is great news. I am using teh coomunity version runing on a raspberry pi.
How will I know when this feature is ready and how can I integrate it into my existing code?

Is there a way to refresh the list of devices automatically? 
The way I do it currently is to manually remove all devices exceot of the gateway and wait till the live devices come up again. This way dead devices are "removed". If there was an automatic refresh function, I could use that process every x seconds.

zbeacon, 2022-02-01:
> Is it possible to write a function in "Use row style function" inside the "Entities table" which will look at the value of the last appearence time and if it is more than x sec ago, it will delete the line? Can you please suggest how to write this function?

This question does not relate to the gateway and we cannot help you with this. But you can create an issue on [ThingsBoard repository](https://github.com/thingsboard/thingsboard) and ask there.

closed 2022-02-02 by samson0v