thingsboard/thingsboard-gateway #722: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 4: invalid continuation byte

santobedi, 2022-02-02:
**Describe the issue**
Random value (unexpected) as telemetry while using Modbus connector. The  Last Update Time always toggles with fixed past times. However, everything is fine with the Attributes (Client).

**Connector name (If you need help with some connector/converter):** Modbus Connector

**Error traceback (If it was raised):** 
```
""2022-02-02 09:58:18" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 90 - 'utf-8' codec can't decode byte 0xe2 in position 4: invalid continuation byte"
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0.1-py3.8.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 76, in convert
    decoded_data = self.__decode_from_registers(decoder, configuration)
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway-3.0.1-py3.8.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 154, in __decode_from_registers
    result_data = decoded.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 4: invalid continuation byte
```

**Versions (please complete the following information):**
 - OS:  Ubuntu 20.04
 - Thingsboard IoT Gateway version (Latest one installed from source)
 - Python version: 3.9

samson0v, 2022-02-02:
Hi @santobedi, send, please, your Modbus config file.
Thanks for your interest in ThingsBoard IoT Gateway!

santobedi, 2022-02-02:
Hello @samson0v,
Following in my Modbus config file.

```
{
  "master": {
    "slaves": [
      {
        "unitId": 150,
        "deviceName": "energy_meter_electrical_1p_1_1a",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "attributes": [
          {
            "tag": "Vendor_Name",
            "type": "string",
            "functionCode": 3,
            "objectsCount": 16,
            "address": 31044
          },
          {
            "tag": "RF-ID",
            "type": "string",
            "functionCode": 4,
            "objectsCount": 4,
            "address": 31026
          },
          {
            "tag": "Product_Code",
            "type": "string",
            "functionCode": 4,
            "objectsCount": 16,
            "address": 31060
          },
          {
            "tag": "Firmware_Revision",
            "type": "string",
            "functionCode": 4,
            "objectsCount": 6,
            "address": 31076
          },
          {
            "tag": "Hardware_Revision",
            "type": "string",
            "functionCode": 4,
            "objectsCount": 6,
            "address": 31082
          },
          {
            "tag": "Serial_Number",
            "type": "string",
            "functionCode": 4,
            "objectsCount": 10,
            "address": 31088
          },
          {
            "tag": "Asset_Name",
            "type": "string",
            "functionCode": 3,
            "objectsCount": 10,
            "address": 50
          },
          {
            "tag": "Label",
            "type": "string",
            "functionCode": 4,
[... 392 lines cut ...]
            "type": "32int",
            "functionCode": 4,
            "objectsCount": 2,
            "address": 35
          }
        ],
        "host": "10.0.1.244",
        "port": 502,
        "type": "tcp",
        "method": "socket",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": null,
        "retries": null,
        "retryOnEmpty": null,
        "retryOnInvalid": null,
        "baudrate": null,
        "pollPeriod": 5000
      }
    ]
  },
  "slave": null
}

```

santobedi, 2022-02-03:
@samson0v On the MQTT connector within the same IoT gateway (device: raspberry pi 3, Ubuntu 20.04.3 LTS (GNU/Linux 5.4.0-1050-raspi armv7l) ), the attribute is fine and updated. However, telemetry is not.

samson0v, 2022-02-03:
Hi @santobedi, can you, please, write steps to reproduce the bug and also send the registers data (using for example modpoll).

santobedi, 2022-02-03:
Hello @samson0v ,

The Modbus TCP power tags are from [Schneider](https://www.se.com/us/en/). Attached herewith are the screenshots of Telemetry from Gateway device and one of the Modbus power tags. I've noticed that the time shown in Logs of Gateway is same as the time displayed as `last time update`  of the Modbus device (devices created by the gateway in the TB). This `last time update` is not real time. Whereas the time and values displayed in the attribute section are fine. 

Moreover, in the case of the MQTT connector, the values (time-series telemetry and attributes) are fine but the `last time update` in the telemetry is wrong similar to the Modbus connector. 
[image]
[image]

samson0v, 2022-02-03:
@santobedi it seems that your system use the wrong time and it is not a Gateway bug because it uses a system date and time.

santobedi, 2022-02-03:
@samson0v  What about the wrong telemetry? And the raised error?

samson0v, 2022-02-03:
@santobedi, can you, please, write steps to reproduce the error, wrong telemetry and also send the registers data (using for example modpoll).
Thanks!

santobedi, 2022-02-03:
@samson0v , I installed the gateway from [source](https://thingsboard.io/docs/iot-gateway/install/source-installation/). Configured the` tb_gateway.yaml` ,`modbus.json`, and `mqtt.json` (/home/myDevice/thingsboard-gateway/thingsboard_gateway/config/), then ran `python3 ./thingsboard_gateway/tb_gateway.py` command. After this, the devices are created in the TB. 

The wrong telemetry from one of the devices (created in TB via the gateway) is shown in the attached screenshot. I've attached the screenshot showing the client attribute (which has no issue at all) of the same device for your reference. If it's due to the system time, how come it affects only Telemetry, but not the Attributes?  At this moment, I could not run the modpoll in our device (deployed in distance location), however, the same device had no such issues before (the earlier version of the gateway).
[image]
[image]

samson0v, 2022-05-20:
Hi @santobedi, is this bug still actual for you?

santobedi, 2022-05-20:
Hello @samson0v ,

I have not experienced such bugs lately.

closed 2022-05-20 by imbeacon

santobedi, 2022-05-23:
Hello @samson0v, @imbeacon,

The same issue occurred again. For the devices created through the MQTT connector, only the attribute gets updated, Telemetry does not. This time the gateway is DEB.

santobedi, 2022-05-24:
Hi @samson0v , @imbeacon ,

This time, the issue occurred when the same MQTT config file (that results in the same virtual devices) has been used for two different gateways (although only one gateway is present in the server and in practice) connecting to the same TB server. In other words, the attempt to update (with telemetry) already existing virtual devices (those were created from the old gateway) with the new gateway caused the issue.  Removing all those devices including the gateway from the server, reinstalling the gateway, and several attempts of rebooting the gateway device finally solved the issue.