thingsboard/thingsboard-gateway #724: [BUG]

ysimonx, 2022-02-02:
**Writing Int32 Shared Attribute to OPC UA node fails**
Although I am able to modify a opcua node value of type "double" or "string", I don't manage to modify a Int32 value of a OPCUA Node

I choose to modify a "shared attribute" as a "Integer"
[image]

It is defined as Int32 on OPCUA Server

[image]

(note : when i change this value thanks to Prosys OPC UA Client , the value is updated on OPC UA Server)

So; this is really a Thingsboard-gateway Issue

**OPC-UA Connector**
[e.g. OPC-UA Connector]

**Versions :**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.0.1
 - Python version 3.9.9

I will take a look as soon as possible to the source code of opcua-connector if it helps

zbeacon, 2022-02-03:
Hi @ysimonx ,

Thank you for your interest in ThingsBoard IoT gateway, please try to use the version from the master branch and let us know about the results.

ysimonx, 2022-02-08:
Hi @zbeacon 

I tried the master branch 
I had this error

`

""2022-02-08 15:04:00" - |ERROR| - [opcua_connector.py] - opcua_connector - on_attributes_update - 235 - "The value supplied for the attribute is not of the same type as the attribute"s value."(BadTypeMismatch)"
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 231, in on_attributes_update
    server_variables[variable].set_value(content["data"][variable])
  File "/Users/ysimonx/Library/Python/3.9/lib/python/site-packages/opcua/common/node.py", line 217, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/Users/ysimonx/Library/Python/3.9/lib/python/site-packages/opcua/common/node.py", line 263, in set_attribute
    result[0].check()
  File "/Users/ysimonx/Library/Python/3.9/lib/python/site-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadWriteNotSupported: "The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 233, in on_attributes_update
    server_variables[variable].set_attribute(ua.AttributeIds.Value, ua.DataValue(content["data"][variable]))
  File "/Users/ysimonx/Library/Python/3.9/lib/python/site-packages/opcua/common/node.py", line 263, in set_attribute
    result[0].check()
  File "/Users/ysimonx/Library/Python/3.9/lib/python/site-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadTypeMismatch: "The value supplied for the attribute is not of the same type as the attribute"s value."(BadTypeMismatch)
""2022-02-08 15:04:00" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 595 - 'attributes'"

`

ysimonx, 2022-02-08:
this is what i just made, and now, I manage to update a int32 value

before:

```
    def on_attributes_update(self, content):
        log.debug(content)
        try:
            for server_variables in self.__available_object_resources[content["device"]]['variables']:
                for attribute in content["data"]:
                    for variable in server_variables:
                        if attribute == variable:
                            try:
                                server_variables[variable].set_value(content["data"][variable])
                            except Exception:
                                server_variables[variable].set_attribute(ua.AttributeIds.Value, ua.DataValue(content["data"][variable]))
        except Exception as e:
            log.exception(e)
```

after : 

```

    def on_attributes_update(self, content):
        log.debug(content)
        try:
            for server_variables in self.__available_object_resources[content["device"]]['variables']:
                for attribute in content["data"]:
                    for variable in server_variables:
                        if attribute == variable:
                            try:
                                if ( isinstance(content["data"][variable], int) ):
                                    dv = ua.DataValue(ua.Variant(content["data"][variable], server_variables[variable].get_data_type_as_variant_type()))
                                    server_variables[variable].set_value(dv)
                                else:
                                    server_variables[variable].set_value(content["data"][variable])
                            except Exception:
                                server_variables[variable].set_attribute(ua.AttributeIds.Value, ua.DataValue(content["data"][variable]))
        except Exception as e:
            log.exception(e)

```

ysimonx, 2022-02-08:
PR https://github.com/thingsboard/thingsboard-gateway/pull/737

ysimonx, 2022-02-09:
PR https://github.com/thingsboard/thingsboard-gateway/pull/737 merged :) thx @zbeacon

closed 2022-02-09 by ysimonx