thingsboard/thingsboard-gateway #726: Connecting a BLE device to the Thingsboard gateway[BUG]

Pan8284, 2022-02-03:
Hi Team,

Thanks for the response

here is the error I am getting after putting latest build or upgrading the thingsboard-gateway w.r.t ble connector issue

 

 

smartcity@smartcity-VirtualBox:/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/ble$ sudo thingsboard-gateway
[STREAM ONLY] 2022-02-03 16:58:35,676 - ERROR - [device.py] - device - 183 - Characteristic with UUID dac508b1-ee98-417c-95e1-f4f464568ac3 could not be found!
[STREAM ONLY] 2022-02-03 16:58:35,676 - ERROR - [device.py] - device - 183 - Characteristic with UUID 1d923677-d04e-4f47-9aea-2cf8b092581e could not be found!
[STREAM ONLY] 2022-02-03 16:58:35,676 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b9091d could not be found!
[STREAM ONLY] 2022-02-03 16:58:40,699 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b9091d could not be found!
[STREAM ONLY] 2022-02-03 16:58:45,728 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b9091d could not be found!

 

I am using BLE Scanner app which has Advertiser to send data. UUID mentioned above are UUID’s in my advertiser.

 

I used this link https://github.com/thingsboard/thingsboard-gateway/tree/master/thingsboard_gateway/connectors/ble

 

I have updated every file under 

/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/ble directory

 

pls help

samson0v, 2022-02-03:
Hi @Pan8284, please, send your ble config file and logs from the device scanner (in device config section set `"showMap": true`).
Thanks for your interest in ThingsBoard IoT Gateway.

Pan8284, 2022-02-04:
PFA ble config file. Pls confirm how to get device scanner logs (url)

{
    "name": "BLE Connector",
    "rescanIntervalSeconds": 100,
    "checkIntervalSeconds": 100,
    "scanTimeSeconds": 5,
    "passiveScanMode": true,
    "devices": [
        {
            "name": "BLETE",
            "MACAddress": "C0:EE:FB:F3:AA:43",
            "addrType":"public",
            "telemetry": [
                {
                    "key": "DEVICETOCLOUD",
                    "method": "notify",
                    "characteristicUUID": "dac508b1-ee98-417c-95e1-f4f464568ac3"
                },
                {
                    "key": "DTC",
                    "method": "notify",
                    "characteristicUUID": "1d923677-d04e-4f47-9aea-2cf8b092581e"
                }
            ],
            "attributes": [
                {
                    "key": "name",
                    "characteristicUUID": "692330a1-5f10-418c-888b-5426d8b9091d",
                    "method": "read",
                    "byteFrom": 0,
                    "byteTo": -1
                }
            ],
            "attributeUpdates": [
                {
                    "attributeOnThingsBoard": "sharedName",
                    "characteristicUUID": "692330a1-5f10-418c-888b-5426d8b9091d"
                }
            ],
            "serverSideRpc": [
                {
                    "methodRPC": "sharedName",
                    "withResponse": true,
                    "characteristicUUID": "692330a1-5f10-418c-888b-5426d8b9091d",
                    "methodProcessing": "write"
                }
            ]
        }
    ]
}

samson0v, 2022-02-07:
Hi @Pan8284, you have to use `"showMap": true` in the device section config, for example:
[image]

Pan8284, 2022-02-07:
I have set the showMap:true but from where to find the logs

samson0v, 2022-02-08:
Hi @Pan8284, you get logs in the place where got them.
[image]

Pan8284, 2022-02-08:
Logs are the same as shared earlier even after setting the ShowMap:true

samson0v, 2022-02-09:
@Pan8284, in the logs, have to be something like that:
[image]

Pan8284, 2022-02-09:
Can we connect over call as today my Bluetooth is not getting with System....
I am working on VM..

So I'll not able to share the logs with you. 

Pls suggest

samson0v, 2022-02-09:
@Pan8284, why are you sending the tail of the logs? The logs you should get are +- at the head of logs.

Pan8284, 2022-02-09:
sudo systemctl status thingsboard-gateway.service
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-02-09 18:39:36 IST; 3s ago
   Main PID: 30925 (python3)
      Tasks: 10 (limit: 4632)
     Memory: 28.6M
     CGroup: /system.slice/thingsboard-gateway.service
             └─30925 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Feb 09 18:39:36 smartcity-VirtualBox systemd[1]: Started ThingsBoard Gateway.
Feb 09 18:39:38 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:38,791 - ERROR - [device.py] - device - 183 - Characteristic with UUID dac508b1-ee98-417c-95e1-f4f464568>
Feb 09 18:39:38 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:38,791 - ERROR - [device.py] - device - 183 - Characteristic with UUID 1d923677-d04e-4f47-9aea-2cf8b0925>
Feb 09 18:39:38 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:38,791 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b90>
Feb 09 18:39:38 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:38,791 - ERROR - [device.py] - device - 183 - Characteristic with UUID 1d923677-d04e-4f47-9aea-2cf8b0925>
Feb 09 18:39:38 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:38,791 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b90>
Feb 09 18:39:43 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:43,902 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b90>
Feb 09 18:39:49 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:49,085 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b90>
Feb 09 18:39:54 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:54,260 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b90>
Feb 09 18:39:59 smartcity-VirtualBox python3[30925]: [STREAM ONLY] 2022-02-09 18:39:59,449 - ERROR - [device.py] - device - 171 - Characteristic with UUID 692330a1-5f10-418c-888b-5426d8b90>

Logs are not creating in var/log/thingsboard-gateway/

I am just sending the output of 

sudo systemctl status thingsboard-gateway

Pan8284, 2022-02-09:
when I do systemctl status thingsboard-gateway then output is different to when I do thingsboard-gateway

Why?

samson0v, 2022-02-09:
@Pan8284, try to use the next commands:
sudo chown thingsboard_gateway. /var/log/thingsboard-gateway  -R
sudo systemctl restart thingsboard-gateway

Pan8284, 2022-02-10:
Its the same logs are not getting created.... Can we get on the call to solve it.
Its been long pending

Pan8284, 2022-02-15:
Any response ? Waiting to hear from you.

samson0v, 2022-02-15:
Hi @Pan8284, I was sick, sorry for the late response. We don't know what the problem is. You can send credentials for access to your server on my email vbidochka@thingsboard.io and we will investigate.

Pan8284, 2022-02-15:
I am running this whole thing on VM on my laptop, I am not sure how u can access the same...

Pan8284, 2022-02-16:
Pls check is it visible to u

IP: 192.168.1.7
Username: smartcity
password: Pan1701234!

samson0v, 2022-02-16:
@Pan8284, you have to make port forwarding and send me the outer IP address. And please send them to my email (vbidochka@thingsboard.io), not here.

samson0v, 2022-02-23:
Hi @Pan8284, any updates on this issue? Can we close the issue?

Pan8284, 2022-02-23:
Actually I dont know how to provide you the access on VM as Its on my laptop. Let me check and revert.

Pan8284, 2022-02-28:
We can do the I can share the scene and give control to you, pls suggest....

samson0v, 2022-02-28:
Hi @Pan8284, sorry for the late response, we have some problem with russian dolboebamy) 
There is some other idea, you can separately run bleak and scan the charId. Please, go to Bleak repository and there are examples. When you finish to scan and take the charId, you can easily paste it to Gateway config file.

Pan8284, 2022-02-28:
I guess my approach to be on webex is good because you will get to operate directly on my machine. It will save both of us time.

I also ran the gatttool and I am getting the output as expected. 
The same is not happening with Thingsboard gateway.

Pls come on webex issue will be resolved quickly

Pan8284, 2022-03-04:
Any response

samson0v, 2022-03-09:
Hi @Pan8284, ok, let's do it via Webex. I can at 14:00 (Kyiv time) tomorrow.

Pan8284, 2022-03-09:
I respect you, In such a situation you are still working.  My VM is not working now, I am trying to fix the issue.
If It got fixed I'll share the invite.

Tc of yourself and your family.

Regards
Pankaj Chaudhary

samson0v, 2022-03-09:
@Pan8284, thanks for your support! 🇺🇦

closed 2022-05-20 by imbeacon