thingsboard/thingsboard-gateway #729: [BUG] Modbus read_holding_registers (funcCode 3) returns Exception IllegalAddress

jeffreylauck, 2022-02-07:
**Describe the bug**
When I use the modbus connector to read from a modbus slave with function code 3 using the below config, an error is returned.

I found that if I remove "* 8" in `connectors/modbus/modbus_connector.py`, the code will run correctly. May I check if "* 8" can be removed? (The "*8" was recently added in #716 )

Tried with PyModbus REPL, the `count` parameter need not be multiplied by 8. If multiplied by 8, it will result in IllegalAddress exception.
[image]

```
if function_code in (1, 2, 3, 4):
            result = device.config['available_functions'][function_code](address=config[ADDRESS_PARAMETER],
                                                                         count=config.get(OBJECTS_COUNT_PARAMETER,
                                                                                          config.get("registersCount",
                                                                                                     config.get(
                                                                                                         "registerCount",
                            this is the problematic "*8" -->                     1))) * 8,
                                                                         unit=device.config['unitId'])
```

Also on line 411, can we reconfirm if the "*8" is necessary?

```
elif function_code in (5, 15):
            result = device.config['available_functions'][function_code](address=config[ADDRESS_PARAMETER] * 8,
                                                                         value=config[PAYLOAD_PARAMETER],
                                                                         unit=device.config['unitId'])
```

Thank you!

Below log for your reference
```
""2022-02-07 17:57:00" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 100 - Gateway starting..."
""2022-02-07 17:57:00" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 105 - ThingsBoard IoT gateway version: 3.0.1"
""2022-02-07 17:57:00" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 176 - Subscribed to *|* with id 1 for device *"
""2022-02-07 17:57:01" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 139 - connection SUCCESS"
""2022-02-07 17:57:04" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import ModbusConnector from /usr/local/lib/python3.7/dist-packages/thingsboard_gateway-3.0.1-py3.7.egg/thingsboard_gateway/connectors/modbus."
""2022-02-07 17:57:04" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 194 - Gateway started."
""2022-02-07 17:57:04" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __process_remote_logging_update - 341 - Remote logging has being updated. Current logging level is: DEBUG "
""2022-02-07 17:57:04" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 82 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:09" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
NoneType: None
""2022-02-07 17:57:09" - |ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 424 - Exception Response(131, 3, IllegalAddress)"
[... 116 lines cut ...]
        "host": null,
        "port": "/dev/ttyS0",
        "type": "serial",
        "method": "rtu",
        "timeout": 10,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": null,
        "retryOnEmpty": null,
        "retryOnInvalid": null,
        "baudrate": 9600,
        "pollPeriod": 5000
      }
    ]
  },
  "slave": null
}
```

**Error traceback (If available):**

**Versions (please complete the following information):**
 - OS: Raspbian Buster
 - Thingsboard IoT Gateway version: 3.0.1
 - Python version: 3.7

jeffreylauck, 2022-02-08:
I have submitted a PR #732 for this issue, please check and comment. Thanks!

closed 2022-02-08 by jeffreylauck

jiang-sss, 2022-05-22:
> I have submitted a PR #732 for this issue, please check and comment. Thanks!

I use the latest version of the source code also appeared this problem, I would like to ask how you solve it

jeffreylauck, 2022-05-23:
Since thingsboard-gateway modbus connector is using pymodbus library, I would suggest you try using the pymodbus REPL to connect to your modbus device and see how it responds. Sometimes "Illegal Address" exception happens because the input address is "really wrong" (happened to me a few times, the user manual of the device says the address is 40001 for example, but in fact it means using function code 4 to read address 1...) Do try the REPL method and see if the configs work first, before running it with the gateway connector

Link: https://pymodbus.readthedocs.io/en/latest/source/library/REPL.html

DuXiaoChuang, 2024-03-26:
> Since thingsboard-gateway modbus connector is using pymodbus library, I would suggest you try using the pymodbus REPL to connect to your modbus device and see how it responds. Sometimes "Illegal [...]

I'm having the same problem as you, how should I solve it, the modbus address is real, but the prompt illegaladdress.