thingsboard/thingsboard-gateway #730: Raspberry 4 - Thinkboard installed but not starting

herken, 2022-02-07:
**Describe the issue**
Hello,
i just installed thinkboard and configured it but the service never start:

systemctl status thingsboard-gateway                     
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2022-02-07 10:42:17 UTC; 8s ago
  Process: 5192 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILU
 Main PID: 5192 (code=exited, status=1/FAILURE)

fév 07 10:42:17 rb0004 systemd[1]: thingsboard-gateway.service: Service RestartSec=100ms expired, scheduling restart.
fév 07 10:42:17 rb0004 systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
fév 07 10:42:17 rb0004 systemd[1]: Stopped ThingsBoard Gateway.
fév 07 10:42:17 rb0004 systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
fév 07 10:42:17 rb0004 systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
fév 07 10:42:17 rb0004 systemd[1]: Failed to start ThingsBoard Gateway.

I cleaned the installation and re install it with the default configuration file (no host, no token) but still not starting.

I searched in service.log but it's empty. (/var/log/thingsboard-gateway/service.log )

**Configuration** (Attach your configuration file) it's the default one
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector] no configured for now

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: Raspbian GNU/Linux 10 (buster)
 - Thingsboard IoT Gateway version ? The last one i suppose
 - Python version Python 2.7.16

Thank you !

zbeacon, 2022-02-07:
Hi @herken ,

Please try to find the error in output of command: `sudo journalctl` (run it after restart the gateway).

herken, 2022-02-07:
Hello,

here the output 
`sudo journalctl
-- Logs begin at Thu 2019-02-14 10:12:01 UTC, end at Mon 2022-02-07 11:49:46 UTC. --
fév 14 10:12:01 rb0004 kernel: Booting Linux on physical CPU 0x0
fév 14 10:12:01 rb0004 kernel: Linux version 4.19.97-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460
fév 14 10:12:01 rb0004 kernel: CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
fév 14 10:12:01 rb0004 kernel: CPU: div instructions available: patching division code
fév 14 10:12:01 rb0004 kernel: CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
fév 14 10:12:01 rb0004 kernel: OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
fév 14 10:12:01 rb0004 kernel: Memory policy: Data cache writealloc
fév 14 10:12:01 rb0004 kernel: cma: Reserved 256 MiB at 0x000000001ec00000
fév 14 10:12:01 rb0004 kernel: On node 0 totalpages: 1012736
fév 14 10:12:01 rb0004 kernel:   DMA zone: 1728 pages used for memmap
fév 14 10:12:01 rb0004 kernel:   DMA zone: 0 pages reserved
fév 14 10:12:01 rb0004 kernel:   DMA zone: 196608 pages, LIFO batch:63
fév 14 10:12:01 rb0004 kernel:   HighMem zone: 816128 pages, LIFO batch:63
fév 14 10:12:01 rb0004 kernel: random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
fév 14 10:12:01 rb0004 kernel: percpu: Embedded 17 pages/cpu s36928 r8192 d24512 u69632
fév 14 10:12:01 rb0004 kernel: pcpu-alloc: s36928 r8192 d24512 u69632 alloc=17*4096
fév 14 10:12:01 rb0004 kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
fév 14 10:12:01 rb0004 kernel: Built 1 zonelists, mobility grouping on.  Total pages: 1011008
fév 14 10:12:01 rb0004 kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=DC:A6:32:73:BA
fév 14 10:12:01 rb0004 kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
fév 14 10:12:01 rb0004 kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
fév 14 10:12:01 rb0004 kernel: Memory: 3735564K/4050944K available (8192K kernel code, 687K rwdata, 2408K rodata, 2048K init, 850K bss
fév 14 10:12:01 rb0004 kernel: Virtual kernel memory layout:
                                    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                    vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                                    lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                                    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                                    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                                      .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                                      .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                                      .data : 0x(ptrval) - 0x(ptrval)   ( 688 kB)
                                       .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
fév 14 10:12:01 rb0004 kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
fév 14 10:12:01 rb0004 kernel: ftrace: allocating 28692 entries in 85 pages
fév 14 10:12:01 rb0004 kernel: rcu: Hierarchical RCU implementation.
fév 14 10:12:01 rb0004 kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
fév 14 10:12:01 rb0004 kernel: GIC: Using split EOI/Deactivate mode
fév 14 10:12:01 rb0004 kernel: arch_timer: cp15 timer(s) running at 54.00MHz (phys).
fév 14 10:12:01 rb0004 kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 44079520312
fév 14 10:12:01 rb0004 kernel: sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
fév 14 10:12:01 rb0004 kernel: Switching to timer-based delay loop, resolution 18ns
fév 14 10:12:01 rb0004 kernel: Console: colour dummy device 80x30
fév 14 10:12:01 rb0004 kernel: console [tty1] enabled
fév 14 10:12:01 rb0004 kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
fév 14 10:12:01 rb0004 kernel: pid_max: default: 32768 minimum: 301
fév 14 10:12:01 rb0004 kernel: Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
fév 14 10:12:01 rb0004 kernel: Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
fév 14 10:12:01 rb0004 kernel: CPU: Testing write buffer coherency: ok
fév 14 10:12:01 rb0004 kernel: CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
fév 14 10:12:01 rb0004 kernel: Setting up static identity map for 0x200000 - 0x20003c
fév 14 10:12:01 rb0004 kernel: rcu: Hierarchical SRCU implementation.
fév 14 10:12:01 rb0004 kernel: smp: Bringing up secondary CPUs ...
fév 14 10:12:01 rb0004 kernel: CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
fév 14 10:12:01 rb0004 kernel: CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
fév 14 10:12:01 rb0004 kernel: CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
fév 14 10:12:01 rb0004 kernel: smp: Brought up 1 node, 4 CPUs
fév 14 10:12:01 rb0004 kernel: SMP: Total of 4 processors activated (432.00 BogoMIPS).
fév 14 10:12:01 rb0004 kernel: CPU: All CPU(s) started in HYP mode.
fév 14 10:12:01 rb0004 kernel: CPU: Virtualization extensions available.
fév 14 10:12:01 rb0004 kernel: devtmpfs: initialized
fév 14 10:12:01 rb0004 kernel: VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
lines 1-63`

I don't found the error? 

Thank you for help

zbeacon, 2022-02-10:
Unfortunately, there is not error, related to the gateway, please try to stop the gateway, using `sudo systemctl stop thingsboard-gateway`, move to /etc/thingsboard-gateway folder using `cd /etc/thingsboard-gateway/` and run the gateway using `sudo thingsboard-gateway`. Will it run and what is the output?

herken, 2022-02-10:
Hello,

here the output 
`eall-user@rb0004:/etc/thingsboard-gateway $ sudo thingsboard-gateway
Traceback (most recent call last):
  File "/usr/bin/thingsboard-gateway", line 33, in <module>
    sys.exit(load_entry_point('thingsboard-gateway==3.0.1', 'console_scripts', 'thingsboard-gateway')())
  File "/usr/bin/thingsboard-gateway", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/local/lib/python3.7/dist-packages/importlib_metadata/__init__.py", line 194, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 32, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
    from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 21, in <module>
    import grpc
  File "/usr/lib/python3/dist-packages/grpc/__init__.py", line 22, in <module>
    from grpc._cython import cygrpc as _cygrpc
  File "src/python/grpcio/grpc/_cython/cygrpc.pyx", line 18, in init grpc._cython.cygrpc
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3191, in <module>
    @_call_aside
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'PyInquirer' distribution was not found and is required by thingsboard-gateway
`

I followed the tutorial here : https://thingsboard.io/docs/iot-gateway/install/deb-installation/

Whats is missing? Thank you for help

zbeacon, 2022-02-17:
Hi @herken ,

Please try to run the following command and restart the gateway:
`sudo pip3 install --upgrade protobuf`

samson0v, 2022-02-23:
Hi @herken, any updates on this issue? Can we close id?

herken, 2022-02-24:
Hello it's started thank you !

closed 2022-03-04 by zbeacon