thingsboard/thingsboard-gateway #744: Cannot receive modbus slave telemetry data

pystudy123, 2022-02-12:
**Describe the issue**
I deployed thingsboard and tb-gateway based on docker under win10.
Execute under cmd: curl -v -X POST -d "{\"temperature\":1.25}" http://127.0.0.1:8080/api/v1/T2_TEST_TOKEN/telemetry --header "Content-Type:application/ json"
Everything works fine and data can be received.
However, I use Modbus Slave simulation, unable to receive telemetry data.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**
tb_gateway.yaml:
`thingsboard:
  host: 127.0.0.1
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: ""
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
  -
    name: Modbus Connector
    type: modbus
    configuration: modbus.json`

modbus.json:
`{
    "server":{
        "type":"tcp",
        "host":"127.0.0.1", 
        "port":502,
        "timeout":35, 
        "method":"socket",
        "byteOrder":"BIG",
        "devices":[ 
            {
                "unitId":1,
                "deviceName":"testDevice",
                "attributesPollPeriod":5000,
                "timeseriesPollPeriod":5000,
                "sendDataOnlyOnChange":false,
                "attributes":[ 
                    {
                        "byteOrder":"BIG",
                        "tag":"count",
                        "type":"16int", 
                        "functionCode":3,
                        "objectsCount":1,
                        "address":0 
                    }
                ],
                "timeseries":[
                    {
                        "byteOrder":"BIG",
                        "tag":"count",
                        "type":"16int",
                        "functionCode":3,
                        "objectsCount":1,
                        "address":0
                    }
                ]
            },
            {
                "unitId":2,
                "deviceName":"testDevice2",
                "attributesPollPeriod":5000,
                "timeseriesPollPeriod":5000,
                "sendDataOnlyOnChange":false,
                "attributes":[
                    {
                        "byteOrder":"BIG",
                        "tag":"count",
                        "type":"16int",
                        "functionCode":3,
                        "objectsCount":1,
                        "address":0
                    }
                ],
                "timeseries":[
                    {
                        "byteOrder":"BIG",
                        "tag":"count",
                        "type":"16int",
                        "functionCode":3,
                        "objectsCount":1,
                        "address":0
                    }
                ],
                "rpc":[ 
                   {
                        "tag":"setValue", 
                        "type":"16int",
                        "functionCode":6,
                        "objectsCount":1,
                        "address":8 
                    },
                    {
                        "tag":"getValue",
                        "type":"16int",
                        "functionCode":1,
                        "objectsCount":1,
                        "address":9
                    }
                ]
            }
        ]
    }
}`

pystudy123, 2022-02-12:
modbus.json:
```{
    "server":{
        "type":"tcp",
        "host":"127.0.0.1", 
        "port":502,
        "timeout":35, 
        "method":"socket",
        "byteOrder":"BIG",
        "devices":[ 
            {
                "unitId":1,
                "deviceName":"testDevice",
                "attributesPollPeriod":5000,
                "timeseriesPollPeriod":5000,
                "sendDataOnlyOnChange":false,
                "attributes":[ 
                    {
                        "byteOrder":"BIG",
                        "tag":"count",
                        "type":"16int", 
                        "functionCode":3,
                        "objectsCount":1,
                        "address":0 
                    }
                ],
                "timeseries":[
                    {
                        "byteOrder":"BIG",
                        "tag":"count",
                        "type":"16int",
                        "functionCode":3,
                        "objectsCount":1,
                        "address":0
                    }
                ]
            },
            {
                "unitId":2,
                "deviceName":"testDevice2",
                "attributesPollPeriod":5000,
                "timeseriesPollPeriod":5000,
                "sendDataOnlyOnChange":false,
                "attributes":[
                    {
                        "byteOrder":"BIG",
                        "tag":"count",
                        "type":"16int",
                        "functionCode":3,
                        "objectsCount":1,
                        "address":0
                    }
                ],
                "timeseries":[
                    {
                        "byteOrder":"BIG",
                        "tag":"count",
                        "type":"16int",
                        "functionCode":3,
                        "objectsCount":1,
                        "address":0
                    }
                ],
                "rpc":[ 
                   {
                        "tag":"setValue", 
                        "type":"16int",
                        "functionCode":6,
                        "objectsCount":1,
                        "address":8 
                    },
                    {
                        "tag":"getValue",
                        "type":"16int",
                        "functionCode":1,
                        "objectsCount":1,
                        "address":9
                    }
                ]
            }
        ]
    }
}`

pystudy123, 2022-02-12:
12 06:11:14" - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _connect_with_connectors - 473 - Connector implementation not found for Modbus Connector"

santobedi, 2022-04-07:
Hi @imbeacon  @pystudy123, 

I'm exactly in a similar situation now. How did you solve this issue?