thingsboard/thingsboard-gateway #746: [BUG] OPC UA identity cert.pem with Basic256Sha256 needs application_uri = "urn:xxxxx"

ysimonx, 2022-02-15:
**OPC UA with Basic256Sha256 needs application_uri = "urn:xxxxx"**
I am trying to connect an opc ua server, with a Basic256Sha256 connection

**OPC-UA Connector:**

**Error traceback:**
```
""2022-02-15 16:45:55" - |ERROR| - [opcua_connector.py] - opcua_connector - __connect - 137 - "The URI specified in the ApplicationDescription does not match the URI in the certificate."(BadCertificateUriInvalid)"
```

**Versions (please complete the following information):**
 - OS: Mac OSx
 - Thingsboard IoT Gateway version 3.0.1, last master branch
 - Python version 3.9
 
 config is : 
```
 
     "identity": {
      "type": "cert.PEM",
      "privateKey": "/Users/ysimonx/Developpement-eurevia/x509v3/key.pem", 
      "cert": "/Users/ysimonx/Developpement-eurevia/x509v3/cert.pem",
      "mode": "SignAndEncrypt",
      "username": "operator",
      "password": "xxxxxxx"
    },
 
```

ysimonx, 2022-02-15:
```

    def __set_auth_settings_by_cert(self):
        try:
            ca_cert = self.__server_conf["identity"].get("caCert")
            private_key = self.__server_conf["identity"].get("privateKey")
            cert = self.__server_conf["identity"].get("cert")
            security_mode = self.__server_conf["identity"].get("mode", "SignAndEncrypt")
            policy = self.__server_conf["security"]
            if cert is None or private_key is None:
                log.exception("Error in ssl configuration - cert or privateKey parameter not found")
                raise RuntimeError("Error in ssl configuration - cert or privateKey parameter not found")
            security_string = policy + ',' + security_mode + ',' + cert + ',' + private_key
            log.info("yannick security_string = %s", security_string)
            if ca_cert is not None:
               security_string = security_string + ',' + ca_cert
            self.client.set_security_string(security_string)

```

if I add this hardcoded line

`self.client.application_uri = "urn:192.168.85.19"`

the connection is done

ysimonx, 2022-02-15:
I tested with a python script ... the application_uri should be specified ...

```
from opcua import Client, ua

url = "opc.tcp://10.8.0.5:4841"
client = Client(url)

client.set_user("operator")
client.set_password("xxxxxxxx")
client.set_security_string("Basic256Sha256,SignAndEncrypt,cert.pem,key.pem")
client.application_uri = "urn:192.168.85.19"

client.connect()
print(f"Connected to: {url}")

client.disconnect()
```

So, I think that thingsboard should consider adding this application_uri parameter to opcua.json

samson0v, 2022-02-16:
Hi @ysimonx, thanks for your interest in ThingsBoard IoT Gateway! We will add it in the short future.

closed 2022-02-16 by zbeacon