thingsboard/thingsboard-gateway #749: Is it possible to change convert HEX text string to JSON format in Thingsboard Rule Engine?

debacs83, 2022-02-22:
The protocol is HTTP(S), the data format is JSON HEX TEXT. Is it possible to convert that to ThingsBoard compatible JSON format by Rule Engine? Or I have to write a convert module for that? Have read there is some 'Converter' in PE edition? Does this converter has such data transformation abilities? Even if it cannot be done in existing ThingsBoard functions/modules any hints how to approach would be very helpful.  ThingsBoard GW has some conversion capabilities, so I post here. Sorry if it is a wrong forum. 

Regards
Deb

zbeacon, 2022-02-23:
Hi @debacs83,

It is a really wrong repository, but AFAIK you can use Script rule node to convert messages in rule engine.

debacs83, 2022-02-23:
Thank you and sorry for the posting in wrong repository. I will be careful next time.

closed 2022-03-04 by zbeacon