thingsboard/thingsboard-gateway #750: [HELP] CustomRestUplinkConverter is not called

FacelookHK, 2022-02-22:
**Describe the issue**
I created a custom_rest_uplink_converter.py file and put it in the extensions/rest folder but it is not called.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
REST connector

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.0.1
 - Python version 3.9

rduivenvoorde, 2022-02-22:
I second this: running 3.0.1 docker gateway

I created a custom mqtt_converter called: `SensorcomputerNetworkConverter` in a file `sensorcomputer_network_converter.py`

Used (at first):

```
"topicFilter": "sensorcomputer/+/network",
"converter": {
   "type": "custom",
   "extension": "SensorcomputerNetworkConverter",
   "extension-config": {}
}
```

and it was never loaded, and I kept seeing `Cannot find converter for sensorcomputer/+/network topic"`

By going into the container and trying to call `TBModuleLoader.find_paths()` and `TBModuleLoader.import_module(...)` I found out some (I think) issues:

- although in all docs it writes that 'type' should be 'custom', I could only  load my Class when I used type 'mqtt'.

- although I have a volume /thingsboard_gateway/extensions/mqtt this path does NOT pop up in the TBModuleLoader.find_paths() ...  When I output find_paths() in my container I see:
```[
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions', 
'/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors']"
```

Note:  it is copying/adding the default path several times

Note2: it does NOT contain the `/thingsboard_gateway/extensions/` or `/thingsboard_gateway/extensions/mqtt` in which I put the custom Class?

So I only got this to work when I add it in `/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/extensions`

So to me too, it is not clear where to put custom python converters when using the docker container.

Tried: `ADD ./extensions/* /thingsboard_gateway/extensions` in a custom docker build file
And now run the container with: `docker run -it -v ~/thingsboard_gateway/extensions:/thingsboard_gateway/extensions  ...`
but both are apparently not picked up?

rduivenvoorde, 2022-02-22:
Mmm, looking to the original issue, these are probably very different issues. Maybe create a separate one for mine?

FacelookHK, 2022-02-22:
> I second this: running 3.0.1 docker gateway
>
> I created a custom mqtt_converter called: `SensorcomputerNetworkConverter` in a file `sensorcomputer_network_converter.py`
>
> Used (at first):
>
> ``` [...]

if you are using docker, you need to bind the /var/lib/thingsboard_gateway/extensions folder instead.

FacelookHK, 2022-02-22:
For my original issue, I got the mqtt custom converter working but not rest custom converter.

rduivenvoorde, 2022-02-22:
@FacelookHK 

There is no: /var/lib/thingsboard_gateway/extensions in the container currently. Should it be?

About your error:

```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

you did not add your rest.json, but in the rest.json I have here, `deviceName` is never used, but `deviceNameExpression`  is:

```
       "deviceNameExpression": "SuperAnonDevice",
        "deviceTypeExpression": "default",
```

Could that be the issue?

Note that for what I know you cannot assign people to your issues (as in: I do not think you should assign @zbeacon to your issue...)

FacelookHK, 2022-02-22:
> @FacelookHK
>
> There is no: /var/lib/thingsboard_gateway/extensions in the container currently. Should it be?
>
> About your error:
>
> ```
> 'deviceName'
> Traceback (most recent call last):
>   File "<input>", [...]

You have to do these steps.

1) docker run -it -v ~/.tb-gateway/logs:/thingsboard_gateway/logs -v ~/.tb-gateway/extensions:/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/thingsboard_gateway/config --name tb-gateway --restart always thingsboard/tb-gateway

2) Exit and remove the container

3) docker run -it --network host -v ~/.tb-gateway/extensions:/var/lib/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/thingsboard_gateway/config --name tb-gateway --restart always thingsboard/tb-gateway

FacelookHK, 2022-02-22:
I seem to have found the issue.

In rest_connector.py on line 77 and 78, instead of

            converter = TBModuleLoader.import_module(self._connector_type,
                                                     mapping.get("extension", self._default_converters["uplink"]))

it should be

            converter = TBModuleLoader.import_module(self._connector_type,
                                                     mapping["converter"].get("extension", self._default_converters["uplink"]))

samson0v, 2022-02-23:
Hi @FacelookHK, thanks for your interest in ThingsBoard IoT Gateway! You are right, it is the bug. We will fix it in the nearest future.

closed 2022-02-23 by zbeacon