thingsboard/thingsboard-gateway #752: [BUG] Request connector device name Json Expression not working

RiccardoFelicetti, 2022-02-22:
**Describe the bug**
Using json expression in "deviceNameJsonExpression" field is not working. 
With this configuration:
`"converter": {
        "type": "json",
        "deviceNameJsonExpression": "${name}",
        "deviceTypeJsonExpression": "${type}",
        "telemetry": [
          {
            "key": "intensity",
            "type": "int",
            "value": "${intensity}"
          }
        ]
      }`

I get the correct device type, but the device will be created with `name` as name.

**Connector name (If bug in the some connector):**
[request connector]

**Versions (please complete the following information):**
 - Thingsboard IoT Gateway version [Latest Docker image]

samson0v, 2022-02-23:
Hi @RiccardoFelicetti, send, please, your configuration, server response and Gateway logs (set the logs level to DEBUG instead of INFO in logs.conf file).
Thanks for your interest in ThingsBoard IoT Gateway!

RiccardoFelicetti, 2022-02-23:
*Part of logs, after the configuration update*
`tb-gateway_1  | ""2022-02-23 08:20:28" - DEBUG - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - 296 - Logs configuration has been updated."
tb-gateway_1  | ""2022-02-23 08:20:28" - DEBUG - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - 105 - Current configuration has been sent to ThingsBoard: {"request": [{"name": "Request", "config": {"host": "https://webhook.site/", "security": {"type": "anonymous"}, "mapping": [{"url": "0f11d003-3e5c-45a9-8e30-9bfaf04b2625?", "httpMethod": "GET", "httpHeaders": {"ACCEPT": "application/json"}, "allowRedirects": true, "timeout": 0.5, "scanPeriod": 60, "converter": {"type": "json", "deviceNameJsonExpression": "${name}", "deviceTypeJsonExpression": "${name}", "telemetry": [{"key": "intensity", "type": "int", "value": "${intensity}"}]}}]}}], "thingsboard": {"thingsboard": {"host": "mytb", "remoteConfiguration": true, "port": 1883, "security": {"accessToken": "9E4K5d3HAg25o1vL26Su"}, "remoteShell": true}, "connectors": [{"configuration": "request.json", "name": "Request", "type": "request"}], "storage": {"type": "memory", "read_records_count": 100, "max_records_count": 100000}, "logs": "W2xvZ2dlcnNdfX1rZXlzPXJvb3QsIHNlcnZpY2UsIGNvbm5lY3RvciwgY29udmVydGVyLCB0Yl9jb25uZWN0aW9uLCBzdG9yYWdlLCBleHRlbnNpb259fVtoYW5kbGVyc119fWtleXM9Y29uc29sZUhhbmRsZXIsIHNlcnZpY2VIYW5kbGVyLCBjb25uZWN0b3JIYW5kbGVyLCBjb252ZXJ0ZXJIYW5kbGVyLCB0Yl9jb25uZWN0aW9uSGFuZGxlciwgc3RvcmFnZUhhbmRsZXIsIGV4dGVuc2lvbkhhbmRsZXJ9fVtmb3JtYXR0ZXJzXX19a2V5cz1Mb2dGb3JtYXR0ZXJ9fVtsb2dnZXJfcm9vdF19fWxldmVsPUVSUk9SfX1oYW5kbGVycz1jb25zb2xlSGFuZGxlcn19W2xvZ2dlcl9jb25uZWN0b3JdfX1sZXZlbD1ERUJVR319aGFuZGxlcnM9Y29ubmVjdG9ySGFuZGxlcn19Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19cXVhbG5hbWU9Y29ubmVjdG9yfX1bbG9nZ2VyX3N0b3JhZ2VdfX1sZXZlbD1ERUJVR319aGFuZGxlcnM9c3RvcmFnZUhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fXF1YWxuYW1lPXN0b3JhZ2V9fVtsb2dnZXJfdGJfY29ubmVjdGlvbl19fWxldmVsPURFQlVHfX1oYW5kbGVycz10Yl9jb25uZWN0aW9uSGFuZGxlcn19Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19cXVhbG5hbWU9dGJfY29ubmVjdGlvbn19W2xvZ2dlcl9zZXJ2aWNlXX19bGV2ZWw9REVCVUd9fWhhbmRsZXJzPXNlcnZpY2VIYW5kbGVyfX1mb3JtYXR0ZXI9TG9nRm9ybWF0dGVyfX1xdWFsbmFtZT1zZXJ2aWNlfX1bbG9nZ2VyX2NvbnZlcnRlcl19fWxldmVsPURFQlVHfX1oYW5kbGVycz1jb252ZXJ0ZXJIYW5kbGVyfX1mb3JtYXR0ZXI9TG9nRm9ybWF0dGVyfX1xdWFsbmFtZT1jb252ZXJ0ZXJ9fVtsb2dnZXJfZXh0ZW5zaW9uXX19bGV2ZWw9REVCVUd9fWhhbmRsZXJzPWNvbm5lY3RvckhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fXF1YWxuYW1lPWV4dGVuc2lvbn19W2hhbmRsZXJfY29uc29sZUhhbmRsZXJdfX1jbGFzcz1TdHJlYW1IYW5kbGVyfX1sZXZlbD1ERUJVR319Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19YXJncz0oc3lzLnN0ZG91dCwpfX1baGFuZGxlcl9jb25uZWN0b3JIYW5kbGVyXX19bGV2ZWw9REVCVUd9fWNsYXNzPWxvZ2dpbmcuaGFuZGxlcnMuVGltZWRSb3RhdGluZ0ZpbGVIYW5kbGVyfX1mb3JtYXR0ZXI9TG9nRm9ybWF0dGVyfX1hcmdzPSgiLi9sb2dzL2Nvbm5lY3Rvci5sb2ciLCAiZCIsIDEsIDcsKX19W2hhbmRsZXJfc3RvcmFnZUhhbmRsZXJdfX1sZXZlbD1ERUJVR319Y2xhc3M9bG9nZ2luZy5oYW5kbGVycy5UaW1lZFJvdGF0aW5nRmlsZUhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fWFyZ3M9KCIuL2xvZ3Mvc3RvcmFnZS5sb2ciLCAiZCIsIDEsIDcsKX19W2hhbmRsZXJfc2VydmljZUhhbmRsZXJdfX1sZXZlbD1ERUJVR319Y2xhc3M9bG9nZ2luZy5oYW5kbGVycy5UaW1lZFJvdGF0aW5nRmlsZUhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fWFyZ3M9KCIuL2xvZ3Mvc2VydmljZS5sb2ciLCAiZCIsIDEsIDcsKX19W2hhbmRsZXJfY29udmVydGVySGFuZGxlcl19fWxldmVsPURFQlVHfX1jbGFzcz1sb2dnaW5nLmhhbmRsZXJzLlRpbWVkUm90YXRpbmdGaWxlSGFuZGxlcn19Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19YXJncz0oIi4vbG9ncy9jb252ZXJ0ZXIubG9nIiwgImQiLCAxLCAzLCl9fVtoYW5kbGVyX2V4dGVuc2lvbkhhbmRsZXJdfX1sZXZlbD1ERUJVR319Y2xhc3M9bG9nZ2luZy5oYW5kbGVycy5UaW1lZFJvdGF0aW5nRmlsZUhhbmRsZXJ9fWZvcm1hdHRlcj1Mb2dGb3JtYXR0ZXJ9fWFyZ3M9KCIuL2xvZ3MvZXh0ZW5zaW9uLmxvZyIsICJkIiwgMSwgMywpfX1baGFuZGxlcl90Yl9jb25uZWN0aW9uSGFuZGxlcl19fWxldmVsPURFQlVHfX1jbGFzcz1sb2dnaW5nLmhhbmRsZXJzLlRpbWVkUm90YXRpbmdGaWxlSGFuZGxlcn19Zm9ybWF0dGVyPUxvZ0Zvcm1hdHRlcn19YXJncz0oIi4vbG9ncy90Yl9jb25uZWN0aW9uLmxvZyIsICJkIiwgMSwgMywpfX1bZm9ybWF0dGVyX0xvZ0Zvcm1hdHRlcl19fWZvcm1hdD0iJShhc2N0aW1lKXMgLSAlKGxldmVsbmFtZSlzIC0gWyUoZmlsZW5hbWUpc10gLSAlKG1vZHVsZSlzIC0gJShsaW5lbm8pZCAtICUobWVzc2FnZSlzIiB9fWRhdGVmbXQ9IiVZLSVtLSVkICVIOiVNOiVTIg=="}}"
tb-gateway_1  | ""2022-02-23 08:20:28" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 316 - Shared attributes received (configuration, RemoteLoggingLevel)."
tb-gateway_1  | ""2022-02-23 08:20:28" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 319 - Client attributes received (current_configuration)."
tb-gateway_1  | ""2022-02-23 08:20:28" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 123 - Disconnected client: <paho.mqtt.client.Client object at 0x7fdb2cac92e0>, user data: None, result code: 0"
tb-gateway_1  | ""2022-02-23 08:20:28" - ERROR - [request_connector.py] - request_connector - 220 - Timeout error on request https://webhook.site//0f11d003-3e5c-45a9-8e30-9bfaf04b2625?."
tb-gateway_1  | ""2022-02-23 08:20:28" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 595 - 'attributes'"
tb-gateway_1  | ""2022-02-23 08:21:27" - DEBUG - [request_connector.py] - request_connector - 186 - /0f11d003-3e5c-45a9-8e30-9bfaf04b2625?"
tb-gateway_1  | ""2022-02-23 08:21:27" - DEBUG - [request_connector.py] - request_connector - 188 - https://webhook.site//0f11d003-3e5c-45a9-8e30-9bfaf04b2625?"
tb-gateway_1  | ""2022-02-23 08:21:27" - DEBUG - [request_connector.py] - request_connector - 199 - https://webhook.site//0f11d003-3e5c-45a9-8e30-9bfaf04b2625?"
[... cut ...]
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${name}",
        "deviceTypeJsonExpression": "${name}",
        "telemetry": [
          {
            "key": "intensity",
            "type": "int",
            "value": "${intensity}"
          }
        ]
      }
    }
  ]
}`

**Server Response**
`{"intensity":70, "name":"TESTNAME01"}`

I also notice some delay between configuration update and telemetry update.
And different timeout error which I can't understand.
Thank you

samson0v, 2022-02-23:
@RiccardoFelicetti, you are right, it is the bug. We will fix it in the nearest feature.

closed 2022-02-23 by zbeacon

RiccardoFelicetti, 2022-02-23:
Thank you for the quick answer. Can you give me some information about the timeout error? Or shoud I open another issue?