thingsboard/thingsboard-gateway #755: Data from Modbus Connector cannot be sent to storage. [HELP]

e-canga, 2022-02-23:
**Describe the issue**
I am trying to setup the thingsboard-gateway with mobus.json and runs without any traceback. 

When I check the debug logs the converter seems to be working fine as following; 
2022-02-23 13:11:30,738 - DEBUG - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 87 - datatype: attributes key: power_OnVoltage value: 2500

Similar with the telemetry data.
2022-02-23 13:11:30,733 - DEBUG - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 87 - datatype: telemetry key: st_PhaseGridVoltage value: 3885

My problem is they cannot be sent to storage and shown under the devices.

When I check the service log files I see an error like this:

""2022-02-23 13:11:30" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - 'deviceType'"
""2022-02-23 13:11:30" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - list indices must be integers or slices, not dict"

**Configuration** (Attach your configuration file)
thingsboard:
  host: 127.0.0.1
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: MY_TOKEN
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json
#
  -
    name: Modbus Connector
    type: modbus
    configuration: modbus.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest

**Connector name (If you need help with some connector/converter):**
Modbus connector

{
  "master": {
    "slaves": [
      {
        "host": "192.168.2.38", 
        "port": 600,
        "type": "tcp",
        "method": "socket",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "unitId": 13,
        "deviceName": "invertor_SN1017",
        "deviceType": "invertor",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
	"attributes": [
	      {
            "tag": "power_OnVoltage",
            "type": "16uint",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 3043

          }
	],
        "timeseries": [

          {
            "tag": "st_PhaseGridVoltage",
            "type": "16uint",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 3015

          }

        ]
      }
    ]
  }
}

**Error traceback (If it was raised):**
No traceback

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.0.1
 - Python version 3.8.10

e-canga, 2022-02-25:
I have found out that the number of points defined in the modbus gateway is critical. 
If I stick with around 30 of 2Byte variables the converter output is sent to attributes and telemetry without any problems. But if I increase the number of variables or increase number of slaves I still get the following error.   

""2022-02-23 13:11:30" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - 'deviceType'"
""2022-02-23 13:11:30" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - list indices must be integers or slices, not dict"

I will mark the original post as bug.

samson0v, 2022-02-28:
Hi @e-canga, thanks for your interest, we will investigate it in the nearest feature.

e-canga, 2022-03-01:
I found this on change log from 30 Sept

Fix for checking message size to send to ThingsBoard. Added parameter "maxPayloadSizebytes" for messages from the gateway to ThingsBoard, default value is **4096** bytes.

Found out that in "tb_gateway_service.py" this variable is set to 1024. Changing this to 4096 fixed the problem for now. 

On line 541, it is like this.
max_data_size = self.__config["thingsboard"].get("maxPayloadSizeBytes", 4096)

Will test with multiple devices.

samson0v, 2022-03-09:
Hi @e-canga, you can specify this parameter in tb_gateway.yaml config file in thingsboard section. It is not necessary to change the code.

samson0v, 2022-03-09:
@e-canga any updates about your test? Can we close the issue?

e-canga, 2022-03-09:
hi @samson0v, thanks for your help. I did not see the parameter in the sample tb_gateway.yaml in the configuration page.

I have done my test, depending on the number of devices and addresses, I have to increase the max payload size. 
Using 8 Kbytes now. Honestly, I don't know the limit or impact of the max payload size on operation. But I guess the MQTT limit should be the highest limit, if this is the case I should have no problem.

If i keep the number smaller, I get an  exception.

Not sure but still think that there is a bug. 

Does the code normally divide the message into smaller messages using adopted_data if the following statement is true?

if self.__get_data_size(data) >= max_data_size:

In my case, this creates an exception.

closed 2022-03-10 by imbeacon