thingsboard/thingsboard-gateway #759: [BUG] BACnet NULL vaule write error

Kretiss, 2022-02-27:
**Describe the bug**
I am trying to clear priority in priorityArray in simulated device and when I use value with `"NULL"` or `""` I get an error described bellow. I think this is a bug, because clearing priority in bacnet is done by using `NULL` as value with desired priority (for example 15).

I was looking into the `tb_gateway_bacnet_appliacation.py` file and I **think that problem is on line 211**, because before this line is `if (isinstance(value, str) and value.lower() == 'null') or value is None` statement which assigns `null` as `value` when value is not defined or `"NULL"` is used as in my example , which seems to be ok.

I can do pull request, but I dont know where `datatype()` function came from.

[image]

**Connector name:**
BACnet connector

**Error traceback:**
```
27 12:32:02" - |ERROR| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - form_iocb - 216 - WriteProperty cast error: TypeError('invalid constructor datatype')"

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 211, in form_iocb
    value = datatype(value)

  File "/root/.local/lib/python3.9/site-packages/bacpypes/primitivedata.py", line 806, in __init__
    raise TypeError("invalid constructor datatype")

TypeError: invalid constructor datatype

""2022-02-27 12:32:02" - |ERROR| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - __general_cb - 143 - object: unknownObject"

NoneType: None

""2022-02-27 12:32:02" - |ERROR| - [bacnet_connector.py] - bacnet_connector - __rpc_response_cb - 184 - Received callback with Error: <bacpypes.apdu.Error(15,158) instance at 0xffffa16d2430>"

NoneType: None
```

**Versions:**
 - OS: [MacOS Monterey 12.2.1]
 - Thingsboard IoT Gateway version [3.0.1]
 - Python version: [3.10]

samson0v, 2022-02-28:
Hi @Kretiss, it will be nice if you prepare the PR for this issue.
Thanks for you interest in ThingsBoard IoT Gateway!

Kretiss, 2022-03-01:
Hi @samson0v, I have made pull request, check it out please. 
https://github.com/thingsboard/thingsboard-gateway/pull/761

closed 2022-03-04 by Kretiss